lardemua/atom #233: Save zed point clouds when collecting data

miguelriemoliveira, 2020-09-16:
Too late now. Will work on this tomorrow ok @aaguiar96 ?

miguelriemoliveira, 2020-09-17:
Hi @aaguiar96 ,

there is a difference in the topics for this new bag file 

Before it was:

```bash
 agrob rosbag info agrob_2020-03-12-10-09-49_1.bag
path:        agrob_2020-03-12-10-09-49_1.bag
version:     2.0
duration:    60.0s
start:       Mar 12 2020 10:09:49.87 (1584007789.87)
end:         Mar 12 2020 10:10:49.87 (1584007849.87)
size:        1.4 GB
messages:    25494
compression: none [854/854 chunks]
types:       sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/Image           [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /tf                                    20934 msgs    : tf2_msgs/TFMessage          (7 connections)
             /velodyne_points                         594 msgs    : sensor_msgs/PointCloud2    
             /zed/left/image/compressed               500 msgs    : sensor_msgs/CompressedImage
             /zed/right/image/compressed              401 msgs    : sensor_msgs/CompressedImage
             /zed/zed_node/depth/depth_registered     259 msgs    : sensor_msgs/Image          
             /zed/zed_node/left/camera_info          1403 msgs    : sensor_msgs/CameraInfo     
             /zed/zed_node/right/camera_info         1403 msgs    : sensor_msgs/CameraInfo
```

and now it is:

```bash
  agrob rosbag info miguel_bag.bag
path:        miguel_bag.bag
version:     2.0
duration:    29.6s
start:       Sep 15 2020 10:18:44.50 (1600161524.50)
end:         Sep 15 2020 10:19:14.05 (1600161554.05)
size:        1.2 GB
messages:    15664
compression: none [359/359 chunks]
types:       sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /camerat265/fisheye1/camera_info               887 msgs    : sensor_msgs/CameraInfo     
             /camerat265/fisheye1/image_raw/compressed      887 msgs    : sensor_msgs/CompressedImage
             /camerat265/fisheye2/camera_info               887 msgs    : sensor_msgs/CameraInfo     
             /camerat265/fisheye2/image_raw/compressed      886 msgs    : sensor_msgs/CompressedImage
             /tf                                          10313 msgs    : tf2_msgs/TFMessage         
             /velodyne_points                               293 msgs    : sensor_msgs/PointCloud2    
             /zed/zed_node/left_raw/camera_info             390 msgs    : sensor_msgs/CameraInfo     
             /zed/zed_node/left_raw/image/compressed        330 msgs    : sensor_msgs/CompressedImage
             /zed/zed_node/point_cloud/cloud_registered      62 msgs    : sensor_msgs/PointCloud2    
             /zed/zed_node/right_raw/camera_info            390 msgs    : sensor_msgs/CameraInfo     
             /zed/zed_node/right_raw/image/compressed       339 msgs    : sensor_msgs/CompressedImage
```

You are now using the raw images for the zed ... was this intentional? What was the reason?

miguelriemoliveira, 2020-09-17:
The problem we have now is that we want some of the data in the bag file to be saved in synchronous fashion with our collections, to ensure a fair comparison later on. My suggestion is to add a functionality in which you can specify additional data to store for each collection.  That we we get a general solution to this particular problem, which we can ue for other stuff later on.

Here are the details:

1. Add to the yaml a new item called **additional_data**, in which you can create a list of topics to save when recording a collection.

Something like:

```yaml
(...)
additional_data:
    fisheye_camera:
         topic_name: "/camerat265/fisheye1/image_raw/compressed"

    zed_point_cloud:
         topic_name: "/zed/zed_node/point_cloud/cloud_registered"
(...)
```

2. Then, consider the json we have:

```json
{
  "collections": {
    "0": {
      "data": {
        "world_camera": ...
      },  
      "labels": ,
      "transforms": 
    },  
      },  
  "calibration_config": ...,
  "sensors": {
    "world_camera": ... 
  }
}
```

and the goal would be to add these topics in the collections dictionary, as follows

```json
{
  "collections": {
    "0": {
      "data": {
        "world_camera":  ...
      },  
      "labels": ...,
      "transforms" : ... 
      "additional_data" : {
               "fisheye_camera": (data obtained from the ros to json converted)
                                 {
                                      "data_file": "fisheye_camera_0.jpg",
                                      "encoding": "rgb8",
                                      "header": { ...
                                 }
               "zed_point_cloud": (data obtained from the ros to json converted)
    }
    },  
      },  
  "calibration_config": ...,
  "sensors": {
    "world_camera": ... 
  }
}
```

makes sense?

aaguiar96, 2020-09-17:
Yes @miguelriemoliveira 

Thanks! I'll do that. 👍

aaguiar96, 2020-09-21:
This is done.
Tested with ZED's pointcloud and RealSense's images, simultaneously.

In theory, the approach supports all types of ROS data.

Here is a screenshot of a test dataset:

[image]

aaguiar96, 2020-09-21:
I will close the issue. If for some reason we find any bug in the future, we open it again.

closed 2020-09-21 by aaguiar96

miguelriemoliveira, 2020-09-21:
That's good news!