lardemua/atom #235: Generate noisy initial guesses the original one

aaguiar96, 2020-09-28:
Generate new initial guesses from the one set in the input `json` file.

Do it as @miguelriemoliveira wrote:

> I would add an option on the loader to add the noise to the transformations (the question is to which of those).
> It is better than to have lots of datasets.
> 
> For example, suppose you want to add 1% noise to some transformation T.
> 
> You have the rotation angles (must convert from the quaternion) and the translation vector.
> 
> rx, ry, rz , tx, ty, tz
> 
> Just randomize a number between 0.99 and 1.01 (for 1%) and multiply by those 6 parameters.

aaguiar96, 2020-10-01:
Hi @miguelriemoliveira and @eupedrosa 

This is implemented!
The result is interesting. With 10% of noise, for example, the calibration goes starts from diverging for weird solutions, and then slowly converges to a solution similar to the ones we obtained before, with no noise!

aaguiar96, 2020-10-01:
I'll put these results on the paper, and all the atom results are generated.

eupedrosa, 2020-10-01:
> The result is interesting. With 10% of noise, for example, the calibration goes starts from diverging for weird solutions, and then slowly converges to a solution similar to the ones we obtained before, with no noise!

Do you have a video (or a gif) that show the effect?

aaguiar96, 2020-10-01:
> The result is interesting. With 10% of noise, for example, the calibration goes starts from diverging for weird solutions, and then slowly converges to a solution similar to the ones we obtained before, with no noise!

This was due to an error on the conversion from quaternion-to-euler. Now the behavior is standard.

Here's an example of a starting point from a 10% noise addition.

[image]

eupedrosa, 2020-10-01:
And what about 50%, 70%, 100%, 200%, etc..

aaguiar96, 2020-10-01:
> And what about 50%, 70%, 100%, 200%, etc..

Here's an example with 50%.
The result is not so good, but I think it's quite acceptable for this amount of noise.

[](https://youtu.be/-G_qDZjmeqU)

miguelriemoliveira, 2020-10-01:
Hi, good news, and great results.

Just a comment: because we are talking about angles I think we should go at most to 20-25% of noise.

I was actually thinking of something like 0.1%, 1%, 5%, 10%. A 10% error for an angle is a lot.

Great work, congratulations.

aaguiar96, 2020-10-01:
I just finished generating the papers results for atom.
I considered 5%, 10%, and 50%. Even with 50%, the results are great... Do you think I should change?

I tested with 100% and the calibration was not converging for the right spot, so I stopped at 50% eheh!

eupedrosa, 2020-10-01:
How many times did you obtained the results for each % level ? Remember, if it is random, just doing an experiments is not enough.

aaguiar96, 2020-10-01:
I opted for not doing it randomly. If the user chooses 1% error, I add 1% error to each component of the transform to be calibrated.

eupedrosa, 2020-10-01:
That is not the definition of noise. It is just adding offsets to the initial guess. And you can be adding bias to that offset...
Using random is more work I know. If you want I can also run experiments.

@miguelriemoliveira, what is your opinion?

eupedrosa, 2020-10-01:
By the way, @aaguiar96, you pushed an empty script in 1ac7ba7e54f7b9d227314b17d6ba2a3e83ca1902, see `convert_pcl.py`.

aaguiar96, 2020-10-01:
Yes, but it has the same effect using our initial guess setter toolbox, right? If I use this tool, I can add this offset to the initial guess, and end up with a worst initial guess right?

But I understand what your saying. I can re-do this if you think it's appropriate.

aaguiar96, 2020-10-01:
> By the way, @aaguiar96, you pushed an empty script in [1ac7ba7](https://github.com/lardemua/atom/commit/1ac7ba7e54f7b9d227314b17d6ba2a3e83ca1902), see `convert_pcl.py`.

Sorry, I'll remove it.

miguelriemoliveira, 2020-10-01:
> That is not the definition of noise. It is just adding offsets to the initial guess. And you can be adding bias to that offset...
> Using random is more work I know. If you want I can also run experiments.
> 
> @miguelriemoliveira, what is your opinion?

I agree with @eupedrosa , if we call it random then it should not be a bias, and because its random we should run it 3 ou 4 times each configuration.

The easy way out is not to call it noise but just a "bad", "reasonable", "good" and "excelent " first guesses ...

aaguiar96, 2020-10-01:
> I agree with @eupedrosa , if we call it random then it should not be a bias, and because its random we should run it 3 ou 4 times each configuration.
> 
> The easy way out is not to call it noise but just a "bad", "reasonable", "good" and "excelent " first guesses ...

Ok thanks. I'll re-do the experiences so that we consider noise.

eupedrosa, 2020-10-01:
>Yes, but it has the same effect using our initial guess setter toolbox, right? If I use this tool, I can add this offset to the initial guess, and end up with a worst initial guess right?

No, it is more generic. The toolbox is an instantiation of possible errors.

> The easy way out is not to call it noise but just a "bad", "reasonable", "good" and "excelent " first guesses ...

True, but that is subjective to our appreciation. The random search is more elegant.

> .. we should run it 3 ou 4 times each configuration.

It has to be more. @aaguiar96, if you commit the script I can use our servers to do a battery of tests.

aaguiar96, 2020-10-01:
> It has to be more. @aaguiar96, if you commit the script I can use our servers to do a battery of tests.

Ok @eupedrosa, that is nice. The code is available now.
You just have to add the flag `-nig` with a float number representing the percentage of noise (e.g. 2.0 for 200%).

aaguiar96, 2020-10-01:
So, you will run the experiments for a given number of noise percentages?

eupedrosa, 2020-10-01:
> You just have to add the flag -nig with a float number representing the percentage of noise (e.g. 2.0 for 200%).

> So, you will run the experiments for a given number of noise percentages?

Yes and yes.

But there are a few things that needs to be changed from the script. For example:
https://github.com/lardemua/atom/blob/ffd6125ec495587962b72f2d32b0d163803c7aba/atom_core/src/atom_core/dataset_io.py#L669-L670

Here you are applying proportional error, which is not what we want. For example, angle zero will result in an angle with zero error.
Furthermore, you are also adding noise to the anchored sensor.

aaguiar96, 2020-10-01:
> Here you are applying proportional error, which is not what we want. For example, angle zero will result in an angle with zero error.

Ok, makes sense. What do you suggest? For angles the percentage can be in relation with 360º, and for translation in relation with 1 meter, for example.

>  Furthermore, you are also adding noise to the anchored sensor.

Right, I'll correct that.

eupedrosa, 2020-10-01:
> Ok, makes sense. What do you suggest? For angles the percentage can be in relation with 360º, and for translation in relation with 1 meter, for example.

What I suggest is  100% corresponds to the 1.0 of the measurement unit.
For example, for translation 75% corresponds to 0.75 * 1 meter, and for rotation corresponds to 0.75 x 1 radian.

aaguiar96, 2020-10-01:
> What I suggest is 100% corresponds to the 1.0 of the measurement unit.
> For example, for translation 75% corresponds to 0.75 * 1 meter, and for rotation corresponds to 0.75 x 1 radian.

Ok, that's what I suggested... :+1: 
I'll do it, and commit soon.

aaguiar96, 2020-10-01:
Done @eupedrosa!

eupedrosa, 2020-10-01:
Nice :+1: 

Now I need the datasets that you are considering for the tests. Can you send them to me?

aaguiar96, 2020-10-01:
Yes. You must run the calibrations in the `train_dataset`, and then the experiment scripts in the `test_dataset`.

Here:
* train: [https://drive.google.com/file/d/12I0T_Fb54HUPpQkNyNCzKuieilZl-BBs/view?usp=sharing](https://drive.google.com/file/d/12I0T_Fb54HUPpQkNyNCzKuieilZl-BBs/view?usp=sharing)
* test: [https://drive.google.com/file/d/1vAJ0OdoWc3FjKG_cy974k2znxd19T_HL/view?usp=sharing](https://drive.google.com/file/d/1vAJ0OdoWc3FjKG_cy974k2znxd19T_HL/view?usp=sharing)

To run the experiments:

* camera-to-camera:
```
rosrun atom_evaluation camera_to_camera_evalutation.py -train_json "/home/andreaguiar/Documents/datasets/train_dataset/atom_calibration.json" -test_json "/home/andreaguiar/Documents/datasets/test_dataset/data_collected.json" -ss "left_camera" -ts "right_camera" -si
```

* right_camera-velodyne
```
rosrun atom_evaluation range_sensor_to_camera_evaluation.py -train_json "/home/andreaguiar/Documents/datasets/train_dataset/atom_calibration.json" -test_json "/home/andreaguiar/Documents/datasets/test_dataset/atom_calibration.json" -ss "vlp16" -ts "right_camera" -ef "/home/andreaguiar/Documents/datasets/test_dataset/annotations_right.json
```

* left_camera-velodyne
```
rosrun atom_evaluation range_sensor_to_camera_evaluation.py -train_json "/home/andreaguiar/Documents/datasets/train_dataset/atom_calibration.json" -test_json "/home/andreaguiar/Documents/datasets/test_dataset/atom_calibration.json" -ss "vlp16" -ts "left_camera" -ef "/home/andreaguiar/Documents/datasets/test_dataset/annotations_left.json
```

aaguiar96, 2020-10-01:
Thanks for the help!

eupedrosa, 2020-10-02:
@aaguiar96, I am having some trouble with your datasets. The errors are to high from the expected.

aaguiar96, 2020-10-02:
That is strange... I'm having good results with the datasets.

Do you want to meet after lunch?

eupedrosa, 2020-10-02:
Ok.. We can meet after launch

aaguiar96, 2020-10-02:
Ok, 14h? Zoom?

eupedrosa, 2020-10-02:
Ok

aaguiar96, 2020-10-08:
Hi @eupedrosa, any news?

eupedrosa, 2020-10-08:
It is delayed due to some unforseen circumstances. But the experiments are running now. Tomorrow we can share results as planned.

aaguiar96, 2020-10-08:
Really nice @eupedrosa!

I'll send an email now to talk about tomorrow's meeting.

eupedrosa, 2020-10-12:
I have some results about the first guess *noise*.

This one is for distance:
[image]

And this one is for angle
[image]

What are your opinions?

aaguiar96, 2020-10-12:
Seems really great, and the results are expected, right?

The calibration is much more sensitive to orientation errors on first guess, the execution time increases for bad first guesses, and ATOM can deal with realistic first guesses.

I think we got all we need, right? Or are you thinking to generate more results?

eupedrosa, 2020-10-12:
Yes, the values are expected.
I only run 10 trial for each step in angle/distance. But it may be enough, but I am doing a run with 100 trials for each step, just in case. I was to do 1000 trials but it would take forever....

aaguiar96, 2020-10-12:
> Yes, the values are expected.
> I only run 10 trial for each step in angle/distance. But it may be enough, but I am doing a run with 100 trials for each step, just in case. I was to do 1000 trials but it would take forever....

Ok, when you have your final results, just add the pictures to the paper please!

miguelriemoliveira, 2020-10-12:
It looks great. And the results are the expected which is also great... sometimes I wonder why we take so much time to get results we were already expecting : - )

eupedrosa, 2020-10-12:
https://www.youtube.com/watch?v=v02L4xE_smk
[image]

miguelriemoliveira, 2020-10-13:
: - )

aaguiar96, 2020-10-19:
Hi @eupedrosa, do you have the results? Can you send me the images, or put them on the paper please?

Also, when you have some time, can you write section 2.3? Its just a paragraph on residuals normalization.

eupedrosa, 2020-10-19:
I will put the results' graphics in Overleaf.

I am going to read the full paper today and *maybe* :wink:  I will write something in section 2.3.

aaguiar96, 2020-10-19:
Thanks! I don't like that *maybe*, but thanks! :rofl:

aaguiar96, 2021-03-05:
This can be closed.

closed 2021-03-05 by aaguiar96