lardemua/atom #236: Get atomic tranformation from sensor-to-sensor calibration

aaguiar96, 2020-09-30:
Hi @miguelriemoliveira and @eupedrosa 

I was thinking on this problem, and this is what we have:

[image]

So, if we anchor the velodyne, we want to change `zed_camera_center-zed_camera_left_frame` atomic transformation, according to the ICP calibration `tf` (or opencv/ZED factory in the other case).

I don't know if I done it well, but my idea was to:

* Compute the original `base_link-zed_camera_center` (`TF1`) chain of transformations
* Compute a new chain: `base_link-zed_left_camera_frame` from `base_link-vlp16 x tf` (`TF2`).
* So, now the atomic transformation should be `TF2 x inv(TF1)`.

What am I doing wrong? Eheh :smile:

miguelriemoliveira, 2020-09-30:
Hi @aaguiar96 ,

can you zoom now?

aaguiar96, 2020-09-30:
Yes!

eupedrosa, 2020-09-30:
Zoom?

eupedrosa, 2020-09-30:
I am alone in the Zoom ... :disappointed:

aaguiar96, 2020-09-30:
Ok guys, here's my plan:

* Implement script to convert pointcloud from a frame into another, and save it as a pcd
* Compute ICP between both pcls for each collection and save the result in a file
* Implement script that reads each result and re-writes the transformation in the json.

Btw, @miguelriemoliveira and @eupedrosa, I checked and the ZED's pointcloud is in `zed_left_camera_frame`.

I'll start working on this now. I'll ask you too check the last step, as soon as I have results.
Thanks!

miguelriemoliveira, 2020-09-30:
Hi @aaguiar96 ,

a couple of suggestions:

> Implement script to convert pointcloud from a frame into another, and save it as a pcd
> Compute ICP between both pcls for each collection and save the result in a file
> Implement script that reads each result and re-writes the transformation in the json.

I think these should be in a single script. Remember you must have all transformations in order to manipulate them as we discussed this morning. If that's in separate scripts it is more difficult.

> For now, I'll put these scripts in atom_evaluation/scripts/others, since they are not as generic as the ones implemented before in atom_evaluation. We can refine them later.

Ok, great.

> Btw, @miguelriemoliveira and @eupedrosa, I checked and the ZED's pointcloud is in zed_left_camera_frame.

That's really strange. I am almost sure this is a problem. One easy to do trick that I often use is to check the Z coordinate of all points. Do you have in your point cloud any point with Z<0, or are they are Z>0?

>

aaguiar96, 2020-10-02:
Hi @miguelriemoliveira and @eupedrosa 

I think I was able to implement a generic function to do this.
I created a utilities script in atom_evaluation for these functions.

Here's the idea (that is entirely based on our conversation).
Given this generic configuration graph:

[image]

To recover `T1`, we just need to apply the following equation:

[image]

This can easily be done in a function like this:

```
def atomicTfFromCalibration(dataset, anchored_sensor_key, other_sensor_key, calib_tf):
    # ----------------------------------------------------------------------------------------------------------------
    # ------ Get input data: required links and transformation
    # ----------------------------------------------------------------------------------------------------------------
    world_link = dataset['calibration_config']['world_link']
    anchored_sensor_link = dataset['calibration_config']['sensors'][anchored_sensor_key]['link']
    other_sensor_link = dataset['calibration_config']['sensors'][other_sensor_key]['link']
    other_sensor_child_link = dataset['calibration_config']['sensors'][other_sensor_key]['child_link']
    other_sensor_parent_link = dataset['calibration_config']['sensors'][other_sensor_key]['parent_link']

    selected_collection_key = dataset['collections'].keys()[0]
    base2anchored = getTransform(world_link, anchored_sensor_link,
                                 dataset['collections'][selected_collection_key]['transforms'])
    base2other_parent = getTransform(world_link, other_sensor_parent_link,
                                     dataset['collections'][selected_collection_key]['transforms'])
    other_child2other = getTransform(other_sensor_child_link, other_sensor_link,
                                     dataset['collections'][selected_collection_key]['transforms'])

    Ta = np.dot(np.linalg.inv(base2other_parent), base2anchored)
    Tb = np.dot(Ta, calib_tf)
    Tc = np.dot(Tb, np.linalg.inv(other_child2other))

    return Tc
```

To test this function, I used a simple approach. Pass to the function the original sensor-to-sensor calibration present in the json.
So, it is expected that the function return a transformation `T1` also equal to the one present in the json.

Here's the result:

* Original `T1`

```
[[1.   0.   0.   0.  ]
 [0.   1.   0.   0.06]
 [0.   0.   1.   0.  ]
 [0.   0.   0.   1.  ]]
```

* Obtained `T1`

```
[[ 1.00000000e+00  3.27502792e-20  9.17832980e-18  5.55150813e-17]
 [-1.59700041e-21  1.00000000e+00 -2.91556511e-21  6.00000000e-02]
 [ 2.01850775e-17  4.95282169e-21  1.00000000e+00  5.94338603e-22]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
```

So, I think it is working, and it is generic. So we can use it for any case!

miguelriemoliveira, 2020-10-02:
Hi @aaguiar96 ,

looks really nice. I just have a suggestion (I always do have at least one suggestion, because that's what boring people have :- )

The getTransform function is really flexible, and I think that instead of doing:

```
base2anchored = getTransform(world_link, anchored_sensor_link,
                                 dataset['collections'][selected_collection_key]['transforms'])
base2other_parent = getTransform(world_link, other_sensor_parent_link,
                                     dataset['collections'][selected_collection_key]['transforms'])
Ta = np.dot(np.linalg.inv(base2other_parent), base2anchored)
```

which is the transform from **anchored_sensor_link** to **other_sensor_parent_link** you can simplify by just calling

```
Ta = getTransform(anchored_sensor_link, other_sensor_parent_link,
                                     dataset['collections'][selected_collection_key]['transforms'])
```

You can even compute both ways to check its the same (it should be).

Perhaps similar shortcuts can be made of the rest of the code, not sure.  The takeaway here is that you can rely of the getTransform to do all the np.dots inside the function (provided the transformations exist in the transformation pool).

Also, I think you can just leave it as is. It works! It was just to be boring :- )

aaguiar96, 2020-10-05:
Hi @miguelriemoliveira 

Thanks for the tips! I'll do that, as soon as I have some time.
I'll close this issue since it's solved!

closed 2020-10-05 by aaguiar96