lardemua/atom #237: Colorize point clouds on atom_evaluation package

aaguiar96, 2020-10-12:


aaguiar96, 2020-10-13:
I @miguelriemoliveira and @eupedrosa 

I created the script to colorize the point cloud.
The result is a bit strange.

I checked the projection of the point into the image and they are correct.
However, the colorizing process does not seem ok... The points behind the velodyne are also mapped to the image.

Here:

[image]

If you want to try:
```
rosrun atom_evaluation colorize_pointcloud.py -json /home/andreaguiar/Documents/datasets/train_dataset/atom_calibration_b.json -ls vlp16 -cs left_camera
```

miguelriemoliveira, 2020-10-13:
Hi André,

I do not have time to try right now (only during the weekend), but I
suspect this has something to do with you not checking whether the z
coordinate of the point (when represented in the camera's reference frame)
is > 0. Only those should be colored.

Take a look at this paper

http://www.cvc.uab.es/~asappa/publications/J__Elsevier_IF_Vol_24_July_2015_pp_108-121.pdf

fig 4, I think that could be your problem.

Regards,
Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

aaguiar96, 2020-10-14:
Ok @miguelriemoliveira thanks!

@eupedrosa  can you try it and see the result? I don't find it very appealing visually...

miguelriemoliveira, 2020-10-14:
I will try to test it saturday. Sorry for the delay, I have 23 hours of classes this week : - (

aaguiar96, 2020-10-14:
Hi @miguelriemoliveira! No problem, we have time. I'll continue writing today!

eupedrosa, 2020-10-14:
> @eupedrosa can you try it and see the result? I don't find it very appealing visually...

And I think it will never be. The velodyne has a considerable gap between horizontal scans and that makes it look bad.
Did you already try @miguelriemoliveira suggestion?

aaguiar96, 2020-10-14:
> And I think it will never be. The velodyne has a considerable gap between horizontal scans and that makes it look bad.

So, but is this still interesting for the paper?...

>  Did you already try @miguelriemoliveira suggestion?

Not yet.

eupedrosa, 2020-10-14:
It is **NOT** interesting for the paper.

aaguiar96, 2020-10-14:
Ok, so I'll close the issue. I was just working on this because we agreed in the previous meeting that we'll put these colored point clouds in the paper...

aaguiar96, 2020-10-14:
Can I close it?

eupedrosa, 2020-10-14:
If it is hard to get any kind of information from the colorization, then it is not worth the time. So yes, close it.

miguelriemoliveira, 2020-10-14:
Hi guys, I am late as always these past weeks : - (.

One suggestion is to try it the other way arround. Project the lidar points onto the image and show the points on the image.

Take a look at this image

http://www-personal.acfr.usyd.edu.au/akas9185/AutoCalib/AutoLaserCamDoc/index.html

aaguiar96, 2020-10-15:
Hi @miguelriemoliveira, thanks for the hint.
I substituted the colorizing script by a generic projection script in `atom_evaluation`.

Usage:

```
rosrun atom_evaluation point_cloud_to_image.py -train_json /home/andreaguiar/Documents/datasets/train_dataset/atom_calibration_b.json -test_json /home/andreaguiar/Documents/datasets/test_dataset/data_collected.json -ls vlp16 -cs left_camera -si
```

Result (the colormap represents depth):

[image]

These ones are great for the paper right?

miguelriemoliveira, 2020-10-15:
Hi @aaguiar96 ,

yes, the picture looks nice. We can see changes in color (abrupt changes in the measured depth) in the lidar jump edges.
Thats a great image for the paper.

I think the color of hte lidar points could be improved ... which colormap are you using?

eupedrosa, 2020-10-15:
I think it can be improved:
* Use filled circles with anti-aliasing.
* The colour mapping needs more resolution.
* Maybe crop the image bottom, it has no information.

aaguiar96, 2020-10-15:
I am using this one:

```
    if show_images:
        colors = cm.tab20b(
            np.linspace(0, 1, 20 * 4))  # we consider a maximum of 20 meters with a discretization of 0.25 meters
```

eupedrosa, 2020-10-15:
You should find the minimum and maximum depth that are visible in the image. Then use that range to improve the colour gamut.

aaguiar96, 2020-10-15:
> You should find the minimum and maximum depth that are visible in the image. Then use that range to improve the colour gamut.

Ok, good idea. I can add also more resolution.

miguelriemoliveira, 2020-10-15:
Hi @aaguiar96 ,

another thing, for this case in particular we should use a progressive colormap, meaning that the color will change progressively.
That way, large jumps in color will denote large jumps in depth.

tab20b is not progressive, take a look:
https://stackoverflow.com/questions/43938425/matplotlib-change-colormap-tab20-to-have-three-colors

all of the first 14 colormaps here are progressive, you can use any one of these:

https://www.mathworks.com/help/matlab/ref/colormap.html