lardemua/atom #240: How do I simulate the ur10e in Gazebo?

miguelriemoliveira, 2021-02-07:
Hi @eupedrosa ,

I need this for creating a gazebo world for the paper. 

Also, @danifpdra  should be using this for the intelligent cell in the Bosch project.

Can you tell me how you do it? I am using your iris_ur10e repo ...

Thanks in advance.

eupedrosa, 2021-02-08:
There is a launch file for that. If you execute `rosrlaunch iris_ur10e sim.launch` a simulation is launched.
You can then check the launch file and create you our simulation environment.

miguelriemoliveira, 2021-02-08:
Thanks @eupedrosa ,

I have tried it. It is working with rviz only. Did you integrate gazebo in this or not?

Thanks in advance

eupedrosa, 2021-02-08:
The Gazebo GUI is disabled, nonetheless the simulation is working. Try `roslaunch iris_ur10e sim.launch gui:=true`

miguelriemoliveira, 2021-02-08:
Thanks. It does not work but I think it is related to me using ros noetic. I will try in melodic and get back to you.

miguelriemoliveira, 2021-02-08:
It works. I was missing the ros-noetic-joint-trajectory-controller. Thanks for the help!

closed 2021-02-08 by miguelriemoliveira