lardemua/atom #241: Setting up mmbot for using as case study in the paper

miguelriemoliveira, 2021-02-09:
Hi @eupedrosa and @aaguiar96 ,

Just prepared a robotic system to use as case study

[image]

the tf tree

[mmt.pdf](https://github.com/lardemua/atom/files/5953582/mmt.pdf)

Any comments are welcome.

aaguiar96, 2021-02-09:
Hi @miguelriemoliveira, wow!!

Looks really nice. Will you test ATOM in this setup and include the results on the paper?
What type of calibration will you make? This can mix the hand-eye and LiDAR calibration scenarios, right?

Do you have a picture of the real setup? I'm curious. :-)

miguelriemoliveira, 2021-02-09:
Hi André,

it does not exist. It is just a conceptual robotic system. If all goes well
we will calibrate it using simulated data.

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2021-02-11:
Hi guys,

I have the simulation almost complete. It is nice because we can get sensor data and we can also move the manipulator as well as the pattern. 

I think we can extract a nice "virtual" bag file which we can use to test the calibration. And the good thing is that we will have ground truth (because we are simulating everything)!

Here's a video:

https://youtu.be/e8wEC8DjxDc

aaguiar96, 2021-02-11:
Hi @miguelriemoliveira,

Looks really really nice!
Have you checked if you can detect the pattern corners on the simulated image?

miguelriemoliveira, 2021-02-11:
Thanks.

Not yet. I will record a bag file soon, and then i will test.

Any ideas are welcome.

aaguiar96, 2021-02-11:
> Not yet. I will record a bag file soon, and then i will test.

Ok @miguelriemoliveira, nice!
This seems important, since from the image I could not see the pattern squares with clarity.
Have you considered making the scene white to increase contrast?

miguelriemoliveira, 2021-02-11:
It is the light of the scene. I will add more lights and it will be better.
I will post an image soon.

eupedrosa, 2021-02-11:
Looks great... But do you know what would be awsome??? Two robotic arms :smile: , so we could also calibrate the transformation between the arms.

miguelriemoliveira, 2021-02-11:
We can do that. I think Atom could work with that.

But the example for the paper would turn out very large graphs.

I will create this scenario and then we decide.

Thanks for the feedback.
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2021-02-14:
Hi @aaguiar96 and @eupedrosa,

https://youtu.be/wRW2SyDoLp8

miguelriemoliveira, 2021-02-15:
I have a first bag file collected from the mmtbot.

https://drive.google.com/file/d/1VgsDDFW9gK3ZyEv6bZObFxYka5MChXaW/view?usp=sharing

eupedrosa, 2021-02-15:
Cool :+1:

The current version of ATOM can calibrate this system, right?

miguelriemoliveira, 2021-02-17:
Hi @eupedrosa and @aaguiar96 ,

So we have a new bag file:

https://drive.google.com/file/d/1Q-aRLCZhoJ7xhopbsg9nvW6XJcbDzEOs/view?usp=sharing

And I created a test dataset from it (just a couple of collections):

https://drive.google.com/file/d/1byrQmiy_8iJdLi3o7sMPsxITUOGnlKkm/view?usp=sharing

All seems to be fine ... which is nice since I had to change a lot of code because of #242 .

miguelriemoliveira, 2021-02-17:
Next is trying to run a simple calibration procedure...

miguelriemoliveira, 2021-02-17:
> The current version of ATOM can calibrate this system, right?

We are all convinced it can ... gulp : )

aaguiar96, 2021-02-23:
Hi @miguelriemoliveira 

Can you send me the lataste bagfile of mmbot?

miguelriemoliveira, 2021-02-23:
Here it goes:

https://drive.google.com/file/d/1mnzHkKeuQmjPCMyq_iGF67GByIx4sQk-/view?usp=sharing

and the dataset

https://drive.google.com/file/d/1Vr_mSSZU24fMTfJdCmcwA9dumiedEbl2/view?usp=sharing

miguelriemoliveira, 2021-02-23:
Just updated the readme with more info.

Anything else let me know with an issue.

aaguiar96, 2021-02-23:
Hi @miguelriemoliveira  and @eupedrosa 

I'm trying to set up mmbot.
I'm stuck in this issue:

```
Traceback (most recent call last):
  File "/home/andresaguiar/catkin_ws_ros1/src/atom_pkgs/iris_ur10e/ur_e_gazebo/scripts/fake_wr-cr200-85.py", line 9, in <module>
    from SocketServer import ThreadingMixIn
ModuleNotFoundError: No module named 'SocketServer'
[fake_gripper-9] process has died [pid 230109, exit code 1, cmd /home/andresaguiar/catkin_ws_ros1/src/atom_pkgs/iris_ur10e/ur_e_gazebo/scripts/fake_wr-cr200-85.py __name:=fake_gripper __log:=/home/andresag
uiar/.ros/log/34aa00ba-75f9-11eb-9d52-db1ca39a088a/fake_gripper-9.log].
log file: /home/andresaguiar/.ros/log/34aa00ba-75f9-11eb-9d52-db1ca39a088a/fake_gripper-9*.log
```

This a python3 issue, since in python3 the module name should be `socketserver`.
@miguelriemoliveira how does this work for you?

eupedrosa, 2021-02-23:
@aaguiar96, that script is a fake controller for the gripper, you can disable it.
Search for the file `ur10e.launch` in the directory `ur_e_gazebo/launch` and comment the line that launches the `fake_gripper`.

miguelriemoliveira, 2021-02-23:
Hi @aaguiar96 ,

this happens when you run which command?

aaguiar96, 2021-02-23:
Hi @eupedrosa, thanks.
I had to disable it in mmbot launch file (`mmtbot/mmtbot_gazebo/launch/mmtbot.launch`).

@miguelriemoliveira, this one:
```
roslaunch mmtbot_bringup sim.launch
```

miguelriemoliveira, 2021-02-24:
Hi,

good that its solved. But it would be best, if possible, to understand why it works for me and not for you ...

Let's discuss this afternoon.

miguelriemoliveira, 2021-02-24:
Hi @aaguiar96 , 

just talked to @eupedrosa about how to manage the two repos we need, ur10e and ur10e_calibration, and the solution was to create a noetic-devel branch in both. 

You should use these branches instead. Let me know if it runs or not ...

closed 2021-02-27 by miguelriemoliveira