lardemua/atom #242: Branch Ros Noetic

miguelriemoliveira, 2021-02-15:
Hi @eupedrosa and @aaguiar96 ,

I opened this branch since now I am working on Ubuntu 20.04.

Will commit changes to this branch and then we decide when to move the master branch to noetic (and python3).

miguelriemoliveira, 2021-02-17:
Hi @eupedrosa and @aaguiar96 ,

What is your opinion about moving towards ros noetic / Ubuntu 20.04.

I have made ATOM work up until the collecting of a dataset. Had to change a lot of things, but in most of them I think 
the change was more or less obvious.

If you can please revise the commits in the noetic branch to see if it makes sense for you.

https://github.com/lardemua/atom/tree/ros_noetic

I will continue with trying to make the calibration work (didn't try it yet).

aaguiar96, 2021-02-17:
> What is your opinion about moving towards ros noetic / Ubuntu 20.04.

I @miguelriemoliveira, I also started using Ubuntu 20.04 and ROS noetic. Even so, I think the master should have the version most people still use. I think for now is melodic right? (not sure)

I also started working on ROS2, maybe in the future, we can support it too.

miguelriemoliveira, 2021-02-17:
Hi @aaguiar96 ,

not sure what is the most used ros distro.

We had it working on melodic before, my suggestion is to create a melodic branch and leave it in the current state, and migrate the master branch to the noetic.

@eupedrosa are you ok with this? Does it make sense?

About ROS2, let's see how it goes but for now I would say definitely not a priority.

eupedrosa, 2021-02-17:
In the ROS community, the most common thing is to have a branch for each ROS version. For example:
* `kinetic-devel` for kinetic
* `melodic-devel` for melodic
* `noetic-devel` for noetic

Then, in GitHub repo options, set the default branch to whatever you want. If your main target is melodic just use that branch as default.

What I do with [iris_lama](https://github.com/iris-ua/iris_lama), is to have a `master` that works in all three main ROS versions: kinetic, melodic and noetic. I then have the specific branches just for releasing the package.
However, because we use `python` this my approach is more difficult, mainly because noetic uses python3.

To reach more people, we should maintain versions for **melodic** and **noetic**. And the best course of action is to "abandon" the `master` branch in favor of the `melodic-devel` and `noetic-devel`.

> About ROS2, let's see how it goes but for now I would say definitely not a priority.

Definitely not a priority

miguelriemoliveira, 2021-02-17:
Sounds good to me as @eupedrosa suggests. 

But in practice the melodic part should not be touched by our research group.
I mean, if some outsider raises an issue about melodic that's ok, but we should be coordinated in using all the same distro.
So the question is: @eupedrosa , can you also jump to noetic and we advance the development there?

eupedrosa, 2021-02-17:
I do not know if it is a good idea to "neglet" the **melodic** version. Along with **kinetic**, they are the most used versions.
I am willing to maintain the **melodic** version.

miguelriemoliveira, 2021-02-17:
That's great if you can take the melodic branch.

Notice, what I meant is when we develop a new functionality are you going to translate it to melodic? I think this is very time consuming, but if you can great.

miguelriemoliveira, 2021-02-17:
Did the changes we discussed.

Two branches 

   - noetic-devel (default)
   - melodic-devel (old master renamed)

Github says we must run this locally to update properly.

git branch -m noetic-devel <BRANCH>
git fetch origin
git branch -u origin/<BRANCH> <BRANCH>

eupedrosa, 2021-02-18:
> Notice, what I meant is when we develop a new functionality are you going to translate it to melodic? I think this is very time consuming, but if you can great.

That is exactly what I was saying

closed 2021-02-27 by miguelriemoliveira