lardemua/atom #245: Charuco detection in simulated images is failing

miguelriemoliveira, 2021-02-18:
Hi @eupedrosa and @aaguiar96 ,

Can you tell me if this config for this pattern makes sense?

[image]

I copied the configuration from agrob (which uses the same pattern) as follows:

```yaml
# The calibration requires a detectable pattern.
# This section describes the properties of the calibration pattern used in th calibration.
calibration_pattern:

    # The frame id (or link) of the pattern.
    # This link/transformation will be optimized.
    link: "pattern_link"

    # The parent frame id (or link) of the pattern.
    # For example, in hand-eye calibration the parent link
    # of the pattern can be the end-effector or the base of the arm
    parent_link: "world"

    # Defines if the pattern link is the same in all collections (i.e. fixed=true),
    # or each collection will have its own estimative of the link transformation.
    fixed: false

    # The type of pattern used for the calibration.
    # Supported pattern are:
    # - chessboard
    # - charuco
    pattern_type: "charuco"

    # If the pattern type is "charuco" you need to define
    # the aruco dictionary used by the pattern.
    # See https://docs.opencv.org/trunk/dc/df7/dictionary_8hpp.html
    dictionary: "DICT_5X5_100"

    # Mesh file (collada.dae or stl) for showing pattern on rviz. URI or regular path.
    # See: description_file
    mesh_file: "package://atom_calibration/meshes/charuco_5x5.dae"

    # The border width from the edge corner to the pattern physical edge.
    # Used for 3D sensors and lidars.
    # It can be a scalar (same border in x and y directions), or it can be {'x': ..., 'y': ,,,}
    border_size: {'x': 0.04, 'y': 0.03}

    # The number of corners the pattern has in the X and Y dimensions.
    # Note: The charuco detector uses the number of squares per dimension in its detector.
    # Internally we add a +1 to Y and X dimensions to account for that.
    # Therfore, the number of corners should be used even for the charuco pattern.
    dimension: {"x": 11, "y": 8}

    # The length of the square edge.
    size: 0.06

    # The length of the charuco inner marker.
    inner_size: 0.045
```

But I am not being able to detect the pattern ... 

One I have time I will dig into this, but in the meantime if you have any ideas ...

eupedrosa, 2021-02-18:
I believe the dictionary is not the same, @aaguiar96 what dictionary you use to generate the image?

aaguiar96, 2021-02-18:
In agrob I used 5X5... Here: https://github.com/aaguiar96/agrob/blob/master/agrob_calibration/calibration/config.yml

eupedrosa, 2021-02-18:
That is for the real pattern. I was asking for the 3D model.

aaguiar96, 2021-02-19:
Sorry about the late response guys... 
You mean the colada file right?

I looked for the respective commit and I founded this: https://github.com/lardemua/atom/commit/07955dede712c19bf999bf3134ec094670418757#diff-311fe3abefb2e039187006d73a477430f29daa264e167e2a1ed78b625a1a3ec5

So I think that it is the dictionary `charuco_5X5_800x600.dae`.

miguelriemoliveira, 2021-02-19:
Thanks @aaguiar96 ,

that's what I was using and I can't seem to be able to detect anything ... I will try again later ...

miguelriemoliveira, 2021-02-20:
Hi guys,

so the dict is what you said: DICT_5X5_100

the problem is I can only detect when very close...

[image]

aaguiar96, 2021-02-20:
Hi @miguelriemoliveira 

It's a start! One problem that might be happening is the image contrast... The supposed white pixels of the pattern do not seem very white. So the detector may be struggling to find the corners.

miguelriemoliveira, 2021-02-20:
Hi @aaguiar96 ,

its the size of the pattern in the image. Its too small. Gazebo is producing 640x480 images and in agrob you had 1280x720.

I will try to increase the image resolution in gazebo ...

closed 2021-02-21 by miguelriemoliveira