lardemua/atom #246: Read and write PCDs in python3

miguelriemoliveira, 2021-02-21:
Hi @aaguiar96 ,

the code of reading and writing pcds is one of the codes which has been the hardest to convert to python3.

I had a log of trouble with the write pcd code but eventually was able to make it work by forcing that the 
output file was written in ascii.
 
Check here what I changed.

https://github.com/lardemua/atom/compare/melodic-devel...noetic-devel#diff-7afaa9ac850ab3f038b763d652e44eca06f11b352f4995d94e969e8537e318c1

Now the problem is that when reading the pcd files (for the calibration part) I get to a point where I have the 
abort message "[read_pcd] Only binary .pcd files are readable."

https://github.com/lardemua/atom/blob/ae85b9fd6902cee46f23c90c1de29207924522d8/atom_core/src/atom_core/dataset_io.py#L471-L472

Can you help? What we need is a pcd read/write functionality that works for python3.

Perhaps I did something wrong when trying to adapt  the write pcd.I remember the problem was that in python3 
the function 

open

can only open files for writing in ascii or in binary, not both. Ans our pcds have an ascii header and binary data ... We can discuss more ...

So I guess

miguelriemoliveira, 2021-02-21:
Can we try to use something like this?

https://github.com/dimatura/pypcd

aaguiar96, 2021-02-21:
I @miguelriemoliveira 

I'll look at it this week. Maybe we can try a new library as you suggested.

miguelriemoliveira, 2021-02-21:
Hi @aaguiar96 ,

sorry, had a little time and could not resist working on this ... if you had the bureaucracy that I have to do you would understand : ).

So I found out that pypcd for python3 is much more advanced in a specific fork 

https://github.com/klintan/pypcd

I installed (the ros way) and did a simple python test which is working.

https://github.com/lardemua/atom/blob/noetic-devel/atom_calibration/scripts/deprecated/test_pcd_io.py

aaguiar96, 2021-03-01:
Hi @miguelriemoliveira  and @eupedrosa 

This is solved with 07312f2. 
Closing it.

closed 2021-03-01 by aaguiar96

miguelriemoliveira, 2021-03-01:
So you testes and it works fine? Writing and loading? Great!

miguelriemoliveira, 2021-03-01:
And the code is so much simpler : - )

eupedrosa, 2021-03-01:
@aaguiar96, I believe the authorship and licencse for `pcdpy.py` is missing. It is not completly ours, am I right?

eupedrosa, 2021-03-01:
@aaguiar96, I already add the licence to the file, so everything is good now. 
But we have to be mindfull with this kind of things.

aaguiar96, 2021-03-01:
Hi @eupedrosa 

I just extracted the absolutely necessary routines for read and write.
But yes, it makes sense anyway.

Thanks!

aaguiar96, 2021-03-01:
> So you testes and it works fine? Writing and loading? Great!

Yes @miguelriemoliveira 
I removed some old code that was useless on `dataset_io`.

And I think the added code does not have any external dependency. :)