lardemua/atom #248: mmtbot minimal test

miguelriemoliveira, 2021-02-24:
So the idea is to understand why we get a large error in the world camera and not in the hand camera.

The minimal test would be:

- only 2 cameras, no lidar
- single collection
- a single execution of the objective function
- do not touch the original simulation generated sensor poses, which would make the first guess a "perfect initial guess"

And the error should be close to 0 on both cameras in the first iteration of the objective function

miguelriemoliveira, 2021-02-24:
Collected a new bagfile

https://drive.google.com/file/d/14SAEUC2wQI3EnBJMfi3et7J3agPmaYw3/view?usp=sharing

and dataset

https://drive.google.com/file/d/14dl56sBWCCktH1ogeKqUtk364zK9Qrog/view?usp=sharing

using the constraints above. Manipulator does not move in the bag

miguelriemoliveira, 2021-02-24:
rosrun tf tf_echo world hand_camera_link

```
- Translation: [-0.117, 0.159, 1.361]
- Rotation: in Quaternion [-0.055, 0.094, 0.665, 0.739]
            in RPY (radian) [0.045, 0.214, 1.470]
            in RPY (degree) [2.553, 12.257, 84.237]
```
rosrun tf tf_echo world world_camera_link

```
- Translation: [1.051, 0.071, 1.345]
- Rotation: in Quaternion [-0.135, 0.078, 0.855, 0.494]
            in RPY (radian) [-0.000, 0.314, 2.094]
            in RPY (degree) [-0.000, 18.000, 120.000]
```

and the output of the first execution of the objective function is ...

```
world to hand_camera_link = [[ 0.09813254 -0.99300833  0.06560835 -0.11699567]
 [ 0.97226443  0.10972914  0.20654636  0.15856462]
 [-0.21230141  0.04351974  0.97623468  1.36080263]
 [ 0.          0.          0.          1.        ]]
quat = [-0.05515598  0.09402382  0.66490093  0.73893443]
world to world_camera_link = [[-0.47552826 -0.8660254  -0.1545085   1.05129503]
 [ 0.8236391  -0.5         0.26761657  0.07115441]
 [-0.30901699  0.          0.95105652  1.34521843]
 [ 0.          0.          0.          1.        ]]
quat = [-0.13547622  0.07821723  0.85536319  0.49384417]
Errors per sensor:
  hand_camera 0.09429266284947162
  world_camera 3.4105508969868006
Collection 0 hand_camera has 0.09429266284947162
Collection 0 world_camera has 3.4105508969868006
```

Conclusions:

1. In this new dataset the world to hand_camera transform is the same using rosrun tf tf_echo and in the objective function (because the manipulator did not move) and it is the same also for the world camera. This was expected.
2. The world_camera continues to have a much larger error than the hand camera (this dataset 3.4 to 0,9 pixels of error) *which we cannot adequately explain and which may hide some horrific bug**

@eupedrosa and @aaguiar96 any ideas?

eupedrosa, 2021-02-24:
I do not have any idea yet. But I tried to optimize an old dataset and it worked :partying_face:
However, there are some datasets that fail to load the pcds, @miguelriemoliveira, whas this the problem you ask @aaguiar96  to solve?

Can it be the size of the virtual charuco? In the configuration file we use a value of 6cm, is it correct?

miguelriemoliveira, 2021-02-24:
Hi @eupedrosa and @aaguiar96,

got it! The problem is that, for each collection we have to estimate the initial pattern pose i.e., world_T_pattern.

We do this by:

world_T_pattern = world_T_sensor x sensor_T_pattern

where we get sensor_T_pattern throught opencv's pnp and world_T_sensor. Note that this is done once for each collection, but once we find a sensor (camera) that allows us to do this we use that estimate and do not compute for other sensors.

That's why the hand_camera was always the best, because it was the one being used for computing this.

Check here https://github.com/lardemua/atom/blob/c565acd16491d82c7231634fe36f0f2b26cbe374/atom_calibration/src/atom_calibration/calibration/patterns_config.py#L169-L220

Once I change the order of the sensor loop 

https://github.com/lardemua/atom/blob/c565acd16491d82c7231634fe36f0f2b26cbe374/atom_calibration/src/atom_calibration/calibration/patterns_config.py#L175

and the world camera is the one being used to get the pattern pose the result is the oposite, world_camera very good, hand_camera not so good.

```
Errors per sensor:
  hand_camera 6.044381191788381
  world_camera 0.10960033099858442
Collection 0 hand_camera has 6.044381191788381
Collection 0 world_camera has 0.10960033099858442
```

So I guess we were the whole afternoon worried about a non-issue : - ). It is always good to warmup!

Let me know if you agree, if you do we can close this issue...

Or I could retrieve the real (simulated) pattern pose and use it just to see if we would really get error 0 ...

aaguiar96, 2021-02-24:
Hi @miguelriemoliveira 

Good news, nice catch!
You can try now a calibration with a different initial guess right? If it works, we are close to having results for the paper!

eupedrosa, 2021-02-24:
I also reached the same conclusion, but in my opinion it is not the end of it. 
The system corrects the error by changing the pose of the sensor, however, the sensor are in the correct position, this behavior should not happen. 

What should be changing are the estimates of the pattern, and this (change) is not happening. But why it is not changing?
In my opinion, it can be explain by a lack of diversity in the dataset.

miguelriemoliveira, 2021-02-24:
> You can try now a calibration with a different initial guess right? If it works, we are close to having results for the paper!

Yeah, lets advance. I think I will not resist to do the test I mentioned above, just to make sure I really trust the objective function.

> The system corrects the error by changing the pose of the sensor, however, the sensor are in the correct position, this behavior should not happen.
>
> What should be changing are the estimates of the [...]

What dataset are you using? Perhaps you could create a new issue and describe your test in more detail ...

miguelriemoliveira, 2021-02-24:
I will close this one.

closed 2021-02-24 by miguelriemoliveira