lardemua/atom #249: The strange case of charuco detection refinement

miguelriemoliveira, 2021-02-27:
Hi @eupedrosa ,

just found a curious thing while working on #230 .

When creating the dataset, in the data_collector code, we do the detection **and refinement** of the chessboards or charucos:

https://github.com/lardemua/atom/blob/041e771b385514d71fbb62f68db1e47e535e6d58/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py#L334-L361

This is done by calling a method **detect** either from the ChessboarPattern class

https://github.com/lardemua/atom/blob/041e771b385514d71fbb62f68db1e47e535e6d58/atom_calibration/src/atom_calibration/collect/patterns.py#L12

or the CharucoPattern class

https://github.com/lardemua/atom/blob/041e771b385514d71fbb62f68db1e47e535e6d58/atom_calibration/src/atom_calibration/collect/patterns.py#L83

So far so good. The strange this is that, in the calibrate function (which reads the dataset already with the detection produced by the code above), we have an additional refinement of the corners which seems to be designed only for chessboards but operates of both chessboard and charuco datasets!

https://github.com/lardemua/atom/blob/041e771b385514d71fbb62f68db1e47e535e6d58/atom_calibration/scripts/calibrate#L106-L131

cornerSubPix is not to be used by charuco is it?

So either:

1. we pass the code from calibrate into the collect and label or
2. we remove the refinement code from the calibrate.

I vote for 2. @eupedrosa what do you say?

eupedrosa, 2021-02-27:
The function exists because the older datasets do not had the *corner refinement* step. 
Now that all collected corners goes throuhg this step we can remove it from the calibrate code.

miguelriemoliveira, 2021-02-27:
OK, but does it make sense to refine corners in the case of the charuco? Because this is not being done at the data collector ...

eupedrosa, 2021-02-27:
If we are refining the corners in the calibration stage, than it makes sense to make it in the data collection stage.

However, I have one argument against doing the corner refinement in the data collection stage. I noticed that some times, the corner refinement fails and provides a wrong "solution". Specially if the search windows size is too large. I guess if we use a small window like the `(3,3)` we may not encounter this problem.

miguelriemoliveira, 2021-02-28:
OK, thanks.

commit 32b9e8c6 (2021-02-28): Close #249. Removed corner subpix from calibrate into the colllect_data
closed 2021-02-28 by miguelriemoliveira