lardemua/atom #251: Store point cloud idxs instead of point coordinates

miguelriemoliveira, 2021-02-27:
Hi @aaguiar96 because of #230 I was reviewing you code to label the 3d point cloud data which is currently in the calibrate script

https://github.com/lardemua/atom/blob/041e771b385514d71fbb62f68db1e47e535e6d58/atom_calibration/scripts/calibrate#L175-L251

The dictionary  you produce for each collection, which we call  **labels**,  is something like:

```json
{'detected': True, 'idxs': [5416, 5432, 5464, 5480, (...)], 'labelled_points': [{'x': 1.655419945716858, 'y': -0.6398109793663025, 'z': 0.09301099926233292, 'w': 1.0}, (...)], 'limit_points': [{'x': 1.796854019165039, 'y': -0.2608639895915985, 'z': 0.4865120053291321, 'w': 1.0}, (...)], 'middle_points': [{'x': 1.8025050163269043, 'y': -0.2493550032377243, 'z': 0.48757898807525635, 'w': 1.0}, (...)] }
```

my question is: why do you store the points? It would be best to store the indices of the points rather than the xyz right?
Because that way the point coordinates are in a single place (in the collection data dictionary), which makes the code more robust, and also because  its more compact to store.

If you agree I can make this change.

miguelriemoliveira, 2021-02-28:
Hi @aaguiar96 ,

I will proceed. I think it makes sense. If you don't agree for some reason I can't see right now we can always rollback.

aaguiar96, 2021-02-28:
Hi @miguelriemoliveira 

Yes I agree! We're doing that for the entrevista point cloud right? So it makes sense to maintain the same approach in these cases. :)

miguelriemoliveira, 2021-02-28:
Hi André.

Entrevista point cloud???

aaguiar96, 2021-02-28:
Hi @miguelriemoliveira 

I corrected it! :)

miguelriemoliveira, 2021-03-04:
I @aaguiar96 ,

its been really crazy lately. I woke up at 7h and started working on this today at 7h30 but of course my daughter decided to wake up at 8h : - ) 

I advanced a bit so let me try to finish tonight or tomorrow.

Will get back to you later.

aaguiar96, 2021-03-04:
Hi @miguelriemoliveira 

Good luck with that! :)

miguelriemoliveira, 2021-03-04:
Ok, reorganized 3d lidar labelling code so that it only uses indices and cached functions.

Much more straightforward now. This last commit still has the commented old code, for easy reference.

I will commit again soon with the old code removed.

miguelriemoliveira, 2021-03-04:
This new commit is clean. No old code commented. The code is still in calibrate. 

Later, will move it into the data collector later under #230.

miguelriemoliveira, 2021-03-06:
I am having problems in the optimization and it has to do with the limit points. So I am opening this one.

miguelriemoliveira, 2021-03-06:
Hi @aaguiar96 ,

I need your help here. There is a strange thing concerning the extraction of the 3d lidar limit points.

I changed a lot of code since last week and only now understood that it occurred last week.

The problem is the limit points all have very very small values for x,y and z, e.g. 

```
limit_points=
[{'x': 6.409690556097303e-10, 'y': 7.712782838993783e-33, 'z': 4.3435200495878235e-05, 'w': 1.0}, {'x': 7.712782838993783e-33, 'y': 6.591305834469097e-10, 'z': 4.3914831621805206e-05, 'w': 1.0}, {'x': 6.409690556097303e-10, 'y': 7.712782838993783e-33, 'z': 4.3435200495878235e-05, 'w': 1.0}, {'x': 7.712782838993783e-33, 'y': 6.591305834469097e-10, 'z': 4.3914831621805206e-05, 'w': 1.0}, {'x': 6.409690556097303e-10, 'y': 7.712782838993783e-33, 'z': 4.3435200495878235e-05, 'w': 1.0}, 
```
I spent a lot of time trying to find what I did wrong, and kept rolling back and back the commits.
I find that this started occurring in a commit you did last week:

https://github.com/lardemua/atom/commit/07312f2052200aab935ef119bff58c7b1e49bfd4?branch=07312f2052200aab935ef119bff58c7b1e49bfd4&diff=unified

The previous commit:

https://github.com/lardemua/atom/commit/f3b8a8dade69fd3b87e3ace544190541ccaed31c

 was printing the limit points just fine:

```
limit_points=
[{'x': 1.3705309629440308, 'y': -0.7009260058403015, 'z': 0.4124720096588135, 'w': 1.0}, {'x': 1.745851993560791, 'y': -0.011707000434398651, 'z': 0.4678100049495697, 'w': 1.0}, {'x': 1.375954031944275, 'y': -0.6920989751815796, 'z': 0.3555850088596344, 'w': 1.0}, {'x': 1.8381379842758179, 'y': 0.0, 'z': 0.1608159989118576, 'w': 1.0}, {'x': 1.395719051361084, 'y': -0.6845390200614929, 'z': 0.302172988653183, 'w': 1.0}, {'x': 1.8381379842758179, 'y': 0.0, 'z': 0.1608159989118576, 'w': 1.0}, {'x': 1.4107880592346191, 'y': -0.6919289827346802, 'z': 0.2488739937543869, 'w': 1.0}, {'x': 1.8381379842758179, 'y': 0.0, 'z': 0.1608159989118576, 'w': 1.0}, {'x': 1.424968957901001, 'y': -0.6870689988136292, 'z': 0.194240003824234, 'w': 1.0}, {'x': 1.8381379842758179, 'y': 0.0, 'z': 0.1608159989118576, 'w': 1.0}, {'x': 1.4509830474853516, 'y': -0.6698700189590454, 'z': 0.13981999456882477, 'w': 1.0}, {'x': 1.8381379842758179, 'y': 0.0, 'z': 0.1608159989118576, 'w': 1.0}, {'x': 1.4548569917678833, 'y': -0.6894950270652771, 'z': 0.08437500149011612, 'w': 1.0}, {'x': 1.6841330528259277, 'y': -0.28500398993492126, 'z': 0.08951599895954132, 'w': 1.0}, {'x': 1.4722150564193726, 'y': -0.685670018196106, 'z': 0.028348000720143318, 'w': 1.0}, {'x': 1.5165189504623413, 'y': -0.6214709877967834, 'z': 0.028606999665498734, 'w': 1.0}]
```

Now I tried to figure out what you did, by comparing the code and cannot understand what happened. Look at the comparison:

https://github.com/lardemua/atom/commit/07312f2052200aab935ef119bff58c7b1e49bfd4?branch=07312f2052200aab935ef119bff58c7b1e49bfd4&diff=split

I have no clue what happened. It could be related to the way the new code for reading the pcd files  works, but I am lost.

Any ideas are welcome. I will try to look into it later on.

aaguiar96, 2021-03-06:
Hi @miguelriemoliveira 

We can meet on Monday to discuss this.
But maybe it has something to do with the pypcd read/write approaches...

Is it possible that they use a different axis convention?...

miguelriemoliveira, 2021-03-07:
Hi @aaguiar96 ,

thanks. I have investigated a little more, and the problem I report above is because of using old datasets with the new way of writing and reading the pcds. I recommend we archive  all previous datasets so this does not happen again.

The problem is that the code does not break, but the point coordinates are very small. 

I still have the limit points error though ... will continue to look into it ...

miguelriemoliveira, 2021-03-07:
Hi again @aaguiar96 ,

so I was going crazy trying to find out what was wrong in your code. A lot of np stuff, which is really hard to follow.

After many attempts I decided I could not understand the code and correct it so I went to my comfort zone: dictionaries.

I have a new implementation that extracts limit points based on dictionaries which, in my opinion, is much easier to follow. 

It works. It uses dictionaries but is faster than your approach using numpy. I was actually a bit surprised with this, but here are the results:

Elapsed time is 0.012063 seconds for dictionary based approach, one collection.
Elapsed time is 0.013940 seconds for numpy based approach, same one collection.

And it works just fine.

[image]

So @aaguiar96 , I need your opinion. Please take a look at the new vs the old code and let me know if you agree with changing to the new dictionary based version.

I think its much simpler, but that's just  me, and I am becoming a dictionary extremist : - )

Here are the two versions of the limit point detection code (your version still has the very small limit point x y and z values which I could not solve):

https://github.com/lardemua/atom/blob/cbfad616b97342da6061cd14bacb9ce282528efc/atom_calibration/scripts/calibrate#L144-L276

aaguiar96, 2021-03-09:
I @miguelriemoliveira 

As we talked, the code is much better with your version.
For me, you can close this issue.

miguelriemoliveira, 2021-03-09:
Hi @aaguiar96 ,

ok, thanks for the reply. I will now try to move the new code to the data collector and labeller (#230).

Closing this one.

commit 6ce7e4fc (2021-03-10): Removed old code for 3d lidar labeling (close #251).
closed 2021-03-09 by miguelriemoliveira