lardemua/atom #254: Initiate Daniela into ATOM

miguelriemoliveira, 2021-03-05:
Lets use mmtbot as example.

1. Install ATOM
2. Install mmtbot
3. download a bag file
4. Try the steps of the atom calibration pipeline
5. Set intial estimate
6. collect data
7. calibrate

miguelriemoliveira, 2021-03-07:
Hi @danifpdra ,

I solved all pending issues this weekend. Can you work tomorrow? From 10h30 ou 11h? I am not sure when, but I can call when available ...

danifpdra, 2021-03-07:
Hi @miguelriemoliveira ,

I can work tomorrow at that time, yes. But I haven't had much time to work this weekend. Is this ok with you or should I work by myself first? I can do it before our meeting.

danifpdra, 2021-03-07:
@miguelriemoliveira, can you tell me where I can find the bag files?

miguelriemoliveira, 2021-03-07:
No problem. We can meet anyway. I solved all problems, so I just want to
walk you through all the procedure so you can repeat it later on.

I have some urgency since I will have to start preparing SVPI classes soon,
so if you could meet tonight it would be even better ...

If not we meet tomorrow around 11h.

Thanks,
Miguel
*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2021-03-07:
> @miguelriemoliveira, can you tell me where I can find the bag files?

Sure, you have some mmtbot bag files in the mmtbot readme page.

https://github.com/miguelriemoliveira/mmtbot

danifpdra, 2021-03-07:
@miguelriemoliveira, tonight will be difficult for me, can we meet tomorrow morning?

miguelriemoliveira, 2021-03-07:
Sure. Tomorrow I call you around 11h.

See you,
Miguel
*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

danifpdra, 2021-03-09:
Fatal error opening the bagfile: [FATAL] [1615307465.689518373]: Error opening file: /opt/ros/noetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/daniela/bagfiles/mmtbot/21_02_2021.bag

@miguelriemoliveira, can you help me?

miguelriemoliveira, 2021-03-09:
Hi @danifpdra ,

which command are you running when you get that error? (please edit the post above to include that information)

Also, do you have that bagfile? Where?

danifpdra, 2021-03-09:
@miguelriemoliveira , I have the bagfile in the home/daniela/bagfiles/mmtbot folder. I created an env variable $ROS_BAGS pointing to to home/daniela/bafiles .

When I run roslaunch mmtbot_calibration set_initial_estimate.launch it give me a fatal error: FATAL] [1615309043.508660631]: Error opening file: /opt/ros/noetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/daniela/bagfiles/mmtbot/21_02_2021.bag

miguelriemoliveira, 2021-03-09:
Hi @danifpdra ,

just a comment w.r.t. the style of posting. I usually do it like this, it is better to read:

@miguelriemoliveira , I have the bagfile in the **home/daniela/bagfiles/mmtbot** folder. I created an env variable $ROS_BAGS pointing to to **home/daniela/bafiles** (this is wrong, the g in bag is missing. Is it a typo?).

When I run:

`   roslaunch mmtbot_calibration set_initial_estimate.launch` 

it give me a fatal error: 

`FATAL] [1615309043.508660631]: Error opening file: /opt/ros/noetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/daniela/bagfiles/mmtbot/21_02_2021.bag`

miguelriemoliveira, 2021-03-09:
Now about the content:

Which bagfile have you configured in your **mmtbot_calibration/calibration/config.yml**?

https://github.com/miguelriemoliveira/mmtbot/blob/d9fe3efc12f9b0a2ac221fd2c4aeca3b397edcbb/mmtbot_calibration/calibration/config.yml#L35-L39

danifpdra, 2021-03-09:
Yes, it is a typo. Sorry.

I have the same bag configured in **mmtbot_calibration/calibration/config.yml**:

`bag_file: "$ROS_BAGS/mmtbot/21_02_2021.bag"

miguelriemoliveira, 2021-03-09:
OK, how about your **$ROS_BAGS** environment variable. If you type  (in the same terminal as you are launching the set initial estimate)

`echo $ROS_BAGS`

what do you get?

Di you try replacing $ROS_BAGS by the absolute path (not a good solution, but just to try)

danifpdra, 2021-03-09:
It outputs 

`/opt/ros/noetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/daniela/bagfiles`

It works when replaced with the absolute path.

miguelriemoliveira, 2021-03-09:
that's not correct. Mine outputs (as it should):

/home/mike/bagfiles

danifpdra, 2021-03-09:
Like this:

`ROS_BAGS="/home/daniela/bagfiles"´

miguelriemoliveira, 2021-03-09:
Where? .bashrc?

Mine is as follows:

export ROS_BAGS="/home/mike/bagfiles"
export ATOM_DATASETS="/home/mike/datasets"

danifpdra, 2021-03-10:
Hi @miguelriemoliveira,

The charuco doesn't appear on the image. Am I doing something wrong?

[image]

miguelriemoliveira, 2021-03-10:
Hi @danifpdra ,

lets create one issue for each new topic. It will be easier to find the problems later on. 

This issue should be named something like "problems with $ROS_BAG configuration".

And the last question should be an issue like "Cannot visualize charuco in cameras".

Please create this second issue and then I will respond there.

miguelriemoliveira, 2021-03-10:
and sorry for being such a nit-picker, but I am really convinced the organization helps, especially in complex projects like this. 
Also, I am trying to teach you my methodologies so please be patient :  - )

miguelriemoliveira, 2021-03-11:
This could also be closed?

closed 2021-04-01 by miguelriemoliveira