lardemua/atom #259: Charuco not appearing on camera image

danifpdra, 2021-03-10:
Hi @miguelriemoliveira,

The charuco doesn't appear on the camera image. Am I doing something wrong?

[image]

miguelriemoliveira, 2021-03-10:
OK, you should always write the command you are using ... I suspect it is the? 

`roslaunch mmtbot_bringup sim.launch gui:=false`

danifpdra, 2021-03-10:
gui:=true. I already edited the initial comment.

miguelriemoliveira, 2021-03-10:
Ok, so what do you see in the gazebo frontend? Where is the pattern? Can you post a screenshot of gazebo?

danifpdra, 2021-03-10:
The pattern does not appear in gazebo.

[image]

miguelriemoliveira, 2021-03-10:
OK, what the code is doing is sending the pose of the pattern to the gazebo. Sometimes (I could not understand why) the communication between the rviz and the gazebo fails. The initial pose of the pattern in  gazebo is  under the ground, which is why you don't see it.

Try one of these:

1. move the pattern around in rviz (using the pattern's interactive marker) and see if gazebo updates
2. Just for testing, try to move the pattern in gazebo. Do you know how? Select the pattern from the tree on the left side, then select the manipulate object and move the pattern. If you want I can record a small video...

danifpdra, 2021-03-10:
The hand camera is not point to the ground: 

[image]

I have already tried to move the pattern in rviz but it does not update in the gazebo. I can't move the pattern in gazebo because it is not there. I selected from the tree and it points to nothing.

miguelriemoliveira, 2021-03-10:
> The hand camera is not point to the ground:

Right, you moved the arm already. My bad.

About the gazebo, I will post a video in a moment.

miguelriemoliveira, 2021-03-10:
One possibility is that your machine is not good enough. Can you post your specs? Also, what is the reported Real time factor in gazebo? Mine is 0.4 ...

danifpdra, 2021-03-10:
Mine is 0.29

danifpdra, 2021-03-10:
[image]

miguelriemoliveira, 2021-03-10:
See if this helps https://youtu.be/L_e2gHEY4BM

miguelriemoliveira, 2021-03-11:
This is working now, according to @danifpdra .

closed 2021-03-11 by miguelriemoliveira