lardemua/atom #260: Calibrate - Publish static transforms

miguelriemoliveira, 2021-03-10:
One of the problems with #253 could be the large number of published transforms. One way of reducing it would be to publish static transforms wherever the transformations are fixed ...

miguelriemoliveira, 2021-03-11:
For now things are running smooth. Check #253. As such, I will give this one a low priority.

miguelriemoliveira, 2021-09-18:
Hi @danifpdra when running calibrate with visualization the rviz becomes a bit slow. 
Does it happen in your system too?

I think the problem is that we are publishing the larcc's meshes (i.e. gantry, manipulator, sensors, etc) over 20 times, one per collection.

[image]

The idea to solve this is to try to publish once the meshes that do not move, and continue to publish the others one per collection.

I will try to implement, can I then ask you to try it out with larcc to see if the user experience in rviz becomes better?

danifpdra, 2021-09-18:
Hi @miguelriemoliveira,

Of course. Tell me if there's anything else I can help you with.

miguelriemoliveira, 2021-09-18:
BTW, there is an override to the automatic detection of whether the robot is static or not. Its the -ajf flag. 
Setting this flag results in 18 FPW in RVIZ for me. The solution should give something in between ...

miguelriemoliveira, 2021-09-18:
Hi @danifpdra ,

I think I got it. The code now only sends multiple times meshes which are not immovable, which makes it run faster while showing the same.

In my system I now get 15 FPS  and 

```
rostopic bw /calibrate/robot_meshes
subscribed to [/calibrate/robot_meshes]
average: 72.56KB/s
	mean: 35.45KB min: 35.45KB max: 35.45KB window: 2
```

Can you confirm it works from your side?

In the future I will change the detection mechanism to make it more general (i.e. to also with the use_tfs option). The idea is to look at the transforms in the dataset to see if they are immovable.

miguelriemoliveira, 2021-09-19:
New version implemented. @danifpdra  when you can please test and if it works you may close the issue.

danifpdra, 2021-09-20:
Hi @miguelriemoliveira ,

I've just tried and got this 

```
average: 38.35KB/s
	mean: 35.45KB min: 35.45KB max: 35.45KB window: 4
average: 37.73KB/s
	mean: 35.45KB min: 35.45KB max: 35.45KB window: 5
average: 31.10KB/s
	mean: 35.45KB min: 35.45KB max: 35.45KB window: 5
average: 31.74KB/s
	mean: 35.45KB min: 35.45KB max: 35.45KB window: 6
average: 31.17KB/s
	mean: 35.45KB min: 35.45KB max: 35.45KB window: 7
```

So I assume it's working

miguelriemoliveira, 2021-09-20:
Thanks @danifpdra ... how many FPS in RVIZ, before and after?

danifpdra, 2021-09-20:
Before around 7, now around 14-15Hz

miguelriemoliveira, 2021-09-20:
So it works. Thanks. Closing...

closed 2021-09-20 by miguelriemoliveira