lardemua/atom #262: Instructions for evaluating on readme

miguelriemoliveira, 2021-03-11:
Hi @aaguiar96 ,

could you write down some instructions on how to evaluate the results of a calibration and compare them against the standard methods?

If you did this we could ask @danifpdra to validate those instructions by  evaluating  the calibration of mmtbot.

aaguiar96, 2021-03-11:
I @miguelriemoliveira 

Yes, it is a good idea.
I will do that soon.

miguelriemoliveira, 2021-03-11:
Hi @aaguiar96 ,

just remembered. This is not yet migrated to python3. 

Can you do it?

@danifpdra , when you have them send two datasets to @aaguiar96 (calibrated train and test) so he can try the code.

aaguiar96, 2021-03-11:
Yes @miguelriemoliveira, I'll do it. :-)

aaguiar96, 2021-03-11:
Hi @miguelriemoliveira  and @danifpdra 

I migrated the `atom_evaluation` code to python3 and tested the three scripts with a mmtbot test calibration.
It worked really well.

I will now explain the process on the readme file.

miguelriemoliveira, 2021-03-11:
Great news. How did you get the datasets? You made some by yourself?

aaguiar96, 2021-03-11:
> Great news. How did you get the datasets? You made some by yourself?

Yes, I made one just for test purposes.

aaguiar96, 2021-03-11:
Done! 

@danifpdra let me know if you have any doubt.

closed 2021-03-11 by aaguiar96

miguelriemoliveira, 2021-03-11:
Thanks @aaguiar96 ,

I looked at it and it looks really nice.

Thanks. @danifpdra its on you. Let us know if you need something else.