lardemua/atom #263: Problems calibrating

danifpdra, 2021-03-11:
Hi @miguelriemoliveira ,

I am having a problem when trying to calibrate the system, that produces the error:

`Traceback (most recent call last):
  File "/home/daniela/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/daniela/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 530, in <module>
    main()
  File "/home/daniela/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 467, in main
    opt.startOptimization(optimization_options=options)
  File "/home/daniela/.local/lib/python3.8/site-packages/OptimizationUtils-1.0.0-py3.8.egg/OptimizationUtils/OptimizationUtils.py", line 375, in startOptimization
  File "/home/daniela/.local/lib/python3.8/site-packages/OptimizationUtils-1.0.0-py3.8.egg/OptimizationUtils/OptimizationUtils.py", line 346, in errorDictToList
AttributeError: 'collections.OrderedDict' object has no attribute 'has_key'`

Do you know what this is about?

miguelriemoliveira, 2021-03-11:
Hi Daniela,

Did you git pull optimization utils?

danifpdra, 2021-03-11:
It's already up to date.

miguelriemoliveira, 2021-03-11:
Hum, I don't think so. Lets zoom for 10 mins?
Now?

danifpdra, 2021-03-11:
yes, now

miguelriemoliveira, 2021-03-12:
Closing. It was a problem with not using the correct branch.

closed 2021-03-12 by miguelriemoliveira