lardemua/atom #264: Large .json file

danifpdra, 2021-03-11:
Hi @miguelriemoliveira and @aaguiar96,

The **atom_calibration.json** file generated after calibration has a size of 202MB. Is this a bug?

miguelriemoliveira, 2021-03-11:
That's strange, my bagfile has only 3 MB. 

I see that the recording of the calibrated json is done at

https://github.com/lardemua/atom/blob/ba744e21b9746bb36992531d7d5530e2dcc7c4b5/atom_calibration/scripts/calibrate#L476-L480

@aaguiar96 , that should go through your new code right? How does it behave if the images and point clouds already exist? It should skip the recording and create a "data_file" field replacing the "data" field, right?

miguelriemoliveira, 2021-03-11:
@danifpdra when reporting a bug it is always important to say which command you were using. Can you edit your initial comment?

danifpdra, 2021-03-11:
> @danifpdra when reporting a bug it is always important to say which command you were using. Can you edit your initial comment?

Done

miguelriemoliveira, 2021-03-11:
My test was without the **nig** flag, can you run a calibration without it and see if the bagfile is still huge?

aaguiar96, 2021-03-11:
> @aaguiar96 , that should go through your new code right? How does it behave if the images and point clouds already exist? It should skip the recording and create a "data_file" field replacing the "data" field, right?

I @miguelriemoliveira, are you referring to the point cloud write/read new routines? That's one what I've changed. It should not interfere with the atom's workflow...

aaguiar96, 2021-03-11:
@danifpdra can you check if the point clouds are being written to the output optimization files? That is, what's the difference between the input and the output json?

miguelriemoliveira, 2021-03-11:
Hi @aaguiar96 , yes, it should not but I saw binary arrays inside the calibrated json, so it seems the mechanism to save (or not save if the file already exists) is not working properly ...

@danifpdra , can you scan the json file and look for a "data:" or the "data_file:" pattern and output here. Something like:

`   grep "\"3dlidar\":" atom_calibration.json -A 4 -B 4`

My output is a lot of this for several collections:

```json
  },
  "collections": {
    "0": {
      "additional_data": {},
      "data": {
        "3dlidar": {
          "data_file": "3dlidar_0.pcd",
          "fields": [
            {
```

danifpdra, 2021-03-11:
> My test was without the **nig** flag, can you run a calibration without it and see if the bagfile is still huge?

I did a calibration with it and the output has the same size.

danifpdra, 2021-03-11:
> `grep "\"3dlidar\":" atom_calibration.json -A 4 -B 4`

  `  },
    "description_file": "package://mmtbot_description/urdf/mmtbot.urdf.xacro",
    "max_duration_between_msgs": 1000,
    "sensors": {
      "3dlidar": {
        "child_link": "3dlidar_base_link",
        "link": "3dlidar",
        "parent_link": "tripod_left_support",
        "topic_name": "/3dlidar/points"
--
  "collections": {
    "0": {
      "additional_data": {},
      "data": {
        "3dlidar": {
          "data_file": "3dlidar_0.pcd",
          "fields": [
            {
              "count": 1,
--
          "width": 1280
        }
      },
      "labels": {
        "3dlidar": {
          "detected": true,
          "idxs": [
            5357,
            5373,
--
    },
    "1": {
      "additional_data": {},
      "data": {
        "3dlidar": {
          "data_file": "3dlidar_1.pcd",
          "fields": [
            {
              "count": 1,
--
          "width": 1280
        }
      },
      "labels": {
        "3dlidar": {
          "detected": true,
          "idxs": [
            5482,
            5498,
--
    },
    "10": {
      "additional_data": {},
      "data": {
        "3dlidar": {
          "data_file": "3dlidar_10.pcd",
          "fields": [
            {
              "count": 1,
--
          "width": 1280
        }
      },
      "labels": {
        "3dlidar": {
          "detected": true,
          "idxs": [
            5236,
[... 466 lines cut ...]
        "3dlidar": {
          "data_file": "3dlidar_9.pcd",
          "fields": [
            {
              "count": 1,
--
          "width": 1280
        }
      },
      "labels": {
        "3dlidar": {
          "detected": true,
          "idxs": [
            5236,
            5237,
--
      ]
    }
  },
  "sensors": {
    "3dlidar": {
      "_name": "3dlidar",
      "calibration_child": "3dlidar_base_link",
      "calibration_parent": "tripod_left_support",
      "chain": [ `

danifpdra, 2021-03-11:
> > Hi @miguelriemoliveira, are you referring to the point cloud write/read new routines? That's only what I've changed. It should not interfere with the atom's workflow...
> 
> Hi @aaguiar96 , yes, it should not but I saw binary arrays inside the calibrated json, so it seems the mechanism to save (or not save if the file already exists) is not working properly ...
> 
> @danifpdra , can you scan the json file and look for a "data:" or the "data_file:" pattern and output here. Something like:
> 
> ` grep "\"3dlidar\":" atom_calibration.json -A 4 -B 4`
> 
> My output is a lot of this for several collections:
> 
> ```json
>   },
>   "collections": {
>     "0": {
>       "additional_data": {},
>       "data": {
>         "3dlidar": {
>           "data_file": "3dlidar_0.pcd",
>           "fields": [
>             {
> ```

The output is huge. Do you want me to paste it here?

danifpdra, 2021-03-11:
> @danifpdra can you check if the point clouds are being written to the output optimization files? That is, what's the difference between the input and the output json?

I can't even open the file properly. The idx are being written in the file but I don't think that the point clouds are.

miguelriemoliveira, 2021-03-11:
>I did a calibration with it and the output has the same size.

You mean without it (without the nig flag)?

danifpdra, 2021-03-11:
> > I did a calibration with it and the output has the same size.
> 
> You mean without it (without the nig flag)?

Yes, without the nig flag, the output has the same size.

danifpdra, 2021-03-11:
Is this what you mean?

`"12": {
      "additional_data": {},
      "data": {
        "3dlidar": {
          "data_file": "3dlidar_12.pcd",
          "fields": [
            {
              "count": 1,
--
          "width": 1280
        }
      },
      "labels": {
        "3dlidar": {
          "detected": true,
          "idxs": [
            5190,
            5206,`

aaguiar96, 2021-03-11:
Hi @miguelriemoliveira  and @danifpdra 

I found the cause of the problem. The image data is being written to the output json file...

aaguiar96, 2021-03-11:
Here:

```
...

world_camera": {
  "data": "s7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7    Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7O    zs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Oz    s7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs    7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7    Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7O    zs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Oz    s7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs    7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7    Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7O    zs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Oz    s7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs    7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7    Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7O    zs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Oz

....
```

miguelriemoliveira, 2021-03-11:
Ah, so I guess a 

grep "\"world_camera\":" atom_calibration.json -A 4 -B 4

Would show the **data:** instead of the **data_file** ... can you confirm @danifpdra ?

miguelriemoliveira, 2021-03-11:
OK, I will try to take a look why this is occurring later tonight. Thanks @aaguiar96 for finding the problem.

danifpdra, 2021-03-11:
> grep ""world_camera":" atom_calibration.json -A 4 -B 4

This command does not output anything.

miguelriemoliveira, 2021-03-11:
Strange ... is the name of the sensor correct?

Try to find some camera ...

danifpdra, 2021-03-11:
The command was not right, the \ were missing. 

It outputs a million characters like:

`/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj4+Pj49/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39vb29vb29vb29vb29vb29vb29vb29vb29vb29vb29vb29vb29vb29vb29vb29vb29/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39/f39fX19fX19fX19fX19fX19fX19fX19fX19fX19fX19fX19fX19fX19fX19fX19fX19vb29vb29vb29vb29vb29vb29vb29vb29fX19fX19fX19fX19fX19fX19fX19fX19PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT09PT08/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz8/Pz9PT09PT09PT09PT09PT09PT09PT09PT08vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8vLy8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx8PDw8PDw8PDw8PDw8PDw8PDw8PDw8PDw8fHx8fHx8fHx8fHx8fHx8fHx8fHx8fHx7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v8PDw8PDw8PDw8PDw8PDw8PDw8PDw8PDw7+/v7+/v7+/v7+/v7+/v7+/v7+/v7+/v7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7e3t7e3t7e3t7e3t7e3t7e3t7e3t7e3t7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7u7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs7Ozs6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6+vr6enp6enp6enp6enp6enp6enp6enp6enp6enp6enp6enp6enp6enp6enp6enp6enp6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo6Ojo5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5+fn5ubm5ubm5ubm5ubm5ubm5ubm5ubm5ubm5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5eXl5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk5OTk4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4eHh4eHh4eHh4eHh4eHh4eHh4eHh4eHh4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg`

miguelriemoliveira, 2021-03-11:
So that confirms, as @aaguiar96 said, we are writing the binary blob for cameras, which is why the file is 200MB. It contains 52 images inside (26 collections x 2 cameras ).

I will look into it tonight,

miguelriemoliveira, 2021-03-11:
Solved it. It was a problem with the conversion from the dictionary back to message, which could not have the field data_file. 
Now I remove the field in a copy of the dictionary, rather than the dataset. IT is working. The json now has 3MB instead of 200MB.

@danifpdra can you try it and confirm it is working? If so please close the issue ...

danifpdra, 2021-03-12:
I confirm that this is solved.

commit bc7386cc (2021-03-11): Fixed problem with json containing data field (Fix #264)
closed 2021-03-12 by danifpdra