lardemua/atom #265: colect_data.launch not saving collections

danifpdra, 2021-03-15:
Hi @miguelriemoliveira , @aaguiar96 ,

When I run 

`roslaunch mmtbot_calibration collect_data.launch  output_folder:=$ATOM_DATASETS/dataset_15_03_2021/ bag_rate:=0.5 overwrite:=true`

I get the following error:

`[ERROR] [1615815232.706686, 1244.637140]: bad callback: <bound method InteractiveMarkerServer.processFeedback of <interactive_markers.interactive_marker_server.InteractiveMarkerServer object at 0x7f728d21e550>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/interactive_markers/interactive_marker_server.py", line 353, in processFeedback
    feedback_cb(feedback)
  File "/opt/ros/noetic/lib/python3/dist-packages/interactive_markers/menu_handler.py", line 143, in processFeedback
    context.feedback_cb(feedback)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 30, in menuFeedback
    data_collector.saveCollection()
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 285, in saveCollection
    atom_core.dataset_io.saveResultsJSON(output_file, D)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 112, in saveResultsJSON
    createDataFile(dataset_in, collection_key, sensor, sensor_key, output_folder)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 138, in createDataFile
    print('Collection ' + collection_key + '. Creating data file for sensor ' + sensor_key + ' msg type ' + sensor[
TypeError: can only concatenate str (not "int") to str `

And I try to save several collections, the folder datasets/dataset_15_03_2021 is created but no collection is saved there. Can you help me? Am I doing anything wrong?

miguelriemoliveira, 2021-03-15:
Great.

miguelriemoliveira, 2021-03-15:
Hi @danifpdra ,

You should push the new version that fixes this issue, otherwise it is only solved for you.

Do a commit with reference to this issue.

danifpdra, 2021-03-15:
Hi @miguelriemoliveira ,

I already did a commit with this, I forgot yo reference this issue but it«s already fixed.

miguelriemoliveira, 2021-03-15:
Ah, in that case you can say something like:

This was fixed by commit 5541ab5d73e7f8463aaf462b28562d262fbb0e25

danifpdra, 2021-03-15:
Ok, sorry

danifpdra, 2021-03-15:
This was fixed by commit 5541ab5d73e7f8463aaf462b28562d262fbb0e25 .

closed 2021-03-15 by danifpdra

miguelriemoliveira, 2021-03-15:
No problem, you are learning : - )