lardemua/atom #268: Camera to camera evaluation

miguelriemoliveira, 2021-03-18:
Hi @aaguiar96 ,

could we change the convention displayed in the images for the **camera_to_camera_evalutation.py**

[image]

Could we change + to dots? Because that's the convention we use in the calibrate script, and it should be uniform.

miguelriemoliveira, 2021-03-18:
Could you also check that the target and source sensors exists and output accordingly if they don't?
Right now I get:

```
in <module>
    K_s[0, :] = train_dataset['sensors'][source_sensor]['camera_info']['K'][0:3]
KeyError: 'hand_camera'
```

Thanks.

miguelriemoliveira, 2021-03-18:
Another error:

```
Traceback (most recent call last):
  File "/home/mike/catkin_ws/src/calibration/atom/atom_evaluation/scripts/camera_to_camera_evalutation.py", line 187, in <module>
    for collection_key, collection in test_dataset['collections'].items():
RuntimeError: dictionary changed size during iteration

```

I had the same error when passing to python3, you want me to fix it?

miguelriemoliveira, 2021-03-18:
Lets use meters always ... SI forever!

aaguiar96, 2021-03-31:
This is also closed.

closed 2021-03-31 by aaguiar96