lardemua/atom #269: Generalize camera_to_range_sensor.py evaluation script to eye in hand cases.

miguelriemoliveira, 2021-03-18:
After #266 we notice that this script should also be generalized to eye in hand cases.

aaguiar96, 2021-03-19:
This is implemented with bc30055. @danifpdra can you test, please?

```
------------------------------------------------------------------------------------------------------------------------------------------------------------
            #                       RMS                    X Error                  Y Error           X Standard Deviation     Y Standard Deviation
------------------------------------------------------------------------------------------------------------------------------------------------------------
Reading point cloud from /home/andresaguiar/Documents/atom_datasets/mmtbot_cam2cam/dataset_15_03_2021_2/3dlidar_0.pcd
            0                        -                     3.1674                   1.7199                   3.9944                   1.9797
Reading point cloud from /home/andresaguiar/Documents/atom_datasets/mmtbot_cam2cam/dataset_15_03_2021_2/3dlidar_1.pcd
            1                        -                     4.1014                   2.2709                   5.4268                   2.7832
Reading point cloud from /home/andresaguiar/Documents/atom_datasets/mmtbot_cam2cam/dataset_15_03_2021_2/3dlidar_2.pcd
            2                        -                     3.0579                   2.1346                   4.1682                   2.8483
Reading point cloud from /home/andresaguiar/Documents/atom_datasets/mmtbot_cam2cam/dataset_15_03_2021_2/3dlidar_3.pcd
```

[image]

danifpdra, 2021-03-19:
Hi @aaguiar96,

I tested it with a different train and test dataset:
`rosrun atom_evaluation camera_to_camera_evalutation.py -train_json $ATOM_DATASETS/dataset_15_03_2021_2/atom_calibration.json -test_json $ATOM_DATASETS/dataset_11_03_2021/data_collected.json -ss hand_camera -ts world_camera`

And the results were:

```
---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
  0      -       1.5393    0.9194    0.2552    0.3671    4.9294    0.1633  
  1      -       1.3626    0.7106    0.2560    0.4157    5.1881    0.1591  
  2      -       1.6145    0.4966    0.2370    0.4099    5.3166    0.1597  
  3      -       1.2917    0.3427    0.2579    0.3736    5.7238    0.1436  
  4      -       1.2917    0.3427    0.2579    0.3736    5.7238    0.1436  
  5      -       1.8736    0.5220    0.2224    0.3199    6.0703    0.1660  
  6      -       1.9148    0.6004    0.2365    0.3762    6.0619    0.1626  
  7      -       0.8622    0.2636    0.2538    0.3002    5.4380    0.1367  
  8      -       1.5435    0.5257    0.2443    0.3843    5.7805    0.1499  
  9      -       1.2753    0.4575    0.2771    0.3284    5.6461    0.1249  
 10      -       1.6239    1.0367    0.3537    0.4738    5.7140    0.1689  
 11      -       1.1858    1.1298    0.2703    0.5319    6.2293    0.1638  
 12      -       0.8966    0.3552    0.2440    0.3843    4.8062    0.1490  
 13      -       0.7753    0.2599    0.2660    0.3126    5.0616    0.1293  
 14      -       1.8985    0.6475    0.2772    0.3488    5.9800    0.1511  
 15      -       1.8000    0.3475    0.2240    0.3498    6.0460    0.1766  
 16      -       0.4802    0.5065    0.1970    0.2047    5.3311    0.1268  
 17      -       1.0027    0.3024    0.2015    0.3174    5.4761    0.1188  
 18      -       1.2257    0.2717    0.1466    0.1756    5.0110    0.1803  
 19      -       0.7915    0.1983    0.1598    0.1960    4.8391    0.1687  
 20      -       1.2111    0.2373    0.1555    0.2617    5.0407    0.1804  
 21      -       0.9924    0.2365    0.2055    0.2680    4.7182    0.1713  
 22      -       2.9583    0.5432    0.1776    0.7562    6.4587    0.2813  
 23      -       1.7053    0.3331    0.3241    0.3752    4.9006    0.2314  
 24      -       1.6747    0.5126    0.3153    0.8076    4.0352    0.3381  
 25      -       0.2295    0.6574    0.2754    0.7758    1.3454    0.2387  
 26      -       1.6907    0.4886    0.3143    0.5467    5.3579    0.2282  
---------------------------------------------------------------------------
 All   1.1390    1.3671    0.4850    0.5768    0.5630    0.0053    0.0030  
--------------------------------------------------------------------------- 
```

I think it is working well. What do you think of the results?

aaguiar96, 2021-03-19:
Hi @danifpdra 

The results are fine. But I was asking you to test the `camera_to_range_sensor.py` evaluation script.

Follow the instructions in the README file and if you have any question feel free to ask.

eupedrosa, 2021-03-19:
I would say that for a virtual dataset the results could be better.
Most of the errors seems to be in the X axis. 

@miguelriemoliveira, @aaguiar96 and @danifpdra, what is you opinion?

danifpdra, 2021-03-19:
> Hi @danifpdra
> 
> The results are fine. But I was asking you to test the `camera_to_range_sensor.py` evaluation script.
> 
> Follow the instructions in the README file and if you have any question feel free to ask.

@aaguiar96 , can you explain me what should be the `ef <path_to_output_annotation_json_file>`?

danifpdra, 2021-03-19:
> I would say that for a virtual dataset the results could be better.
> Most of the errors seems to be in the X axis.
> 
> @miguelriemoliveira, @aaguiar96 and @danifpdra, what is you opinion?

Hi @eupedrosa ,

I induce error in the calibration with the -nig flag, maybe that is why.

aaguiar96, 2021-03-19:
> I would say that for a virtual dataset the results could be better.
> Most of the errors seems to be in the X axis.
> 
> @miguelriemoliveira, @aaguiar96 and @danifpdra, what is you opinion?

I @eupedrosa 
I don't know how careful was this calibration (we are still experimenting).

aaguiar96, 2021-03-19:
> @aaguiar96 , can you explain me what should be the `ef <path_to_output_annotation_json_file>`?

In the LiDAR-to-Camera evaluation, you will have to annotate the pattern limits.
To do so you should activate the `-ua` flag. When you finish the annotation&evaluation a json file will be written with the pattern limits in the path specified by the `-ef` flag.

Thus, if you want to re-run the evaluation for the same input datasets, you can deactivate the `-ua` flag to disable the annotation and set the `-ef` flag to the path of the json with the annotations.

So, the `-ef` flag is either a path to an output file if the annotation is activated or a path to an input file if you have already annotated the pattern limits.

danifpdra, 2021-03-19:
@aaguiar96 , I am doing something wrong, can you help me?

[image]

miguelriemoliveira, 2021-03-19:
Nice drawing : -)

aaguiar96, 2021-03-19:
> @aaguiar96 , I am doing something wrong, can you help me?

Yes, you are. :-)

You want to annotate four classes, one for each edge.
The flow is:

(1) [click + s] from one corner of the edge to the other
(2) [p] to change the class
(3) When you've annotated the four classes [4 times p pressed] the result will appear
(4) [space] to move to the next image

From the image you are showing, you are annotating only one class for all the borders (you did not click on p to change the class).
Also, make sure your first and last points for each class are the edge corners.

danifpdra, 2021-03-19:
@aaguiar96 , what about in this case where we don't see the 4 corners? Should I do just 3 ou 5, 3 corners and 2 points in the limits ?

aaguiar96, 2021-03-19:
> @aaguiar96 , what about in this case where we don't see the 4 corners? Should I do just 3 ou 5, 3 corners and 2 points in the limits ?

In that case, you annotate the left/rightmost pixels of each edge as in the pictures above.

aaguiar96, 2021-03-19:
Hi @danifpdra 

You're almost there. The only thing missing is that you have not started in each corner for all the classes. When you pass from one class to another, you have to clock again on the corner to add it to the new class.

Yes, you should repeat for all the collections of the test dataset. Once you've done it and if you want to re-run the evaluation, you can simply load the annotations from the exported json file.

eupedrosa, 2021-03-19:
I have to say, this is nothing but an easy procedure.
Furthermore, the "user" error will be propagated to the final errors.

Is there a better way to do this? Like, something automatic, or semi-automatic?
Just to follow the instructions you have to be a PhD student. Luckily @aaguiar96 and @danifpdra are such people. 
But I am not :smile: 

In my opinion, this should be improved. I bet @miguelriemoliveira will say that I can do it my self.

aaguiar96, 2021-03-19:
Hi @eupedrosa 

The procedure is very easy if you're familiar with it.
And, like every user-annotated metric, it has errors associated but I think that they are not significant. This approach is used in almost all the DL/ML evaluation metrics...

If you have an automatic procedure perhaps the pattern limits detection will have more error than the ones produced by annotation.

What is your idea for the automatic procedure?

danifpdra, 2021-03-19:
I agree that it is not very easy, and it is time-consuming. What if we had 100 collections?

And what if I get one point in the wrong place? There is no way to delete it?

aaguiar96, 2021-03-19:
> I agree that it is not very easy, and it is time-consuming. What if we had 100 collections?

This happens in all the annotation processes. When you are annotating a Deep Learning dataset with 10,000 images, you just have to do it... 
Of course, if there is some automatic way of doing this, it would be better, but I don't see how.

> 
> And what if I get one point in the wrong place? There is no way to delete it?

Here I agree that we should add this feature. Currently there is no way of deleting.

aaguiar96, 2021-03-19:
@danifpdra if you have some suggestions to ease the process, report them and I can implement them. :)

danifpdra, 2021-03-19:
Don't you think it would be more accurate with lines instead of points?

eupedrosa, 2021-03-19:
> What is your idea for the automatic procedure?

Here is one idea:
1. We detect the pattern (we sould already have that)
2. Calculate the 4 corners of the pattern (I believe this is also done somewhere else)
3. Add intermediate points between edges.
4. profit :partying_face: 

Can this be done?

aaguiar96, 2021-03-19:
> Don't you think it would be more accurate with lines instead of points?

No, because due to distortion, the pattern edges are not lines but curves.

eupedrosa, 2021-03-19:
> No, because due to distortion, the pattern edges are not lines but curves.

We know the distortion .... we can calculate the line points and then project

aaguiar96, 2021-03-19:
> Here is one idea:
> 
>     1. We detect the pattern (we sould already have that)
> 
>     2. Calculate the 4 corners of the pattern (I believe this is also done somewhere else)
> 
>     3. Add intermediate points between edges.
> 
>     4. profit partying_face
> 
> 
> Can this be done?

This is a good idea @eupedrosa. The only thing is that in 3. you will have to know the distortion to compute points in the boundaries. Will this work with accuracy? We can try. :)

eupedrosa, 2021-03-19:
If you do not trust in 3), let it be the only thing the user has to do.

aaguiar96, 2021-03-19:
> If you do not trust in 3), let it be the only thing the user has to do.

Add points? This is the only thing the user has to do by now...

aaguiar96, 2021-03-19:
If you want to make this automatic, I think we should implement your entire suggestion. Otherwise, it will not be worthy.

eupedrosa, 2021-03-19:
I agree with your last comment.

danifpdra, 2021-03-19:
Well I've done it but I made a mistake in two collections and it affected all the results...

```
------------------------------------------------------------------------------------------------------------------------------------------------------------
            #                       RMS                    X Error                  Y Error           X Standard Deviation     Y Standard Deviation   
------------------------------------------------------------------------------------------------------------------------------------------------------------
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_0.pcd
            0                        -                     1.8596                   1.5557                   2.3540                   1.8799          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_1.pcd
            1                        -                     2.0471                   1.6442                   2.9226                   2.2966          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_2.pcd
            2                        -                     2.5486                   2.1386                   3.1574                   2.3560          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_3.pcd
sys:1: RankWarning: Polyfit may be poorly conditioned
            3                        -                    172.4886                 228.7599                 160.2617                 152.1839         
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_4.pcd
sys:1: RankWarning: Polyfit may be poorly conditioned
            4                        -                    171.9319                 228.9178                 160.2703                 152.1413         
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_5.pcd
            5                        -                     2.5344                   2.3068                   2.8296                   3.5261          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_6.pcd
            6                        -                     2.0528                   2.0892                   2.5810                   2.5566          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_7.pcd
            7                        -                     1.9781                   2.1700                   1.9726                   2.7572          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_8.pcd
            8                        -                     1.9542                   2.2141                   2.4378                   2.7832          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_9.pcd
            9                        -                     4.4713                   1.6395                   5.6099                   2.0599          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_10.pcd
           10                        -                     2.0067                   0.9093                   2.8095                   1.2519          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_11.pcd
           11                        -                     2.3907                   0.5380                   3.3131                   0.6827          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_12.pcd
           12                        -                     1.7978                   0.3959                   2.2195                   0.5555          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_13.pcd
           13                        -                     2.8439                   1.0956                   3.7668                   1.3404          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_14.pcd
           14                        -                     2.1687                   0.8503                   2.8275                   1.1241          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_15.pcd
           15                        -                     2.2280                   1.7405                   2.9512                   2.1027          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_16.pcd
           16                        -                     2.5941                   2.2145                   3.2869                   2.8604          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_17.pcd
           17                        -                     1.5098                   1.3201                   1.9517                   1.7204          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_18.pcd
           18                        -                     2.1444                   2.1965                   1.7013                   2.4147          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_19.pcd
           19                        -                     3.8575                   3.5345                   5.0672                   4.6296          
Reading point cloud from /home/daniela/datasets/dataset_11_03_2021/3dlidar_20.pcd
[... cut ...]
------------------------------------------------------------------------------------------------------------------------------------------------------------
           All                    70.6887                  16.3109                  20.4591                  61.2506                  75.8443         
------------------------------------------------------------------------------------------------------------------------------------------------------------

```

I think this is the major problem of this method.... Is there a way of repeating just the two collections?

aaguiar96, 2021-03-19:
Let @miguelriemoliveira talk too. I am afraid that we will not achieve an accurate solution.

Annotation is accurate (the errors are despicable). But we can try of course! :)

aaguiar96, 2021-03-19:
> Let @miguelriemoliveira talk too. I am afraid that we will not achieve an accurate solution.
> 
> Annotation is accurate (the errors are despicable). But we can try of course! :)

There is but is a trick. I will implement a feature to do this, delete annotations in real-time. If you have any other suggestions please let me know.

@danifpdra do you consider the process complex?

danifpdra, 2021-03-19:
@aaguiar96, I don't think it is complex after you get it but I wouldn't have gotten there alone with the explanation on the readme. Maybe this should be improved.

I think it is a bit time consuming and as I mentioned before, I believe the biggest fault is the fact that we can't delete the points or correct the wrong collections. 

Anyhow, are the results in the collections that were not badly annotated ok? Is this in mm? What do you think is a "good" error value?

aaguiar96, 2021-03-19:
> @aaguiar96, I don't think it is complex after you get it but I wouldn't have gotten there alone with the explanation on the readme. Maybe this should be improved.

Can you do this please? You have a better notion of how much the use have to know to replicate the process.

> I think it is a bit time consuming and as I mentioned before, I believe the biggest fault is the fact that we can't delete the points or correct the wrong collections.

Ok, I will work on this.

> Anyhow, are the results in the collections that were not badly annotated ok? Is this in mm? What do you think is a "good" error value?

The results are in pixels, and as @eupedrosa said before, for a simulation we should get better results (less than 1 pixel I guess).

danifpdra, 2021-03-19:
Ok, I will update the readme.

Maybe we should, but as I said before, I induce error in the calibration with the -nig flag. I don't know how much this affects the results.

aaguiar96, 2021-03-19:
> Maybe we should, but as I said before, I induce error in the calibration with the -nig flag. I don't know how much this affects the results.

You can try to recalibrate the system without this flag for the same dataset and see how the error changes.

danifpdra, 2021-03-19:
If I do that with the same collections, I don't need to annotate again, right?

aaguiar96, 2021-03-19:
> If I do that with the same collections, I don't need to annotate again, right?

Right. Just use the annotations file that was saved.

danifpdra, 2021-03-19:
@aaguiar96, I have just notice that the annotation file was not save previously, with the error:

`Saving the json output file to /home/daniela/datasets/annotation/, please wait, it could take a while ...
Traceback (most recent call last):
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_evaluation/scripts/range_sensor_to_camera_evaluation.py", line 341, in <module>
    createJSONFile(eval_file, output_dict)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_evaluation/scripts/range_sensor_to_camera_evaluation.py", line 54, in createJSONFile
    f = open(output_file, 'w')
IsADirectoryError: [Errno 21] Is a directory: '/home/daniela/datasets/annotation/' `

You told me that the -ef flag should be a folder where the annotation json would be saved.

aaguiar96, 2021-03-19:
No, the `-ef` flag is a path to a file. 
Look here:

```
usage: range_sensor_to_camera_evaluation.py [-h] -train_json TRAIN_JSON_FILE -test_json TEST_JSON_FILE -ss
                                            SOURCE_SENSOR -ts TARGET_SENSOR [-si] -ef EVAL_FILE [-ua]

optional arguments:
  -h, --help            show this help message and exit
  -train_json TRAIN_JSON_FILE, --train_json_file TRAIN_JSON_FILE
                        Json file containing input training dataset.
  -test_json TEST_JSON_FILE, --test_json_file TEST_JSON_FILE
                        Json file containing input testing dataset.
  -ss SOURCE_SENSOR, --source_sensor SOURCE_SENSOR
                        Source transformation sensor.
  -ts TARGET_SENSOR, --target_sensor TARGET_SENSOR
                        Target transformation sensor.
  -si, --show_images    If true the script shows images.
  -ef EVAL_FILE, --eval_file EVAL_FILE
                        Path to file to read and/or write the evalutation data.
  -ua, --use_annotation
                        If true, the limit points will be manually annotated.
```

danifpdra, 2021-03-19:
Yes but if there file doesn't exist, shouldn't it be created there? Do I have to previously create an empty file?

aaguiar96, 2021-03-19:
> Yes but if there file doesn't exist, shouldn't it be created there? Do I have to previously create an empty file?

The file is automatically created after annotation.

danifpdra, 2021-03-19:
But I have to choose the file name? I thought it was like the calibration where the atom_calibration.json is automatically created...

I will repeat the annotation later.

aaguiar96, 2021-03-19:
> But I have to choose the file name? I thought it was like the calibration where the atom_calibration.json is automatically created...
> 
> I will repeat the annotation later.

Yes, you have. Is just to have the same parameter for input and output. But we can change this also.

danifpdra, 2021-03-19:
I think it would be easier and would become more uniform with the rest of the ATOM framework.

aaguiar96, 2021-03-19:
Hi @danifpdra @miguelriemoliveira and @eupedrosa 

I just committed a new version of the evaluation script.
New features:

- Don't need to press any key everytime we want to add a point. So, the flow is just [click, click, click, ..., c, click, click, ..., c ...]. *c* (denotes for change) is to change the class.
- It is now possible to correct annotations. For each image, if you want to start over the annotation, just click on *d* (denotes for delete), and the annotations will be deleted and the image re-loaded.

@danifpdra can you try it out and give me some feedback please?

eupedrosa, 2021-03-19:
@aaguiar96, Is there any visual hint that shows which class is active?

aaguiar96, 2021-03-19:
> @aaguiar96, Is there any visual hint that shows which class is active?

Yes, the color of the dots.

eupedrosa, 2021-03-19:
But that only tells you which class is active after clicking..
What about a simple print in the terminal that says which class, by name, is active. Or you can write its name directly in the image, a small label in some corner.

eupedrosa, 2021-03-19:
All of this may seem obvious to you @aaguiar96, because you develop it.
But if we want more people to use it we have to make an effort to make the tool "understandable".
Didn't you have any `human-computer interaction` class?

aaguiar96, 2021-03-19:
Hi @eupedrosa 

That makes sense. I'll do that. :)

miguelriemoliveira, 2021-03-20:
Hi @aaguiar96 @eupedrosa and @danifpdra ,

A large discussion. Sorry, I was not online yesterday. 
I will try to go topic by topic.

---------------------------------------------------------------------------------------------
> I would say that for a virtual dataset the results could be better.
> Most of the errors seems to be in the X axis.
> 
> @miguelriemoliveira, @aaguiar96 and @danifpdra, what is you opinion?

I think @aaguiar96 and @danifpdra gave good reasons. We were just trying to get a compete calibration working. Also, Daniela 
was using some noise in the sensor pose first guess, so I am not very concerned with that for now.

BTW, @danifpdra did you create the issue about that?

---------------------------------------------------------------------------------------------

About the procedure designed by @aaguiar96 to annotate the chessboard limits on the images, and the question about automating the procedure.

I see @eupedrosa 's argument of "let's make it easy to use or else people will not use it", but I also think is not so difficult and @aaguiar96 says. Perhaps a tutorial video could be added to the readme.

About automating the procedure: I am generally not in favor of automating an annotation procedure, for several reasons:

1. It is a one shot process. You annotate each dataset once and that's it.
2. Because its a one shot, it is generally not worth the time you would spend automating the procedure.
3. The annotated ground truth should be as accurate as possible, automatic algorithms often have a tradeoff in which a bit of  accuracy is lost.

In this case, its just clicking in the pattern's corners and in some of the borders. I don't think it is that hard. It's boring, but not hard. 

@eupedrosa suggested the following procedure: 

> 1. We detect the pattern (we sould already have that)
Yes we have done it

> 2. Calculate the 4 corners of the pattern (I believe this is also done somewhere else)

Hum, not sure how we could do it accurately. This is not in 3D, its in the image.

> 3. Add intermediate points between edges.

As @aaguiar96 said, sometimes we do not have a straight line due to distortion. @eupedrosa suggested we could 
 use the distortion to automate the shape of the curve which binds two corners together because we have the distortion.

I have a different view: We do not have the distortion. Rather, we have estimated the distortion as part of the parameters estimated during the calibration. We should not use the estimated parameters, which is what we want to evaluate, to create the ground truth which we will then use to evaluate those same parameters.

> 4. profit partying_face

This I completely agree but also don't know how to do.

--------------------------------------------------

So my opinion is that we should not spend a lot of time automating which could be harder than it seems at first glance, but we should improve as much as possible the interface. In the end I think you three converged to something like this, where @aaguiar96 is using suggestions from @eupedrosa and @danifpdra  to improve the interface and make it as easy to use as possible, whilst still manual.

I only have the additional suggestion of making a video gif of the annotation procedure which can be inserted in the readme.

Great work guys (and girl) : - )

aaguiar96, 2021-03-20:
Hi @miguelriemoliveira 

I agree. I'll make a video tutorial soon.
@danifpdra have you had the opportunity to try the script with the new features? Did it get easier?

danifpdra, 2021-03-21:
Hi @aaguiar96, I've just tested it. Much easier and intuitive now!

aaguiar96, 2021-03-22:
Great @danifpdra. Can you please update the README with the new flow of the script? I will make a video soon.

Are we ready to start generating results for the paper? (@danifpdra, @miguelriemoliveira, @eupedrosa)

danifpdra, 2021-03-22:
Yes, I will do that.

miguelriemoliveira, 2021-03-22:
Don't we need #270 solved first?

I would like to have a talk just to try to define the tables ... can we talk wednesday afternoon @danifpdra and @aaguiar96 ?

aaguiar96, 2021-03-22:
> Don't we need #270 solved first?
> 
> I would like to have a talk just to try to define the tables ... can we talk wednesday afternoon @danifpdra and @aaguiar96 ?

Yes and yes. :)

danifpdra, 2021-03-22:
I have a presential meeting from 14h to 15h, so maybe after 15h30 for me to have time to get home.

miguelriemoliveira, 2021-03-22:
OK, wednesday 15h30.

@Eurico Pedrosa ***@***.***> do you want to come as well?

Regards,
Miguel

*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

eupedrosa, 2021-03-22:
Ok.

[1 more reply]

closed 2021-03-31 by aaguiar96