lardemua/atom #270: -nig flag doesn't seem to be affecting LiDAR data

danifpdra, 2021-03-22:
Hi @eupedrosa,

The -nig flag doesn't seem to be affecting LiDAR data.

For example with -nig 0.01 I get the following errors

`Sensor 3dlidar 0.004305327277845061
Sensor hand_camera 1.4203569325689398
Sensor world_camera 1.063220630675486`

And without the -nig flag I get 

`Sensor 3dlidar 0.004314974079740526
Sensor hand_camera 0.7702079446835115
Sensor world_camera 0.5920475494334818`

Is this expected only to affect cameras?

miguelriemoliveira, 2021-03-22:
@eupedrosa can you see this one please?

eupedrosa, 2021-03-22:
Ok, I will take a look.

eupedrosa, 2021-03-22:
@danifpdra, is any of the sensors defined as "anchored"?

danifpdra, 2021-03-22:
@eupedrosa, I don't think so, in config.yml I have:

`anchored_sensor: ""`

This is it, right? I'm not entirely familiar with ATOM framework.

eupedrosa, 2021-03-22:
> This is it, right? I'm not entirely familiar with ATOM framework.

It is correct.

> For example with -nig 0.01 I get the following errors
>
> Sensor 3dlidar 0.004305327277845061 Sensor hand_camera 1.4203569325689398 Sensor world_camera 1.063220630675486
>
> And without the -nig flag I [...]

I have to say that the noisy initial guess is working. Maybe the 3d lidar was favoured by the gods of random and got a 0 in the initial guess. You need to makes more than one test to get better conclusions.

danifpdra, 2021-03-22:
But I tested it several times and this always happened. Can it be because of the pixel and millimetres difference?

eupedrosa, 2021-03-22:
Yes, it can be.
But it should also be expected to have the cameras with an error similar to running the optimization without the -nig flag.

eupedrosa, 2021-03-22:
Are these the parameters that you are using to optimize?
https://github.com/lardemua/atom/blob/dae9751c1658db648195edfa173aec57a2594334/atom_calibration/scripts/calibrate#L467

The optimization may be finishing before reaching the local minimum.
Try somehing like `1e-6` or even `1e-8`

miguelriemoliveira, 2021-03-22:
> Ok, I will take a look.

Thanks.

danifpdra, 2021-03-22:
No, the parameters are already 

`options = {'ftol': 1e-8, 'xtol': 1e-8, 'gtol': 1e-8, 'diff_step': None, 'x_scale': 'jac'}`

eupedrosa, 2021-03-22:
Try a big value for -nig parameter, for example `-nig 1.0` and see what happens.

danifpdra, 2021-03-22:
The errors are:

`Final errors:
Sensor 3dlidar 0.004289071412951691
Sensor hand_camera 6.872915025914711
Sensor world_camera 3.3463363659081486`

eupedrosa, 2021-03-22:
@danifpdra, can you tell me, or point me to where the dataset is? So I can try it?

danifpdra, 2021-03-22:
[dataset_15_03_2021_2.zip](https://github.com/lardemua/atom/files/6181594/dataset_15_03_2021_2.zip)

This is the dataset I've used.

eupedrosa, 2021-03-22:
I runned the following
```bash
rosrun atom_calibration calibrate -json dataset_15_03_2021_2/data_collected.json -nig 0.1
```
And I was able to get
```
Final errors:
Sensor 3dlidar 0.00487091988349726
Sensor hand_camera 0.29026328244471516
Sensor world_camera 0.30067674339040534
```

This was only possible by turning off the `normalizer`. 

@miguelriemoliveira, I need your atention to this. I believe this has something to do with the error of the 3d lidar, which in my opinion is very low.

miguelriemoliveira, 2021-03-22:
I will look into it on Wednesday, sorry : - (

closed 2021-03-29 by eupedrosa

miguelriemoliveira, 2021-03-29:
@danifpdra now you can start with getting the results.

commit db3b47e1 (2021-03-29): Fix noisy initial pose calculation. Closes #270 and #271.

The -nig option now receives 2 values. The first is for translation and
the second for rotation.