lardemua/atom #272: Tables and graphs for atom framework paper

miguelriemoliveira, 2021-03-24:
- [ ] Results:
- [ ]  camera to camera evaluation
- [ ]  opencv stereo

- [ ]   Image with error dispersion in pixels

- [ ]   lidar to camera1 evaluation
- [ ]   lidar to camera2 evaluation
- [ ]   ICP????

- [ ] Characterization

- [ ]   impact number of collections
- [ ]   impact number of incomplete collections
- [ ]   impact number of partial collections ???
- [ ]   impact noise of initial estimate

Waiting for #271

danifpdra, 2021-03-31:
Hi @aaguiar96 and @miguelriemoliveira,

In camera-camera evaluation, how do you compare with OpenCv if all the agrob train datasets have partial detections?

@miguelriemoliveira told me to have one dataset with partial detection and another one without them, but I checked the agrob datasets and both of them have partial detections.

aaguiar96, 2021-03-31:
> In camera-camera evaluation, how do you compare with OpenCv if all the agrob train datasets have partial detections?
> 
> @miguelriemoliveira told me to have one dataset with partial detection and another one without them, but I checked the agrob datasets and both of them have partial detections.

I @danifpdra 

There was one dataset of agrob that had some non-partial collections. And that was the one we used.
But why do you need agrob datasets? I think you can test with `mmtbot`.

I will just adapt it to hand eye case and is good to go.

danifpdra, 2021-03-31:
I don't need the agrob datasets, I just want to make the mmtbot datasets uniform with the agrob datasets.

According to this table you always have partial detections.

[image]

aaguiar96, 2021-03-31:
> According to this table you always have partial detections.

Yes, I had. But I only use non-partial collections for the opencv calibration. This is a limitation of opencv.

danifpdra, 2021-03-31:
I know that, that's why I was asking. I thought that I was supposed to have a dataset with partial detections and another one without.

aaguiar96, 2021-03-31:
For a real scenario is really difficult to have a dataset only with non-partial collections. But in simulation, you can try! :)

aaguiar96, 2021-03-31:
I just corrected the opencv evaluation script for python3 and executed it with a `mmtbot` dataset. Here's what you should get:

```
➜  catkin_ws_ros1 rosrun atom_evaluation cv_stereo_calib.py -json ~andresaguiar/Documents/atom_datasets/mmtbot_cam2cam/dataset_11_03_2021/atom_calibration.json -rc hand_camera -lc world_camera -si
Partial detection removed: label from collection 0, sensor hand_camera
Partial detection removed: label from collection 1, sensor world_camera
Partial detection removed: label from collection 10, sensor world_camera
Partial detection removed: label from collection 11, sensor world_camera
Partial detection removed: label from collection 12, sensor world_camera
Partial detection removed: label from collection 13, sensor hand_camera
Partial detection removed: label from collection 14, sensor hand_camera
Partial detection removed: label from collection 15, sensor hand_camera
Partial detection removed: label from collection 16, sensor hand_camera
Partial detection removed: label from collection 18, sensor hand_camera
Partial detection removed: label from collection 19, sensor world_camera
Partial detection removed: label from collection 2, sensor world_camera
Partial detection removed: label from collection 20, sensor world_camera
Partial detection removed: label from collection 21, sensor world_camera
Partial detection removed: label from collection 22, sensor hand_camera
Partial detection removed: label from collection 23, sensor hand_camera
Partial detection removed: label from collection 24, sensor hand_camera
Partial detection removed: label from collection 25, sensor hand_camera
Partial detection removed: label from collection 26, sensor hand_camera
Partial detection removed: label from collection 5, sensor hand_camera
Partial detection removed: label from collection 6, sensor hand_camera
Partial detection removed: label from collection 7, sensor hand_camera

Using 5 collections.

---------------------
 Starting stereo calibration ...

---------------------
 Done!

------
Calibration results:
------

K_left [[8.19961233e+02 0.00000000e+00 5.35329947e+02]
 [0.00000000e+00 1.01284493e+03 1.83692080e+02]
 [0.00000000e+00 0.00000000e+00 1.00000000e+00]]
D_left [[ 0.2411337 ]
 [-8.74041987]
 [ 0.09070219]
 [ 0.10736554]
 [41.43056823]]
K_right [[3.38910592e+03 0.00000000e+00 1.74818877e+03]
 [0.00000000e+00 3.18719544e+03 1.15192930e+03]
 [0.00000000e+00 0.00000000e+00 1.00000000e+00]]
D_right [[-0.25388015]
 [ 1.41194698]
 [ 0.04102072]
 [ 0.06885467]
 [-1.49176662]]
R [[ 0.65596347  0.30431845 -0.69072585]
 [-0.3081252   0.94336023  0.12300545]
 [ 0.68903612  0.13214296  0.71257803]]
T [[-0.66791163]
 [-1.45801716]
 [ 3.01130303]]
E [[-0.07676814 -3.03341022 -1.40935768]
 [ 2.43552003  1.00465489 -1.6040457 ]
 [ 1.16220641 -0.18637974 -1.08924692]]
F [[ 1.56446918e-07  5.00457864e-06  1.35200011e-03]
 [-5.27781500e-06 -1.76250242e-06  5.99931164e-03]
 [-2.22085034e-03 -5.67654405e-03  1.00000000e+00]]
Saving the json output file to /home/andresaguiar/Documents/atom_datasets/mmtbot_cam2cam/dataset_11_03_2021/cv_calibration.json, please wait, it could take a while ...
Completed.
```

danifpdra, 2021-04-06:
> I just corrected the opencv evaluation script for python3 and executed it with a `mmtbot` dataset. Here's what you should get:
>
> ```
> ➜  catkin_ws_ros1 rosrun atom_evaluation cv_stereo_calib.py -json [...]

Hi @aaguiar96 ,

After I get this, what are the results that should go to the tables in the paper?

aaguiar96, 2021-04-06:
Hi @danifpdra 

As you see, this script outputs a json file:

```
Saving the json output file to /home/andresaguiar/Documents/atom_datasets/mmtbot_cam2cam/dataset_11_03_2021/cv_calibration.json, please wait, it could take a while ...
Completed.
```

This `json` has the same format as the ones exported by ATOM, so all you have to do is use it in the camera-to-camera evaluation script as you already did for ATOM.

You should use the same train and test datasets to compare fairly with ATOM.

danifpdra, 2021-04-06:
Hi @aaguiar96 ,

I did that but the errors were huge, did I do something wrong?

`rosrun atom_evaluation camera_to_camera_evalutation.py -train_json $ATOM_DATASETS/train_dataset_1/cv_calibration.json -test_json $ATOM_DATASETS/test_dataset/data_collected.json -ss hand_camera -ts world_camera`

```
---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
  0      -      857.0737 2657.2286  120.2267  579.9042 1222.3263  18.9259  
  1      -      862.8784 1947.0261  131.9373  740.6625  586.0206  16.5589  
  2      -      870.6371 1399.5607  171.9327  693.3196  624.0957  23.4468  
  3      -      240.5548  634.6677  164.3765  767.3012  882.0238  21.5654  
  4      -      240.5548  634.6677  164.3765  767.3012  882.0238  21.5654  
  5      -      403.3754 1936.5011  157.5742  691.4149  990.1728  21.1384  
  6      -      442.5352 1957.4686  116.9558  805.6797 1234.5027  17.4034  
  7      -      401.8128 2072.0340  46.5281   783.9833 1002.6496  10.0969  
  8      -      101.6084  997.6073  91.2029   876.0718 1212.3864  14.1328  
  9      -      110.8073 1154.1672  120.0598  736.8698 1051.8256  13.7899  
 10      -     1690.5027 1372.1904  121.1598  693.4060  745.2836  12.9661  
 11      -     1556.9306 2953.1672  248.9739  735.0281 1064.1954  25.9987  
 12      -      964.6402 5164.1388  100.9087  755.0038  467.5796  11.0975  
 13      -      815.5038 4650.9793  83.5275   801.9755  378.1867   7.5991  
 14      -     1116.3291  874.8666  132.9699  618.7443  608.1415  14.7277  
 15      -      148.5068  797.7834  161.3553  768.2578  902.7670  21.9993  
 16      -      939.0749  903.7896  131.1507  754.0781 1323.4161  15.7779  
 17      -      583.6276 1791.6996  136.9372  916.4992 1411.4529  17.6778  
 18      -      113.5350 2639.0200  130.0946  857.2329  214.3969  18.9293  
 19      -     1157.1569 2258.3384  121.0630  949.5642 4384.7288  19.7004  
 20      -      320.5286  918.1955  152.3316  821.8198 1546.3805  27.6625  
 21      -     1140.4892 2081.4994  131.2562  786.5609 4598.0203  32.0441  
 22      -     3773.7423 1971.5002  302.4529 1018.2560 2316.5807  37.2241  
 23      -     1461.8470 5234.6551  117.3641  507.6422 1496.5091  20.6828  
 24      -     1787.1112 5457.0750  118.3061  480.7469 1664.9064  24.4473  
 25      -      788.8580 6014.6073  136.8881  496.4807 1394.2971  24.8214  
 26      -      680.9074 5540.7017  169.0240  689.5400 1150.5927  22.3577  
---------------------------------------------------------------------------
 All 2185.7174  826.4440 2301.4569  924.5902 2572.1884   1.3095    0.3454  
---------------------------------------------------------------------------
```

aaguiar96, 2021-04-06:
Hi @danifpdra 

I don't see anything wrong in the process. This worked well for agrob, but maybe it needs some adaptation for the hand-eye scenario. Can you send me the datasets and I will try to see what's missing?

Just to be sure, @miguelriemoliveira and @eupedrosa. Does this make sense? I.e., in theory, we can use OpenCV stereo calibration even if one of the cameras is moving right? I'm a little bit confused right now.

aaguiar96, 2021-04-06:
Thanks @danifpdra 

@miguelriemoliveira and @eupedrosa, regarding my previous question: OpenCV also has a hand-eye calibration toolbox. Should we use it?

Correct me if I am wrong, but I think the OpenCV stereo calibration uses all collections to find the relative homogeneous transformation between the cameras. I don't think it is prepared for a moving camera...

eupedrosa, 2021-04-06:
> @miguelriemoliveira and @eupedrosa, regarding my previous question: OpenCV also has a hand-eye calibration toolbox. Should we use it?

The hand-eye calibration toolbox from OpenCV only calculates the position of the camera on the end-effector using a fixed pattern. If I am not mistaken, it will not work with a moving pattern. 

However, in the hand-eye paper we already test our solutiong agains the one implemented by openCV.

> Correct me if I am wrong, but I think the OpenCV stereo calibration uses all collections to find a single relative homogeneous transformation between the cameras. I don't think it is prepared for a moving camera...

I believe your are correct. It calculates a transformation of fixed cameras.

> this is why we had to implement the atomicTfFromCalibration routine that extracts the atomic transformations marked for calibration from the relative homogeneous transformation between sensors. Right?

I am not sure about this one, @miguelriemoliveira was the one that did that.

aaguiar96, 2021-04-06:
> I believe your are correct. It calculates a transformation of fixed cameras.
> 
> >

Ok, so it is not suitable for calibrating a moving camera with a fixed camera...

aaguiar96, 2021-04-06:
> I am not sure about this one, @miguelriemoliveira was the one that did that.

I was the one that implemented that routine. My question was that if we need to do this for OpenCV, then it means that opencv computes a sensor-to-sensor transformation. Since one sensor is moving, the calibration will fail.

eupedrosa, 2021-04-06:
> I was the one that implemented that routine. 

My mistake.

Yes, OpenCV computes a sensor-to-sensor transformation and will break with a moving sensor.

miguelriemoliveira, 2021-04-06:
`@miguelriemoliveira and @eupedrosa, regarding my previous question: OpenCV also has a hand-eye calibration toolbox. Should we use it?`

Right, we cannot use opencv for this case. It just does not work. Actually, nor will the hand eye case work because it will only calibrate a moving camera but not the fixed. 
This is what we call the **joint-hand-base use case** in the T-RO paper which, to the best of my knowledge, nobody except use is able to solve.

Sorry @danifpdra forgot about that ...

miguelriemoliveira, 2021-07-12:
This is complete I think. Only thing missing are the atlascar2 results which will have to be recomputed but #309 is tracking that. Closing.

closed 2021-07-12 by miguelriemoliveira