lardemua/atom #273: How to launch several atoms

miguelriemoliveira, 2021-03-30:
@danifpdra wants to run several tests in batch.

@eupedrosa how did you run your tests for the sensitivity to the initial estimate?

I can put an example of a python script that calls the calibrate in sequence, but perhaps @eupedrosa has a more advanced technique ...

eupedrosa, 2021-03-31:
My advance technique is a bash script with greps and stdout redirects. I usualy change the python script to output the data that I want with a prefix that I can easily filter with `grep`.

miguelriemoliveira, 2021-03-31:
Thanks @eupedrosa 

Hum ... if that's the case then I prefer to do it with  python. @danifpdra I started yesterday night and will continue tonight, so give me today to come up with an example ok?

danifpdra, 2021-03-31:
Ok @miguelriemoliveira

miguelriemoliveira, 2021-04-01:
Hi @danifpdra ,

I created a python script which runs a set of commands (could the several executions of calibrate, could be anything).

I think this way you can batch run several experiments.

Take a look and let me know if it works well:

https://github.com/lardemua/atom/blob/noetic-devel/atom_evaluation/scripts/batch_execution

miguelriemoliveira, 2021-04-02:
Hi @danifpdra ,

I could not resist and implemented a functionality to read the commands to be executed in batch from a yaml file.

Just run something like:

```
roscd atom_evaluation && ./batch_execution -v -f batch_example.yml
```

The commands you want to execute are defined in the yaml file, e.g.:

```yaml
# Auxiliary variables. Used to render other variables.
dataset_path: '/home/mike/datasets/mmtbot/train_dataset'
cmd_prefix: 'rosrun atom_calibration calibrate'

# Batch variables. These are read and used for batch execution.
output_folder: '{{ dataset_path }}/results'

batches:
    - name: "nig0.01_0.020"
      cmd: "{{ cmd_prefix }} -json {{ dataset_path }}/data_collected.json -csf 'lambda name: int(name)<1' -nig 0.01 0.020 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro'

    - name: "nig0.03_0.020"
      cmd: "{{ cmd_prefix }} -json {{ dataset_path }}/data_collected.json -csf 'lambda name: int(name)<1' -nig 0.03 0.020 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro'
```

We are using jinja2 for rendering this template file.

Can you try it out and see if it works?

miguelriemoliveira, 2021-04-02:
Implemented recursive variable substitution. Now its possible to nest variables in the configuration file. 
An example:

```yaml
# Auxiliary variables. Used to render other variables.
home: '/home/mike'
dataset_path: '{{ home }}/datasets/mmtbot/train_dataset'
cmd_prefix: 'rosrun atom_calibration calibrate -json {{ dataset_path }}/data_collected.json '

# Batch variables. These are read and used for batch execution.
output_folder: '{{ dataset_path }}/results'

batches:
    - name: "nig0.01_0.020"
      cmd: "{{ cmd_prefix }} -csf 'lambda name: int(name)<1' -nig 0.01 0.020 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro'

    - name: "nig0.03_0.020"
      cmd: "{{ cmd_prefix }} -csf 'lambda name: int(name)<1' -nig 0.03 0.020 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro'
```

danifpdra, 2021-04-08:
Hi @miguelriemoliveira ,

Can you help me figuring out these loops?  https://github.com/lardemua/atom/blob/noetic-devel/atom_evaluation/scripts/batch_example.yml

I get the error 
`jinja2.exceptions.UndefinedError: 'item' is undefined`

But I did this according to the ansible documentations (https://docs.ansible.com/ansible/latest/user_guide/playbooks_loops.html)

miguelriemoliveira, 2021-04-08:
Hi @danifpdra ,

better to leave the batch_example.yml as was before so it can be used by others, and instead create one for you, which should not go into atom BTW, I think it should go into mmtbot.

I don't think I am using ansible. I will try to get an example with loops working and get back to you.

miguelriemoliveira, 2021-04-16:
Closing this one. Implemented and working.

closed 2021-04-16 by miguelriemoliveira