lardemua/atom #276: Problems with incomplete collections

danifpdra, 2021-04-01:
Hi @miguelriemoliveira ,

How does ATOM work with incomplete collections? 

When running:

`rosrun atom_calibration calibrate -json /home/daniela/datasets/dataset_31_03_2021/data_collected.json 'lambda name: int(name) < 50' -nig 0.05 0.025 -ss 1`

I just got the error:

`Removing collection 1 -> pattern was not found in sensor hand_camera (incomplete collection).
Traceback (most recent call last):
  File "/home/daniela/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 543, in <module>
    main()
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 112, in main
    dataset = filterCollectionsFromDataset(dataset, args)  # filter collections
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 361, in filterCollectionsFromDataset
    for collection_key, collection in dataset['collections'].items():
RuntimeError: dictionary changed size during iteration`

miguelriemoliveira, 2021-04-01:
Hi @danifpdra ,

I think it was a problem of having to adjust this particular code to python3. I've done it.

Cannot test because you did not give me the **dataset_31_03_2021**. Can you test and close if its fixed?

danifpdra, 2021-04-01:
It works when removing the incomplete collections. If I use the -uic flag I get the following error:

`Removing collection 8: at least one detection by a camera should be present.
Traceback (most recent call last):
  File "/home/daniela/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 543, in <module>
    main()
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 112, in main
    dataset = filterCollectionsFromDataset(dataset, args)  # filter collections
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 397, in filterCollectionsFromDataset
    for collection_key, collection in dataset['collections'].items():
RuntimeError: dictionary changed size during iteration`

miguelriemoliveira, 2021-04-01:
I will try tonight. Give me the dataset please.

danifpdra, 2021-04-01:
Here it is: [dataset_31_03_2021.zip](https://github.com/lardemua/atom/files/6246239/dataset_31_03_2021.zip)

miguelriemoliveira, 2021-04-01:
I think now its solved. Tested and it works. Let me know if you need anything else.

Also I would like to discuss with you the organization of the results in a new repository or some shared folder. Can we talk 15 ou 20 mins? Tell me when you have time in the next days (or next week).

closed 2021-04-01 by miguelriemoliveira

danifpdra, 2021-04-02:
I don't have anything scheduled for the next few days. Give me a call whenever you want to meet.

commit c1a038a6 (2021-04-01): Adapting incomplete collection filter to python3 (fixes #276).
commit f7182a8c (2021-04-01): Adapting incomplete collection filter to python3 (fixes #276).
commit 8745fef1 (2021-04-01): Adapting filter that removes collections if not detection by camera exists to python3 (fixes #276).