lardemua/atom #277: Problems with camera-lidar evaluation for the hand eye case

danifpdra, 2021-04-06:
Hi @aaguiar96,

When I use the atom_evaluation range_sensor for the 3dlidar and world_camera, it works fine, but when I use it for the 3dlidar and the hand_eye camera this happens:

`rosrun atom_evaluation range_sensor_to_camera_evaluation.py -train_json $ATOM_DATASETS/train_dataset_1/atom_calibration.json -test_json $ATOM_DATASETS/test_dataset/data_collected.json -ss 3dlidar -ts hand_camera -si -ef $ATOM_DATASETS/annotation/15_3__11_3_hand.json -ua
`

[image]

Can you help me? Are we having the same issue here because of the moving camera?

aaguiar96, 2021-04-06:
Hi @danifpdra 

I thought it would work since I included the fix made by @miguelriemoliveira in the camera-to-camera evaluation script of changing the test dataset with the optimized atomic transformations on the train dataset.

So, at first sight I don't see what's happening, but I will take a look.

aaguiar96, 2021-04-06:
Hi @danifpdra 

Can you send me the annotations and the test dataset please?

danifpdra, 2021-04-06:
This is the train and test dataset and the annotations.

[datasets.zip](https://github.com/lardemua/atom/files/6266296/datasets.zip)

[annotation.zip](https://github.com/lardemua/atom/files/6266299/annotation.zip)

aaguiar96, 2021-04-06:
The annotations are not coherent with the dataset. Can you check if the is the right command?

```
rosrun atom_evaluation range_sensor_to_camera_evaluation.py -train_json /home/andresaguiar/Documents/atom_datasets/mmtbot_dataset_11_03_2021/atom_calibration.json -test_json /home/andresaguiar/Documents/atom_datasets/mmtbot_dataset_15_03_2021_2/data_collected.json -ss 3dlidar -ts hand_camera -si -ef /home/andresaguiar/Documents/atom_datasets/mmtbot_dataset_15_03_2021_2/15_3__11_3.json
```

miguelriemoliveira, 2021-04-06:
Hum ... not sure about this one. I mean, one thing are the evaluations (those should work for moving sensors). Another thing is a calibration method such as opencv, which does not support moving sensors ...

Makes sense @aaguiar96 ?

aaguiar96, 2021-04-06:
> Hum ... not sure about this one. I mean, one thing are the evaluations (those should work for moving sensors). Another thing is a calibration method such as opencv, which does not support moving sensors ...
> 
> Makes sense @aaguiar96 ?

Yes @miguelriemoliveira. This should work. What I was saying is that this seems to the problem we had for the camera-to-camera evaluation script for hand-eye. However, I made that fix also for range-to-camera script. I will check if something is missing.

miguelriemoliveira, 2021-04-06:
alright. Let me know if I can help...

aaguiar96, 2021-04-07:
Hello @danifpdra and @miguelriemoliveira 

I am having trouble finding the issue. I recorded new datasets and tested other calibrations and the error persists.

This may mean that the problem is on the evaluation script (?). However, I reduced the problem to the simple projection of the 3dlidar points into the image, and it fails...

Also, I used the same train and test dataset to eliminate the possibility to be some error due to different hand poses between datasets.

So, what I am testing now is simply this code (where the `rangeToImage` routine works for the non hand-eye case):

```
    from_frame = test_dataset['calibration_config']['sensors'][target_sensor]['link']
    to_frame = test_dataset['calibration_config']['sensors'][source_sensor]['link']
    od = OrderedDict(sorted(test_dataset['collections'].items(), key=lambda t: int(t[0])))
    for collection_key, collection in od.items():
        # ---------------------------------------
        # --- Range to image projection
        # ---------------------------------------
        selected_collection_key = list(test_dataset['collections'].keys())[0]
        vel2cam = atom_core.atom.getTransform(from_frame, to_frame,
                                              test_dataset['collections'][selected_collection_key]['transforms'])
        pts_in_image = rangeToImage(collection, test_json_file, source_sensor, target_sensor, vel2cam)

        # ---------------------------------------
        # --- Drawing ...
        # ---------------------------------------
        filename = os.path.dirname(test_json_file) + '/' + collection['data'][target_sensor]['data_file']
        print (filename)
        image = cv2.imread(filename)
        if show_images is True:
            for idx in range(0, pts_in_image.shape[1]):
                image = cv2.circle(image, (int(pts_in_image[0, idx]), int(pts_in_image[1, idx])), 5, (255, 0, 0), -1)

            cv2.imshow("Lidar to Camera reprojection - collection " + str(collection_key), image)
            cv2.waitKey()

        continue
```

Let me know if you have some ideas.

[image]

miguelriemoliveira, 2021-04-07:
Hi @aaguiar96 ,

I will try to look into it this afternoon or tomorrow and get back to you.

miguelriemoliveira, 2021-04-07:
Hi @aaguiar96 ,

I was thinking about this now. I looked at your code and I think using the selected collection key does not make sense.

Note that we have in each **dataset['collections'][collection_key]['transforms']** all the transforms in the system, both the ones being optimized and the others. So we have a pool of transforms per collection.

We use the selected_collection_key when we are sure that the pools of transforms are equal for all collections, which would make it the same to use collection_key X or Y.

When we have the eye in hand case we have dynamic transformations which change from collection to collection (the motion of the arm), and therefore cannot assume that the transformations pools are the same. If they are not the same, we cannot use the selected_transformation_key. 

Try replacing 

```
vel2cam = atom_core.atom.getTransform(from_frame, to_frame,
                                              test_dataset['collections'][selected_collection_key]['transforms'])
```

with 

```
vel2cam = atom_core.atom.getTransform(from_frame, to_frame,
                                              test_dataset['collections'][collection_key]['transforms'])
```

This is what I did when I looked into the camera to camera evaluation script.

aaguiar96, 2021-04-08:
Thanks, @miguelriemoliveira, it makes sense. And it works! :) 

@danifpdra can you try it, and if it works, close this issue, please?

miguelriemoliveira, 2021-04-08:
Great!

danifpdra, 2021-04-08:
Yes, this is working now. Thanks.

closed 2021-04-08 by danifpdra