lardemua/atom #278: Show/Hide functionality for set_initial_estimate

MiguelDRPina, 2021-04-09:
Hello, @miguelriemoliveira !

Can you add a functionality for the set_initial_estimate to show/hide the sensor when it's asked on the panel plugin?

miguelriemoliveira, 2021-04-09:
Hi @MiguelDRPina 

I already have a skeleton for the communication. We will use a service to the name

/set_initial_estimate/<sensor>/set_visible

and the service type is a std_srv SetBool
http://docs.ros.org/en/api/std_srvs/html/srv/SetBool.html

For testing, I used with

   rosservice call /set_initial_estimate/world_camera/set_visible "data: false"

Everything works but the hiding of the markers, which I will work to solve.

miguelriemoliveira, 2021-04-10:
Hi @MiguelDRPina ,

It was tricky but I was able to make it work. Check this:

https://youtu.be/A5gc-yb-Emg

I will assign this issue to you so you can integrate with the rviz pannel.

MiguelDRPina, 2021-04-10:
Hi @miguelriemoliveira ,
Thank you, I will integrate it with the panel first thing tomorrow.

miguelriemoliveira, 2021-04-11:
Hi @MiguelDRPina ,

sorry about the change, but because of #279 which defines more functionalities I decided to change the service that we use.

Its is now the **SetSensorInteractiveMarker.srv**, which allows to set the visibility and the scale (and anything else we may need in the future).

https://github.com/lardemua/atom/blob/noetic-devel/atom_msgs/srv/SetSensorInteractiveMarker.srv

This is is the new package atom_msgs and to use I do as follows:

```
rosservice call /set_initial_estimate/hand_camera/set_sensor_interactive_marker "visible: true
scale: 0.2
```

notice that now we have two fields in the srv request, visible and scale. If you have problems adjusting let me know.

MiguelDRPina, 2021-04-11:
Hi, @miguelriemoliveira ,
Ok, I will keep that in mind, but I am still struggling with implementing it with the rviz panel, because I am trying to use a service from another package (trying to use the SetSensorInteractiveMarker service from the atom package in the atom_rviz_plugin package).

miguelriemoliveira, 2021-04-11:
Hi @MiguelDRPina 

you can push the code and ask for help. It could be from my side, i.e., the atom_msgs package is missing something ...

MiguelDRPina, 2021-04-12:
Hi @miguelriemoliveira ,

The code is committed, but for the service there's not much to see there, because what I have tried so far haven't worked yet. I tried following the [tutorial](http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28c%2B%2B%29) for the client but am having problems creating the client. My questions are: 

- Am I supposed to add something to the CMakeLists file (or any other file)? 
- Basing it on the tutorial, what exactly am I supposed to write within the < > in this line:
```
ros::ServiceClient client = n.serviceClient<beginner_tutorials::AddTwoInts>("add_two_ints");
```
Because whatever I put there I get an error message saying it does not belong there.

miguelriemoliveira, 2021-04-12:
Hi @MiguelDRPina ,

I will try to work with you a little bit, but I have classes until 18h. Can we meet tonight?

MiguelDRPina, 2021-04-12:
Hi @miguelriemoliveira ,
Yes, we can meet tonight at any hour after 21h30/22h, if that works for you. But if you think it's too late, we can schedule for another time, no problem.

miguelriemoliveira, 2021-04-12:
Hi @MiguelDRPina ,

so I was able to compile no problems. Can you compile?

MiguelDRPina, 2021-04-12:
I can, because the code for calling the service is not there. I have tried by following the tutorial as I said earlier (https://github.com/lardemua/atom/issues/278#issuecomment-817723207  but with no success.

miguelriemoliveira, 2021-04-12:
Hi @MiguelDRPina ,

I think I solved it. Give it a try and if you still need we meet tonight.

closed 2021-04-13 by MiguelDRPina