lardemua/atom #280: Integration with rviz panel - set initial estimate - Buttons for save and reset

miguelriemoliveira, 2021-04-11:
Hi @MiguelDRPina

About the buttons for saving and reset, there is a topic **/set_initial_estimate/feedback**
on which rviz publishes a msg everytime the user presses either the save or the reset button.

You can listen to the topic using

`rostopic echo /set_initial_estimate/feedback`

and then pressing the buttons (right click the sphere of any contro and a menu pops upl).

Doing this you get **for when the save button is pushed** a message like this:

```yaml
header: 
  seq: 0
  stamp: 
    secs: 822
    nsecs: 109000000
  frame_id: "ee_link"
client_id: "/rviz/MoveSensors-InteractiveMarker"
marker_name: "hand_camera"
control_name: ''
event_type: 2
pose: 
  position: 
    x: -0.019999999552965164
    y: 0.05000000074505806
    z: 0.07000000029802322
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
menu_entry_id: 1
mouse_point: 
  x: -0.07335437089204788
  y: 0.07233176380395889
  z: 0.11905686557292938
mouse_point_valid: True
---
```

Doing this you get **for when the reset  button is pushed** a message like this:

```yaml
header: 
  seq: 0
  stamp: 
    secs: 868
    nsecs: 649000000
  frame_id: "ee_link"
client_id: "/rviz/MoveSensors-InteractiveMarker"
marker_name: "hand_camera"
control_name: ''
event_type: 2
pose: 
  position: 
    x: -0.019999999552965164
    y: 0.05000000074505806
    z: 0.07000000029802322
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
menu_entry_id: 2
mouse_point: 
  x: -0.03277934342622757
  y: 0.06772848218679428
  z: 0.1465941220521927
mouse_point_valid: True
---
```

All you have to do is to send a message similar to the one you see above when your save and reset buttons are pressed.

MiguelDRPina, 2021-04-14:
Hi @miguelriemoliveira ,
I already have the code for publishing the messages when the save/reset buttons are pressed. However, I am not sure what to put on some of the message parameters. For example, the mouse_point coordinates.
Also, how do I get the marker's pose?

miguelriemoliveira, 2021-04-14:
Its not used. Just put 0 for all. Only the menu entry is used.

MiguelDRPina, 2021-04-14:
None of other msg parameters is used? I assume at least the marker_name is used, no? Or by pressing the save/reset button does it save/reset for all sensors?

MiguelDRPina, 2021-04-14:
I think it's working fine now.

closed 2021-04-14 by MiguelDRPina

miguelriemoliveira, 2021-04-14:
Not even that one i think.

MiguelDRPina, 2021-04-14:
I got it working. The client_id, marker_name and event_type also seemed to be needed.
From issue #279, the only functionality left is the micro movement of the marker through panel.

miguelriemoliveira, 2021-04-14:
Hi @MiguelDRPina ,

great news. Job's not finished though. You have to record a demo video and create a section in your Readme (to be imported later on to the atom readme) in which you explain how to operate the panel.

I am not sure that functionality should have a high priority. Perhaps we should instead move on to the functionalities of the data_collector. What is your opinion?

miguelriemoliveira, 2021-04-14:
Ok, so lets meet tomorrow 14h? for me to set you up with the data collector ok?

@eupedrosa can you jump in?

eupedrosa, 2021-04-14:
Hi @miguelriemoliveira. Yes I can.

MiguelDRPina, 2021-04-14:
14h works for me too, talk to you both tomorrow.