lardemua/atom #281: 3D plot surfaces of calibration error vs initial estimate don't make sense

danifpdra, 2021-04-13:
Hi @miguelriemoliveira,

I used batch execution to calibrate the mmtbot setup with initial estimates from 0-2m in translation and 0-40 degrees in rotation. I don't think these results make much sense. Can you tell me your opinion?

[image]
[image]
[image]
[image]

danifpdra, 2021-04-13:
Hi @miguelriemoliveira,

The batch execution got stuck on 

`rosrun atom_calibration calibrate -json /home/daniela/datasets/train_dataset_1/data_collected.json  -csf 'lambda name: int(name)<1' -nig 0.10 0.30 -ss 1`

for like 5 hours. I don't think this is very normal. I've tried to run the same command separately and it finished really quickly.

danifpdra, 2021-04-14:
New data: 

[image]
[image]
[image]
[image]

danifpdra, 2021-04-14:
More resultion: distance and angle 0.5 to 0.5, 21 points each.

I can only run the batch execution with the calibrate file options like this:

`options = {'ftol': 1e-4, 'xtol': 1e-4, 'gtol': 1e-4, 'diff_step': None, 'x_scale': 'jac'}`

@miguelriemoliveira , do you have any idea why?

[image]
[image]
[image]
[image]

danifpdra, 2021-04-14:
Even more resolution. From 0 to 0.75, 50 points each 

[image]
[image]
[image]
[image]

danifpdra, 2021-04-14:
Hi @miguelriemoliveira ,

The file to generate the plots is here: https://github.com/miguelriemoliveira/mmtbot/blob/c8f9ba72c1b2e50a737eea8e8d39cae80ee515a9/mmtbot_calibration/scripts/mmtbot%20analysis.ipynb

danifpdra, 2021-04-14:
This file also generates the commands to a txt file that you can copy to the batch file. You just need to choose the initial and final value and the number of points for each vector.

miguelriemoliveira, 2021-04-15:
> Hi @miguelriemoliveira,
>
> The batch execution got stuck on
>
> `rosrun atom_calibration calibrate -json /home/daniela/datasets/train_dataset_1/data_collected.json -csf 'lambda name: int(name)<1' -nig [...]

Hi @danifpdra ,

I am trying to debug this,  but its very hard because your **batch_mmtbot,yml** is very large, its running now for 10 minutes and still going...

By the way, you did not say but I assume the command is:

`./batch_execution -v -f batch_mmtbot.yml`

My question is: do you have a smaller example where it crashes? I mean, a smaller batch config file? Otherwise its nearly impossible to debug if every new try takes 10 or 15 minutes...

miguelriemoliveira, 2021-04-15:
I think I finally got stuck here:

```
Batch38, executing command:
rosrun atom_calibration calibrate -json /home/mike/datasets/mmtbot/train_dataset/data_collected.json  -csf 'lambda name: int(name)<1' -nig 0.00 0.58 -ss 1
```

The parameters are not the ones you have, so I suspect it has nothing to do with the command parameters, rather with something in the batch...

danifpdra, 2021-04-15:
Yes, this is what happens to me. But I have run a batch with 2500 calibration and it worked well as you have seen so I don't know what is happening.

miguelriemoliveira, 2021-04-15:
Hi @danifpdra ,

I think I may have solved it. Can you try it please?

Also, I could try to make a batch execution which would run commands in parallel, but I am not really sure if this is very high priority. What do you think?

danifpdra, 2021-04-15:
I will try it, yes.

I don't think it is important. It is working the way it is, just might take longer but it's no big deal.

miguelriemoliveira, 2021-04-15:
OK, lets forget about it for now. About the fix, I am now running trial 69 of a 10-8 setting and going fine ...

danifpdra, 2021-04-15:
I will try to run the big calibration with the 10e-8 and see what happens. That way we can even see the difference in the surfaces of that.

miguelriemoliveira, 2021-04-15:
Right now I am running trial 453 and still going. I think that problem is solved.

danifpdra, 2021-04-15:
Hi, plots for the same distances and angles with 10e-8:

[image]
[image]
[image]
[image]

miguelriemoliveira, 2021-04-16:
Hi @danifpdra ,

tell me your opinion about these new results.

Mine is that they look good. I would say no need to get more results of this kind...

I will try to beautify the graphs a bit more and then ask for your opinion.

danifpdra, 2021-04-16:
Ok. I've updated the code for plotting these surfaces: https://github.com/miguelriemoliveira/mmtbot/blob/c8f9ba72c1b2e50a737eea8e8d39cae80ee515a9/mmtbot_calibration/scripts/mmtbot%20analysis.ipynb.

miguelriemoliveira, 2021-04-16:
Ok, thanks, but the new version generates the images above?

danifpdra, 2021-04-16:
Yes, it does.

eupedrosa, 2021-04-16:
Wouldn't it be better to use something like a heat map? Or contour map ?

Although these graphics are interesting,  and somewhat expeted. Their non-smothness is not very appealling.

miguelriemoliveira, 2021-04-16:
Yes, also something else: we are using two degrees of freedom (height and color) to display a single variable.

@danifpdra , can you try something like the matriz de risco that we see for the pandemic situation? Basically a top down view.

Also, the colors should perhaps be more intuitive: a green (good) to red(bad) colormap perhaps is more suited?

danifpdra, 2021-04-19:
Is this better?

[image]

miguelriemoliveira, 2021-04-19:
Hi @danifpdra ,

much better! Nice work.

For a paper in a top journal we still need to improve the following:

What do the axes represent (including units)? 

I like the one on the left, but the xtick tics should perhaps be similar. You have 5 values (0 0.2 0.4 0.6 and 0.8 ) in x and many more on y. Can you fix it?

The figure on the right requires a legend that explains what is green and what is red.

Can you improve with these tips? Think about a top publication and how every little detail should be looked after.
When you have new graphs then we can discuss and select one of them to insert in the paper.

danifpdra, 2021-04-19:
Is this better? Do you prefer the matlab (previous plot) or python (this one) plots ?

[image]

danifpdra, 2021-04-19:
This is the python version

[image]
[image]
[image]
[image]

miguelriemoliveira, 2021-04-19:
Hi @danifpdra ,

this is getting much better. I would say python, since its completely integrated into atom.

Some more comments:

In the paper the title of the graphs should be removed.
Initial estimate distance -> Linear distance from initial estimate to ground truth (meters)
Initial estimate angle -> Angular distance from initial estimate to ground truth (degrees)
Time -> Appears on top of the zticks
World camera error -> Error (pixels) -> the legend will refer which camera the graph is for
Hand camera error -> Error (pixels) -> the legend will refer which camera the graph is for
Lidar camera error -> Error (meters) -> the legend will refer which sensor the graph is for

I like the graphs on the left more, and you?

danifpdra, 2021-04-19:
Is this what you meant?

Yes, I agree.

[image]

miguelriemoliveira, 2021-04-19:
Yes, looks much better than at the start of the issue. Congratullations.

I think the axis labels have the correct text now but its too large, so its better to add symbols and then refer to them in the figure caption. 

So for example you can use

l (meters) and then in the caption explain what l is
a (degrees) and then in the caption explain what a is

danifpdra, 2021-04-20:
I insert the plots in the paper. Can you check if it's alright?

miguelriemoliveira, 2021-04-20:
Hi @danifpdra ,

hum, not just yet, because the vertical axis legend is colliding with the ztick values.

Also, for reference, never have in a graph text which is smaller that the smallest text in the body of the document, which is normally the figure captions or the footnotes. That means you will have to increase the size of the font of the ticks and the legend a bit.

The colorbar with the legend is not needed because its redundant with the information in the z axis ticks.

[image]

You are going to kill me, but remember we are aiming at the best of the best journals.

danifpdra, 2021-04-20:
Is this ok?

[image]

miguelriemoliveira, 2021-04-20:
Hi @danifpdra ,

yes, definitely. I don't think the colorbar is needed. I would remove.
The values in Angular distance are strange .. this are not degrees are they?

Also, the values of the error are strange. Are you sure those are pixels? Are they not meters? If so perhaps consider using millimeters in the graph for a better visualization.

danifpdra, 2021-04-20:
You are right,  the angular distance is in rad, not degrees. Do you want me to convert to degrees or should I just change the labels?

Yes, that particular plot is from LiDAR, the errors are in mm. Sorry, I copy-pasted and didn't notice it. I have already changed it.

miguelriemoliveira, 2021-04-20:
Ahah ... everytime I catch one more typo in the graph you owe me a coffee :).

I would prefer degrees, its more intuitive

danifpdra, 2021-04-20:
Did I get it right this time?

[image]

miguelriemoliveira, 2021-04-20:
No : - ), the z axis values are in meters and you are saying in the legend that they are millimeters. 

Can you multiply those values by 1000?

Do I get my coffee : - )?

danifpdra, 2021-04-20:
I'm confident that this is still not correct :-)

[image]

miguelriemoliveira, 2021-04-21:
Its great. Thanks for the patience : - ). I hope you agree that now its much better than at the beginning ...

And don't forget my coffee!

danifpdra, 2021-04-21:
Yes, I agree. Thanks!

It's already on its way, I sent it on high priority mail :-)

miguelriemoliveira, 2021-05-01:
Hi @danifpdra . Can we close this issue? Its done, right?

closed 2021-05-03 by miguelriemoliveira