lardemua/atom #284: Rviz panel - set initial estimate - Reset sensor/Reset All

MiguelDRPina, 2021-04-15:
In issue #280, pressing the reset button resets the initial estimate for all sensors.
It would be nice to have an extra button to reset just a given sensor.

The same could maybe be done later for the save button, if seen fit.

miguelriemoliveira, 2021-04-17:
Hi @MiguelDRPina ,

done. Take a look at the video https://youtu.be/sf9JkdeGKrU

The video shows which messages you need to send for each case. 

Will assign the issue to you.

MiguelDRPina, 2021-04-18:
Hello @miguelriemoliveira,
Sorry I took so long to respond, I had a very busy weekend.
I have seen the video, thank you. Tomorrow morning I'll try to have that done.

miguelriemoliveira, 2021-04-18:
Hi @MiguelDRPina ,

no problem. Let me know if you need some help.

MiguelDRPina, 2021-04-19:
Hi @miguelriemoliveira ,

So, I was able to do it. The only thing is, the reset sensor and reset all are sending me to different places. If I move the sensors and then click on Reset All, it is going to the last position saved, as it was doing already. But when I move a sensor and then try to reset just that sensor it always goes to the same position, like this:

[image]

Is there something I am missing in the message I am sending (I basically just changed the marker_name depending on which button is clicked) or is it something from your side?

miguelriemoliveira, 2021-04-19:
Hum, sounds like a bug from my side. Push your code and I will check.

MiguelDRPina, 2021-04-19:
The code is pushed.

miguelriemoliveira, 2021-04-19:
Hi @MiguelDRPina , when you push code related to an issue add a reference to that issue in the commit message. That way the commit appears in these conversations and I will know automatically that you pushed code related to this. Check this example above:

[image]

I will work on this and get back to you.

MiguelDRPina, 2021-04-19:
I am pretty sure I did. Maybe it's because I only referenced "#284" in the commit but because my side is on a different package ([this package](https://github.com/MiguelDRPina/atom_rviz_plugin)), it didn't show up here. How would I reference an issue from a different package in the commit?

miguelriemoliveira, 2021-04-19:
Hi @MiguelDRPina ,

I think this is how to do it:

https://github.com/gogs/gogs/issues/360

MiguelDRPina, 2021-04-19:
Just did a commit to test it and it worked. Thank you.

miguelriemoliveira, 2021-04-20:
I investigated a bit, and using the spheres (the original control it works). 

Using the 3dlidar as a case study, the message that is send when the spheres are selected is (**and this works as intended**):

```
header: 
  seq: 220
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "tripod_left_support"
client_id: "/rviz/MoveSensors-InteractiveMarker"
marker_name: "3dlidar_menu"
control_name: "menu_only_control_u0"
event_type: 2
pose: 
  position: 
    x: -1.8976080417633057
    y: 0.0
    z: 0.9622363448143005
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
menu_entry_id: 1
mouse_point: 
  x: -1.9207031726837158
  y: -0.0872483104467392
  z: 0.9642506241798401
mouse_point_valid: True
---
```

Using your code (pressing the reset sensor button in the panel), the message is:

```
---
header: 
  seq: 251
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
client_id: "/rviz/MoveSensors-InteractiveMarker"
marker_name: "3dlidar"
control_name: ''
event_type: 2
pose: 
  position: 
    x: 0.0
    y: 0.0
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0
menu_entry_id: 2
mouse_point: 
  x: 0.0
  y: 0.0
  z: 0.0
mouse_point_valid: False
---
```

I think the problem should be in the different in value of the field marker_name: "3dlidar_menu" in my case, "3dlidar" in your case. Can you change an see if it works?

MiguelDRPina, 2021-04-20:
Hi @miguelriemoliveira ,
Yes, that was it. Sorry, my bad.
It's working now, thank you. I will be closing this issue.

closed 2021-04-20 by MiguelDRPina

miguelriemoliveira, 2021-04-20:
Great. One less.

miguelriemoliveira, 2021-04-20:
BTW, do you need me to develop any more functionalities? I think I will go into the one of getting information about collections... Any other ideas?