lardemua/atom #285: Integration with Rviz panel - set initial estimate - micro movements of the control

MiguelDRPina, 2021-04-15:
The functionality left to do from the functionalities mentioned in issue #279 is the micro movements of each sensor.

In the panel we should maybe have boxes for each xyz and rpy coordinates and as we changed them, it would send a pose message with those coordinates that would change the sensors position.

miguelriemoliveira, 2021-05-03:
Hi @MiguelDRPina ,

I think you just have to emulate the messages that are sent when you move the marker. Take a look at the video:

https://youtu.be/9-ocrJhv8qk

MiguelDRPina, 2021-05-04:
Hi @miguelriemoliveira ,
Ok, but what would be the control name for changing all xyz and rpy? Or should I be publishing a message for each of the 6? Because in the video you only move in one direction and use the control name "move_y". I tried moving the marker in the sphere to move all to see which message would be published because I remember I used to be able to do that, but for some reason I can't move it in the sphere anymore.

miguelriemoliveira, 2021-05-04:
Hi @MiguelDRPina ,

don't use the sphere. Use the arrows and the rings and do a rostopic echo to get the message that you need to replicate.

MiguelDRPina, 2021-05-04:
I know, what I meant was, depending on which arrow/ring I use, the control_name will be different (move_y if I move the arrow to for y, 'move_z' for the z arrow, 'rotate_y' if I use the y ring, etc). 

My goal was to publish a message to change the entire pose at once, that's why I was asking about the sphere, because with the sphere more than one position was being changed simultaneously, so I was trying to see the message sent for that. Or do you think it's better if I just send 6 messages everytime, one for each position/rotation?

miguelriemoliveira, 2021-05-04:
Hi @MiguelDRPina ,

better that you send a message every time the user pushes the button. You will have 12 buttons I think, minus and plus the 6 DOF. And for each button you send a special message.

That's the best way I think because you get immediate visual feedback after pressing the button

MiguelDRPina, 2021-06-01:
Hi @miguelriemoliveira ,

Today I decided to just pop in for a few moments to try to advance in a few of the issues at hands, before I return to the writing. For this issue, I already send the poses and the sensors move (GIFs below), but I should be able to have their current pose. How can I get it? I thought that I could read it from the /initial_estimate/feedback and /data_labeler/feedback topics, but I am not sure how to do it or even if I can do it that way.

Initial Estimate:
[image]

Data Collection:
[image]

miguelriemoliveira, 2021-06-01:
Hi @MiguelDRPina ,

you can subscribe to **/set_initial_estimate/feedback** and you get the pose of the marker published every time you move it.

[image]

The feedback message is:

```yaml
--
header: 
  seq: 63
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "tripod_right_support"
client_id: "/rviz/MoveSensors-InteractiveMarker"
marker_name: "world_camera"
control_name: "move_x"
event_type: 1
pose: 
  position: 
    x: 0.13281035423278809
    y: 0.07000000029802322
    z: 0.014999999664723873
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
menu_entry_id: 0
mouse_point: 
  x: 0.0
  y: 0.0
  z: 0.0
mouse_point_valid: False
```

from this you can get the pose right?

MiguelDRPina, 2021-06-01:
What I wanted was to get the position when I click (on the table list of the sensors) in one of the sensors. 

For instance, when I first open the panel, all xyz and rpy are at 0 because no sensor is selected. I would select the hand_camera in the list and would see at what position it is, before even moving them. After moving it around, I would then click on 3dlidar and would see the position it is at. And so on. 

Otherwise, if I move a sensor to a certain pose and I then select other sensor and move one of sliders, it will update with the values used for the previous one. Hope I was clear.

miguelriemoliveira, 2021-06-01:
> What I wanted was to get the position when I click (on the table list of the sensors) in one of the sensors.

That could be a problem because the feedback message is only sent when the interactive marker is moved.

In that case you should use the tfs:

[image]

MiguelDRPina, 2021-06-01:
I am a little confused with something.

[image]

I checked the tf for the hand_camera, like you did and also listened to the /set_initial_estimate/feedback topic. When I moved the hand_camera, the pose on the feedback was different than the one on the tf. Do you know why?

miguelriemoliveira, 2021-06-01:
Yes, the the transformation should be different.

For example, the pose you get on the feedback of the hand camera is from the end_effector to the hand_camera_link.

When you do a tf_echo you have to select the same frames to get the same transform ...

closed 2021-06-12 by MiguelDRPina