lardemua/atom #286: Integration with Rviz Panel - set initial estimate - Listing all sensors

MiguelDRPina, 2021-04-15:
In the panel, we currently have a drop-down box with the sensors.
It could maybe be more intuitive to change it for a list box that would allow us to see all sensors.

MiguelDRPina, 2021-04-21:
Hi @miguelriemoliveira ,

Today I have been on this issue, but I came across some difficulties that I will do my best to explain. Basically, when RViz opens, it's supposed to run some functions, and they do run. The problem is with the services to get the state of visibility/scale (issue #278 references this service). 

For some reason I am not able to read the state right away, only when some action happens. This wasn't a problem when I had the combobox because of the way I had set it up where the user would always have to select the sensor and by performing that action, the service would be called as expected. Now that I am trying the list (or a table, still testing which one I like more), it is a problem.

[image]

This GIF should help explaining my problem. Basically, when it starts in the initial estimate tab you see the 'hand_camera' sensor in the combobox, which is being read. But as you see, the marker is visible and with a 0.5 scale and that's not what it shows on the checkbox and spinbox, it only appears correctly when I change in the combobox back to the hand_camera.

You can then see in the 'Test' tab what I need this for. In that table I am supposed to have all the checkboxes ticked as they are all visible, but that is not the case because for some reason the service isn't being called when initialized.

Do you think you know what I could be missing?

miguelriemoliveira, 2021-04-21:
Hi @MiguelDRPina do you have your code pushed? Where is the code for the test tab?

miguelriemoliveira, 2021-04-21:
Hi @MiguelDRPina ,

I did some debugging and added a PFLN in two methods.

```cpp
CalibrationPanel::CalibrationPanel(QWidget* parent) : Panel(parent),  ui_(new Ui::CalibUI())
    {
      ui_->setupUi(this);
      int argc = 0;
      char** argv;
      ros::init(argc, NULL, "rviz_panel");
      PFLN
    }

    CalibrationPanel::~CalibrationPanel() = default;

    void CalibrationPanel::onInitialize()
    {
      // Functions to run when rviz opens
      PFLN
      getSensors();
      initEstimateComboBoxTextChanged();

     // (...)
```

and I think the problem is that your two calibration panel class methods **constructor** and **onInitialize**are only called when the panel is called again. If we load rviz with a configuration where the panel  already existed these methods are not called. Perhaps you can use the event for changing the tab to test tab as the trigger to read the list of sensors ...

I am pushing this version for debugging ...

miguelriemoliveira, 2021-04-21:
oops, forgot to add the correct git message ... https://github.com/MiguelDRPina/atom_rviz_plugin/commit/b5713799ef5b4f19759be4b213d0605c073c7a55

miguelriemoliveira, 2021-04-21:
Hi @MiguelDRPina ,

I think I got it wrong the last time. These two methods run either on rviz initialization, when they are configured in the rviz config file, or when we go to panels -> add display -> atom_calibration.

Now for me the problem is that the list of sensors has nothing ...

miguelriemoliveira, 2021-04-21:
my bad ... the list of sensors prints correctly. I forgot to do the 

rosparam load config.yml

You could create a launch to make this easier.

Now I am a bit lost:

> You can then see in the 'Test' tab what I need this for. In that table I am supposed to have all the checkboxes ticked as they are all visible, but that is not the case because for some reason the service isn't being called when initialized.

which service do you mean? Which code line?

MiguelDRPina, 2021-04-21:
I don't think I can create a launch file for that, because I don't have permission to edit the mmtbot package (don't give me permission though, at least for now, sometimes as I make changes it's good to know I can't mess anything up there).

It's the service GetSensorInteractiveMarker, I call the setTable function at the end of the calibration_panel.cpp file (line 86) and the code for that function where I call the service is right in the beginning of the set_initial_estimate.cpp file.

miguelriemoliveira, 2021-04-21:
Ok so the problem is that the service call fails. This is the correct way to call a service

http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28c%2B%2B%29

```
if (client.call(srv))
  {
    ROS_INFO("Sum: %ld", (long int)srv.response.sum);
  }
  else
  {
    ROS_ERROR("Failed to call service add_two_ints");
    return 1;
  }
```

Now im trying to figure out why it fails ...

MiguelDRPina, 2021-04-21:
But did it fail when you tested it? Because it was calling the service correctly before. If you watch the GIF I sent, when I change a sensor in the combobox (set_initial_estimate tab), it calls the service with no problem and I am able to set the checkbox and spinbox according to the state of the visible/scale values returned by the service.

The problem is that that is not happening when I first launch the launchfile if the panel is already open, only with some action.

miguelriemoliveira, 2021-04-21:
Hi @MiguelDRPina,

`You can then see in the 'Test' tab what I need this for. In that table I am supposed to have all the checkboxes ticked as they are all visible, but that is not the case because for some reason the service isn't being called when initialized.
`
The service **was being called**, the problem was that the call was failing. 

The reason for that was that the service server sometimes takes some time to be ready to operate, and the client was initializing much faster, which meant that when the service client called the service, the server was not yet ready.

The solution is to wait until the service is ready. 

https://answers.ros.org/question/323885/how-to-make-cs-service-call-method-wait-for-servers-response/

Take a look at the solution:

```
client.waitForExistence(); // Wait for the service to be available before calling
if (client.call(srv)){
   ROS_INFO("Service %s called successfully.", service_name.c_str() );
}
else{
   ROS_ERROR("Failed to call service %s", service_name.c_str());
   return; // perhaps here we should do a shutdown or an exit?
}
//       cout <<  << srv.response.visible << endl;
ROS_INFO("srv.response.visible=%d", srv.response.visible); // just for testing, return visible=1 so it should be fine
```

Let me know if it works...

miguelriemoliveira, 2021-04-21:
> But did it fail when you tested it? Because it was calling the service correctly before. If you watch the GIF I sent, when I change a sensor in the combobox (set_initial_estimate tab), it calls the [...]

yes, but in that case, because you use some interaction, you will always call the service after the server is ready. The test tab was calling it on initialization, when the server was not ready.

> The problem is that that is not happening when I first launch the launchfile if the panel is already open, only with some action.

I think it does happen. Check my prints and PFLNs. I left them there.

MiguelDRPina, 2021-04-21:
> yes, but in that case, because you use some interaction, you will always call the service after the server is ready. The test tab was calling it on initialization, when the server was not ready.

Ok, so then is there a way to get the server ready and call the service, both in initialization?

MiguelDRPina, 2021-04-21:
Oh, I just pushed the code. And it is working.

MiguelDRPina, 2021-04-21:
> > yes, but in that case, because you use some interaction, you will always call the service after the server is ready. The test tab was calling it on initialization, when the server was not ready.
> 
> Ok, so then is there a way to get the server ready and call the service, both in initialization?

Nevermind, I understand what you did. Thank you.

miguelriemoliveira, 2021-04-21:
Great.

closed 2021-04-22 by MiguelDRPina