lardemua/atom #288: Integration with Rviz Panel - Collect Data - Get collection state

MiguelDRPina, 2021-04-15:
We should be able to show the collections made.

miguelriemoliveira, 2021-04-16:
Hi @MiguelDRPina ,

could I ask you to create an issue for grouping all the issues in Integration with Rviz Panel - Collect Data, as in

https://github.com/lardemua/atom/issues/279

MiguelDRPina, 2021-04-16:
Hi @miguelriemoliveira ,

Done (issue #290)

miguelriemoliveira, 2021-04-16:
> Done (issue #290)

Thanks

miguelriemoliveira, 2021-04-25:
Hi @MiguelDRPina ,

I implemented the service https://github.com/lardemua/atom/blob/noetic-devel/atom_msgs/srv/GetDataset.srv

I hesitated back and forth and ended up deciding that it made no sensor for the python code data_collector to be filtering the information in the dataset. Might as well send you everything and then you decide what you need and take it. 
So the solution was for the service to send the complete json string which describes a dataset (the content of a data_collected.json file in a dataset).

Granted, this way you get the burden of having to parse a json and navigate it. To make  things easier I looked around on how this could be done. Lets use this cpp json parser https://github.com/nlohmann/json (nothing to install. Just added the hpp to our repo).

The test code from where you can see how you can integrate is 
https://github.com/MiguelDRPina/atom_rviz_plugin/blob/main/atom_rviz_plugin/src/test_json_parser.cpp

Let me know if you need some help.

MiguelDRPina, 2021-04-25:
Hi @miguelriemoliveira ,
Ok, thank you. I will take a look into it and I'll say something if I need to.

MiguelDRPina, 2021-04-26:
Hello @miguelriemoliveira ,

I only had a chance to work on this issue this morning, and I am running into some errors. 

I have ran the launchfile for the data collect, saved a few collections and ran the [file](https://github.com/MiguelDRPina/atom_rviz_plugin/blob/main/atom_rviz_plugin/src/test_json_parser.cpp) you made and it all printed out fine. But then I tried just parsing the json file in the [data_collect.cpp](https://github.com/MiguelDRPina/atom_rviz_plugin/blob/main/atom_rviz_plugin/src/data_collect.cpp) file using your code and for some reason I am getting an error. I tried running the code in the same function that publishes the message to save the collection (when the save button is clicked), but then I also created a button to separate both actions, thinking it would make a difference. It didn't.

Do you know what I could be missing?

miguelriemoliveira, 2021-04-26:
Hi @MiguelDRPina ,

when you run in the data_collect.cpp, is the service call responded successfully, or does it fail?

MiguelDRPina, 2021-04-26:
I am not 100% sure, since I am running with the launchfile I can't see what is being printed in the terminal. But I would assume so, because if I comment everything after calling the service, there's no error.
The Rviz crashes only when I insert this line:
```ruby
json j = json::parse(srv.response.dataset_json);
```

miguelriemoliveira, 2021-04-26:
> I am not 100% sure, since I am running with the launchfile I can't see what is being printed in the terminal. 

you should fix this, otherwise it will be very difficult to debug.

Your code is running inside rviz, and since we are launching rviz with a launch file we must make sure the output of the node is being redirected to the terminal, i.e., this line should have a **output="screen"**

https://github.com/miguelriemoliveira/mmtbot/blob/8dc4f873b80d95b13359079b7fc9360cf8543908/mmtbot_calibration/launch/playbag.launch#L62

Try again and let me know if the service is successfully called or not...

MiguelDRPina, 2021-04-26:
Yes, service called successfully. And the json file is saved on output_folder. The error appears with that line I mentioned.

miguelriemoliveira, 2021-04-26:
and what's the output of this?

`ROS_INFO("srv.response.dataset_json=%s", srv.response.dataset_json.c_str());`

MiguelDRPina, 2021-04-26:
[image]

miguelriemoliveira, 2021-04-26:
Hi @MiguelDRPina ,

not sure I understand your response ...  where is the text?

"srv.response.dataset_json = blablabla ..."

Did you change the launch file as described here?

> you should fix this, otherwise it will be very difficult to debug.
>
> Your code is running inside rviz, and since we are launching rviz with a launch file we must make sure the output of the node is [...]

MiguelDRPina, 2021-04-26:
I did change the launchfile. And I highlighted the line you asked (first line in the terminal).

miguelriemoliveira, 2021-04-26:
ah ... sory, did not see it. Better to copy paste the terminal's output rather than posting an image. Its much easier to spot and copy paste.

In any case, this is the problem, your json is empty.

The procedure is: 

1. launch
`roslaunch mmtbot_calibration collect_data.launch  output_folder:=/home/mike/datasets/mmtbot/tmp overwrite:=true`

2. get at least one collection by setting the save (use the right button click on the sphere's  to make sure it works)

3. Then call the service to getDataset

it should work like this ... are you doing like this?

MiguelDRPina, 2021-04-26:
Well, the way I was doing was: in the function connected to the event of clicking the save Button I had created in the panel (in issue #287) I would call the service and try to read the json so I could then add the collection to the treeWidget in the panel.

MiguelDRPina, 2021-04-26:
Ok, I have now tried separately again (one button just to save the collection and one where I call the service and parse the json file just like you did) and it worked. It wasn't working when I tried earlier. But then my question is why can't I do all of this in the same function? 
Because for me it makes sense that, as I click on the button to save the collection, that collection would be added to the tree list. And that means I would have to parse the json file in the same function as the one I use to save the collection.

miguelriemoliveira, 2021-04-26:
It should work as you have but I am not sure. If you do it as I suggest does it work?

miguelriemoliveira, 2021-04-26:
Push your code so I can try it ...

miguelriemoliveira, 2021-04-26:
You, it should work in the same, but let me check from my side ...

MiguelDRPina, 2021-04-26:
Code is pushed. The way it is right now is the way that works (parsing the json in a separate function). If you want the way it doesn't work, just uncomment the commented part on the save function.

miguelriemoliveira, 2021-04-26:
OK, thanks. A moment please ...

miguelriemoliveira, 2021-04-26:
I am stuck on #291 can you see what's wrong with that so I can test?

miguelriemoliveira, 2021-04-26:
I think I know what's happening. The idea is that two functionalities are called in sequence:

1. save a new collection
2. get the state of all collections

The problem is that we are using a message publication for 1. which does not tell us when the process of saving the collection is finished. Note that you should only go into 2. when the collection has finished saving. Having a publish / subscribe does not give this information. I will implement a service for 1. and then all should work fine.

...

MiguelDRPina, 2021-04-26:
Ok, thank you.

miguelriemoliveira, 2021-04-26:
Done. Tested it. Should work fine.

MiguelDRPina, 2021-04-26:
Seems to be working fine, I'll work on it from here. Thank you again.

miguelriemoliveira, 2021-04-26:
No problem. It is a pleasure to code a bit ...

MiguelDRPina, 2021-04-27:
Hello @miguelriemoliveira,

Just 2 quick questions.
```ruby
      for (auto& collection: j["collections"].items()) // iterates over all collections
      {
        for (auto& sensor: collection.value().at("labels").items()) { // iterates over all sensors in this label
          QTreeWidgetItem *childLevel2TreeItem = new QTreeWidgetItem();
          childLevel2TreeItem->setText(0, "sensor ");
          childLevel1TreeItem->addChild(childLevel2TreeItem);
        }
      }
```
In your json parser, you have 2 cycles like this (forget what's inside the cycles). The first cycle is iterating over the sensors of all collections, but the way I am doing I wanted to iterate over all sensors of JUST the one collection that I am adding when I click on the save button (I have the index of the desired collection). How can I do that?

Second question is, besides the collections, each respective sensors and if they were labeled or not, is there any other thing from the json file you or @eupedrosa would like to see on the tree?

miguelriemoliveira, 2021-04-27:
Hi @MiguelDRPina ,

the collections are named with numbers, i.e. 0,1,2,3, ... .

The last saved one is the highest number. You could use this...

So just print all collection names

 ```
for (auto& collection: j["collections"].items()) // iterates over all collections
 {
count << collection.key() << endl;
}
```

an figure out a way to get the largers value for key (you probably have to convert to numeric to compare).

A tip: copy the contents of the data_collected.json file you have from a dataset and paste them into 

https://jsoneditoronline.org/#left=local.yikeme&right=local.reqaji

It helps you visualize the structure in the json.

> Second question is, besides the collections, each respective sensors and if they were labeled or not, is there any other thing from the json file you or @eupedrosa would like to see on the tree?

hum, not sure for now. Just show that information and then we'll probably remember more stuff... @eupedrosa , any ideas?

closed 2021-04-29 by MiguelDRPina