lardemua/atom #289: Integration with Rviz Panel - Collect Data - Delete collection

MiguelDRPina, 2021-04-15:
We should be able to delete a particular collection made, if desired.

MiguelDRPina, 2021-04-28:
Hi @miguelriemoliveira,

I have 2 questions about this issue.
1. I am already able to delete the collections that I want in the panel, but I don't do anything with the json itself. How do we do that? Do you create a service that receives the index of the collection I want to delete and delete on the ATOM's side?

2. Should I readjust the names of the collections in the panel as I delete a collection? As in, if I have collections 0, 1, 2, 3 and 4, and I delete collection 2, collection 3 becomes 2 and collection 4 becomes 3. It only makes sense to do it if it is done on the json file aswell, otherwise if I want to delete more than one collection, the indexes can become all mixed up.

miguelriemoliveira, 2021-04-28:
> * I am already able to delete the collections that I want in the panel, but I don't do anything with the json itself. How do we do that? Do you create a service that receives the index of the collection I want to delete and delete on the ATOM's side?

Yes, that's it. I can do it.

> * Should I readjust the names of the collections in the panel as I delete a collection? As in, if I have collections 0, 1, 2, 3 and 4, and I delete collection 2, collection 3 becomes 2 and [...]

What you should do is to call the service to delete and when its finished, call again the service to get the collections. That way you are sure you have the updated description of the dataset.

miguelriemoliveira, 2021-04-30:
Hi @MiguelDRPina ,

Implemented the service to delete a collection. Please try it out and let me know if you run into trouble.

Example of calling the service:

`rosservice` call /collect_data/delete_collection "collection_name: '"3"'"

MiguelDRPina, 2021-04-30:
HI @miguelriemoliveira ,
Thanks, will implement it today.

MiguelDRPina, 2021-04-30:
Hello @miguelriemoliveira ,

It's working. Only one thing... I was testing everything and I just noticed that in your [test_json_parser](https://github.com/MiguelDRPina/atom_rviz_plugin/blob/main/atom_rviz_plugin/src/test_json_parser.cpp), if I have more than 10 collections, those 2 digit collections will come right after their respective first digit. Like this:
[image]
Is this the way it is being added into the json file? Is it something easy to fix from the ATOM side or should I try to rearrange it on this side? Because I was adding it to the panel every iteration, so the panel will also look something like this:
[image]

miguelriemoliveira, 2021-05-01:
Hi @MiguelDRPina ,

This occurs because the sorting of string variables puts 10 before 2 and so on. In any case, the data structure that we have to store the collections is a dictionary, which means that the order by which the collections appear is not important. 

If you want, for visualization purposes you can sort the collections by name (also an option to sort them by aquisition time as you have now ) in order to present them as you want in the table.

Concerning your second questions: It would be easy to solve from the ATOM's side, but we have already many collected datasets and changing something could render those datasets unusable, so touching the structure of the jsons is the least adequate option due to the impact ot may have.

miguelriemoliveira, 2021-05-01:
BTW @MiguelDRPina am I missing something to do from my side? Please tell me if you are waiting for me with something ...

MiguelDRPina, 2021-05-01:
Hi @miguelriemoliveira ,
Ok, I will sort it then.

MiguelDRPina, 2021-05-01:
> BTW @MiguelDRPina am I missing something to do from my side? Please tell me if you are waiting for me with something ...

Only for issue #285 regarding the micromovement of the sensors for the initial estimate. I already have in the panel 6 spinboxes (3 for xyz and 3 for rpy) and a button that when clicked would send a message/call a service with the pose from those spinboxes and you would change it on the ATOM's side.

Right now there's only an issue for that in the set initial estimate, but I think that if we are doing that for the initial estimate, maybe we would do it for the data collect aswell?

MiguelDRPina, 2021-05-01:
Hi @miguelriemoliveira ,
I only had the chance to work on this now.

``` ruby
for (auto& collection: j["collections"].items()) // iterates over all collections
      {
        // First level of the tree ("collection 0", "collection 1", ..., "collection X")
        QTreeWidgetItem *childLevel1TreeItem = new QTreeWidgetItem();
        childLevel1TreeItem->setText(0, "collection " + QString::fromUtf8((collection.key().c_str())));
        topTreeItem->addChild(childLevel1TreeItem);

        // Second level of the tree ("labels")
        QTreeWidgetItem *childLevel2TreeItem = new QTreeWidgetItem();
        childLevel2TreeItem->setText(0, "labels");
        childLevel1TreeItem->addChild(childLevel2TreeItem);

//        ROS_INFO_STREAM("Collection " + collection.key());

        for (auto& sensor: collection.value().at("labels").items()) { // iterates over all sensors in this label
          // Level three of the tree (sensor)
          QTreeWidgetItem *childLevel3TreeItem = new QTreeWidgetItem();
          childLevel3TreeItem->setText(0, QString::fromUtf8((sensor.key().c_str())));
          childLevel2TreeItem->addChild(childLevel3TreeItem);

          // Level four of the tree (label detected)
          QString labels_detected = sensor.value().at("detected") ? "detected: true" : "detected: false";

          QTreeWidgetItem *childLevel4TreeItem = new QTreeWidgetItem();
          childLevel4TreeItem->setText(0, labels_detected);
          childLevel3TreeItem->addChild(childLevel4TreeItem);
        }
      }
```

This is the code where I add the lists. How exactly were you suggesting I could sort it? I tried looking into sorting at the end of the cycles where I would cycle through the added collections and tried to sort it using Qt, but in Qt it is not that easy to sort.

miguelriemoliveira, 2021-05-02:
> You said that right now it is being sorted by aqquisition time, but that's not the case, at least I don't think it is. The collection that is saved as being the 10th collection is indeed the 10th collection being saved.

Right, I agree.

>  However, when I iterate like in the first line of the code below, the 10th collection will come after collection 1 (I understand why, it's just not what I wanted). That's either the way the json is being saved or the way this iterator works, but I can't seem to solve this.``

Since you have a dictionary when you travel through the keys in the dictionary there is no order associated with the keys. 

My strategy would be:

1. Travel the dictionary keys and  build a vector of integers **vector<int>** containing the keys of all collections. Important detail: you will have to cast if needed the collection key to integer.
2. Sort the vector of integers  https://www.geeksforgeeks.org/sorting-a-vector-in-c/
3.  Cycle through the sorted vector and, for each element in the vector, you have a key to recover the corresponding dictionary value.

If you cannot implement let me know and I can give it a try ...

MiguelDRPina, 2021-05-02:
I got the first 2 points you made, and I was already trying to do something similar, in a slightly different way. I guess my problem is with your 3rd point. Having the key of the value I want to retrieve, how can I get its corresponding value?

miguelriemoliveira, 2021-05-02:
Hi @MiguelDRPina ,

I think you can do 

`j["collections"].at(key)`

does it work?

MiguelDRPina, 2021-05-02:
I think it does, I'll confirm it in a sec.

MiguelDRPina, 2021-05-02:
Hi @miguelriemoliveira ,
It's working. Thank you.

closed 2021-05-02 by MiguelDRPina