lardemua/atom #290: Integration with Rviz Panel - Collect Data

MiguelDRPina, 2021-04-16:
Functionalities to develop for the collect_data:

- [ ] Save Collection (#287)
- [ ] Get Collection state (#288)
- [ ] Delete Collection (#289)

We can add more functionalities later if we can remember more.

MiguelDRPina, 2021-04-23:
Hi @miguelriemoliveira,

I ran into some problems with running the collect_data launchfile. I launch the file,  everything is good but whenever I try to add the panel, it just takes too long until eventually I get a message saying Rviz is not responding and I have to close it.

I tried opening other panels that I had here (simpler panels, only 1 tab and 1 button) and they open, but the one we've been working on doesn't. Any suggestion?

MiguelDRPina, 2021-04-23:
Hi again @miguelriemoliveira ,
I figured it out. It was because of the service that is being called upon initialization of the panel (issue #286) for the set_initial_estimate part. I have (temporarily) commented that part of the code to work on the collect data part and I was able to add the panel. But long term, how could we solve this problem?

miguelriemoliveira, 2021-04-23:
Hi @MiguelDRPina ,

One option would be to replace the infinite wait for the service:

`client.waitForExistence(); // Wait for the service to be available before calling`

with a wait for a fixed reasonable time such as wait for 2 secs, then call the service.

If the service call still fails then you could disable that panel.

Another idea:

The service should be called not on initialization of the panel but whenever you jump into the tab. You should have an event for that, just call the service there ...

miguelriemoliveira, 2021-04-23:
a third option:

```cpp

#define _MAX_ATTEMPTS_ 10
success = False;

for (size_t i=0; i<MAX_ATTEMPTS; i++){

   if (client.call(srv)){
      success = True;
   }    

   if success == True { // stop cycle if call was successful

       ROS_INFO("Service %s called successfully (at attempt %s).", service_name.c_str(), i);
       break;

   }else if success == True  && i == MAX_ATTEMPTS -1 {

      ROS_ERROR("Failed to call service %s (tried %d times)", service_name.c_str(), i);
      return; // perhaps here we should do a shutdown or an exit?

H erre 's were you can disable the tab
   }
}
```

eupedrosa, 2021-04-23:
There is no need for initializations or even use a `ServiceClient`. ROS also provides a [simpler API](https://docs.ros.org/en/api/roscpp/html/namespaceros_1_1service.html#33d9fb0f32ff8bbac4db4188d657715d) to interact with 
services. 

When pushing the button you can simple do
```c++
if (ros::service::exists("/path/of/service"))
    ros::service::call("/path/of/service", service_object);
else
    ROS_INFO("Complain the services does not exist")
```

MiguelDRPina, 2021-04-23:
Hi @miguelriemoliveira and @eupedrosa ,

At least for now, I've gone with an approach similar to the 2nd suggestion @miguelriemoliveira made (calling the service when the tab changes). That way I am able to keep working on the rest of the issues that might have a bigger priority. After that I can try and test all options on this to see what works best.
Thank you both.

miguelriemoliveira, 2021-04-23:
Hi @MiguelDRPina ,

yes, you can use the timeout argument in the wait for service (check @eupedrosa 's link).

In any case, you will have to handle the possibility of the service not being responded, even after waiting. The best way would be to disable that tab...

closed 2021-04-30 by MiguelDRPina