lardemua/atom #291: Error launching calibration panel

miguelriemoliveira, 2021-04-26:
Hi @MiguelDRPina,

I can't run ... I get this error 

```
terminate called after throwing an instance of 'YAML::BadFile'.978000               
  what():  bad file
```

when I launch the atom calibration panel ...any idea what's wrong?

MiguelDRPina, 2021-04-26:
Yes. Change the file path in line 22 of the [read_write_config](https://github.com/MiguelDRPina/atom_rviz_plugin/blob/f2618c191b0b77fbe2b3864910c32ce6195b1257/atom_rviz_plugin/src/read_write_config.cpp) to your config.yaml absolute path.

MiguelDRPina, 2021-04-26:
I intend on maybe having the path being given within the panel, but for now I am setting it in the code.

miguelriemoliveira, 2021-04-26:
OK, but you should set that so that it does not break the code. Something like if text is empty do not run these functionalities ...

closed 2021-04-26 by miguelriemoliveira