lardemua/atom #293: Integration with Rviz Panel - set initial estimate - calling service to get the state of visible and marker scale

MiguelDRPina, 2021-05-03:
Hi @miguelriemoliveira ,
Can you add a new service that, when called, returns the current state of the services 'visible' and 'scale' (addressed here: https://github.com/lardemua/atom/issues/278)?

miguelriemoliveira, 2021-05-03:
Hi @MiguelDRPina ,

I transferred the repo to the lardemua organization, which is the official group of the DEM Robotics group. Here it is:

https://github.com/lardemua/atom_rviz_plugin

I gave you admin permissions on this repo.

After this I was able to transfer the issues to the atom. Here they are:

https://github.com/lardemua/atom/issues/293
https://github.com/lardemua/atom/issues/294
https://github.com/lardemua/atom/issues/292

I recommend that you remove the atom_rviz code from the repo (make a backup before) and clone it again from the lardemua owner. 

I think that's it no? Do you need anything else?

MiguelDRPina, 2021-05-04:
That's it, yes. Thank you.