lardemua/atom #295: Integration with Rviz Panel - read/write in config file - Dump in config.yml

MiguelDRPina, 2021-05-05:
In the configuration file, the panel is already able to read the configuration file and set the parameters. 

When we alter the parameters and click on the 'Write' button, it does change the ROS parameters, but I can't change on the .yml file. It would be nice to do this, so the parameters wouldn't have to be set again everytime someone wants to do a calibration.

MiguelDRPina, 2021-06-01:
Hi @miguelriemoliveira ,
I am able to dump the values in the config.yml file. The only thing is, the comments disappear after the first dump. I've looked around and it seems to be expected. Do you think that is a problem?

[image]

MiguelDRPina, 2021-06-12:
Helo @miguelriemoliveira ,

I have two yaml files, [one with](https://github.com/lardemua/atom_rviz_plugin/blob/main/atom_rviz_plugin/src/config.yml) and the [other without](https://github.com/lardemua/atom_rviz_plugin/blob/main/atom_rviz_plugin/src/config_no_comments.yml) comments. 

I wanted to be able to get the values from the one without comments to the commented one, without losing the comments. Could you help me?

miguelriemoliveira, 2021-06-12:
Hi @MiguelDRPina ,

Done. I am using a python package called ruamel. Install with 

```
roscd atom_rviz_plugin
sudo pip3 install requirements.txt
```

Then go the the scripts folder and run the script, e.g.:

```
roscd atom_rviz_plugin
./copy_content_to_yaml.py -yc test_config_content.yml -yf test_config_format.yml -yo test_out.yml
```

Let me know if it works.

miguelriemoliveira, 2021-06-12:
Pushed code to the atom_rviz_plugin repo

https://github.com/lardemua/atom_rviz_plugin/commit/be3772b9066672105e9aa933e75217ddd05e120f

MiguelDRPina, 2021-06-14:
Hello @miguelriemoliveira ,
It did work on the terminal. But how do I run the script inside the C++ code? I tried using the system() function, like this:

```ruby
      std::string python_script_path = ros::package::getPath("atom_rviz_plugin") + "/scripts/copy_content_to_yaml.py";
      std::string yaml_format_path = ros::package::getPath("atom_rviz_plugin") + "/scripts/test_config_format.yml";

      std::string command_line = "python3 " + python_script_path + " -yc " + config_file_path + " -yf " + yaml_format_path + " -yo " + config_file_path;
      system(command_line.c_str());

```

And it did recognize the script, but got the following error:
```
Traceback (most recent call last):
  File "/home/miguel/catkin_ws/src/calibration/atom_rviz_plugin/atom_rviz_plugin/scripts/copy_content_to_yaml.py", line 58, in <module>
    main()
  File "/home/miguel/catkin_ws/src/calibration/atom_rviz_plugin/atom_rviz_plugin/scripts/copy_content_to_yaml.py", line 50, in main
    copy_values(content, format)
  File "/home/miguel/catkin_ws/src/calibration/atom_rviz_plugin/atom_rviz_plugin/scripts/copy_content_to_yaml.py", line 40, in copy_values
    for item in din.keys():
AttributeError: 'NoneType' object has no attribute 'keys'
```

The code is pushed. Any idea on how to solve this?

miguelriemoliveira, 2021-06-14:
Hi @MiguelDRPina ,

can you print the command you are running, i.e., 

`cout << command_line << endl;`

I want to check how you are calling the script...

MiguelDRPina, 2021-06-14:
Sure. The output was:
```
python3 /home/miguel/catkin_ws/src/calibration/atom_rviz_plugin/atom_rviz_plugin/scripts/copy_content_to_yaml.py -yc /home/miguel/catkin_ws/src/calibration/mmtbot/mmtbot_calibration/calibration/config.yml -yf /home/miguel/catkin_ws/src/calibration/atom_rviz_plugin/atom_rviz_plugin/scripts/test_config_format.yml -yo /home/miguel/catkin_ws/src/calibration/mmtbot/mmtbot_calibration/calibration/config.yml
```

miguelriemoliveira, 2021-06-14:
Hi @MiguelDRPina ,

are you using the up to date code? Because I have nothing in line 40 ...

Please upload the yamls you are using in this example.

MiguelDRPina, 2021-06-14:
I did remove some of the imports, one of them was giving me an error and didn't seem that it was being used (the OptimizationUtils one).

miguelriemoliveira, 2021-06-14:
OK, I was having a problem pulling. How about hte yamls. Are they the ones you sent before or some others?

miguelriemoliveira, 2021-06-14:
Just tested with the ymls you gave me. It works. Please send the new yamls.

MiguelDRPina, 2021-06-14:
There are no other yamls. The command-line I sent you before works if i run it in the terminal. But using the system() function to run that same command-line inside the C++ code gives me the error that I showed above.

miguelriemoliveira, 2021-06-14:
Hi @MiguelDRPina ,

I created a cpp code for testing a system call of the python script

https://github.com/lardemua/atom_rviz_plugin/commit/a89eaf6032eb04f833f2aeae66a207728a05661a

It seems to work fine. 

```
roscd atom_rviz_plugin/scripts/
rosrun atom_rviz_plugin call_python
```

can you try it?

MiguelDRPina, 2021-06-14:
Thank you @miguelriemoliveira. I thought I did this from the start, but apparently not, it is working now. I will be closing this issue now.

closed 2021-06-14 by MiguelDRPina

miguelriemoliveira, 2021-06-14:
Great.