lardemua/atom #297: Integration with RViz Panel - Testing the panel with other robots

MiguelDRPina, 2021-05-05:
Hi @miguelriemoliveira ,

In our last meeting, you told me to ask you for the procedure for me to be able to test the ATOM panel on different robots, since right now we've only been testing with [mmtbot](https://github.com/miguelriemoliveira/mmtbot).

MiguelDRPina, 2021-06-18:
Hi @miguelriemoliveira ,
Last meeting we had we tried testing the panel with the agrob robot. I have cloned the repository, made the same adjustments you made in the meeting, added the correct bagfile, and when I tried running the set_initial_estimate, got this error:

```
resource not found: lms1xx
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/miguel/ws_moveit/src
ROS path [2]=/home/miguel/catkin_ws/src
ROS path [3]=/opt/ros/noetic/share
when processing file: /home/miguel/catkin_ws/src/calibration/husky/husky_description/urdf/accessories/sick_lms1xx_mount.urdf.xacro
included from: /home/miguel/catkin_ws/src/calibration/husky/husky_description/urdf/husky.urdf.xacro
included from: /home/miguel/catkin_ws/src/calibration/agrob/agrob_calibration/urdf/initial_estimate.urdf.xacro
RLException: while processing /home/miguel/catkin_ws/src/calibration/agrob/agrob_calibration/launch/playbag.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '/home/miguel/catkin_ws/src/calibration/agrob/agrob_calibration/urdf/initial_estimate.urdf.xacro']] returned with code [2]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro $(arg description_file)"/>
The traceback for the exception was written to the log file

```
Also, when I tried running the rosrun agrob_calibration configure command, got this:

```
Loading config file /home/miguel/catkin_ws/src/calibration/agrob/agrob_calibration/calibration/config.yml
Loading bagfile /home/miguel/bagfiles/agrob/calibration_2020-07-01-10-15-43_0.bag
Parsing description file /home/miguel/catkin_ws/src/calibration/agrob/agrob_description/urdf/agrob.urdf.xacro
Executing command: xacro /home/miguel/catkin_ws/src/calibration/agrob/agrob_description/urdf/agrob.urdf.xacro -o /tmp/description.urdf
Executing command: xacro /home/miguel/catkin_ws/src/calibration/agrob/agrob_description/urdf/agrob.urdf.xacro -o /tmp/description.urdf
Traceback (most recent call last):
  File "/home/miguel/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 115, in <module>
    description = URDF.from_xml_file(urdf_file)  # read teh urdf file
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 620, in from_xml_file
    xml_string = open(file_path, 'r').read()
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/description.urdf'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/miguel/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/miguel/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 118, in <module>
    raise ValueError('Could not parse description file ' + description_file)
ValueError: Could not parse description file /home/miguel/catkin_ws/src/calibration/agrob/agrob_description/urdf/agrob.urdf.xacro

```

Do you know how I can fix this?

miguelriemoliveira, 2021-06-18:
I think you are missing the lms1xx ros package

https://github.com/clearpathrobotics/LMS1xx

Git clone to your catkin ws and compile.

MiguelDRPina, 2021-06-18:
Yep, that was it. Thank you!

MiguelDRPina, 2021-06-18:
The only thing now is in RViz, the interactive markers are not visible and I get the following warning:
```
Cannot get tf info for init message with sequence number 1. Error: "world" passed to lookupTransform argument source_frame does not exist. 
```

MiguelDRPina, 2021-06-18:
And in the collect_data.launch I do not get any error or warning, but the interactive markers do not appear as well.

miguelriemoliveira, 2021-06-18:
Lets zoom at 14h?

MiguelDRPina, 2021-06-18:
Yes, see you then.

MiguelDRPina, 2021-06-18:
Hi @miguelriemoliveira ,

Could you send me the latest bagfile you have for the [ATLASCAR2](https://github.com/lardemua/atlascar2) robotic system?

miguelriemoliveira, 2021-06-18:
Hi @MiguelDRPina ,

here it is:

https://drive.google.com/file/d/1M-vqtvB7q0mfUgIEiWNbMQLLXUxe1b8Y/view?usp=sharing

MiguelDRPina, 2021-06-18:
Hello @miguelriemoliveira ,
I only had time to try to look into the atlascar now, I tried to configure and got the following error:
```
miguel@G50-80:~$ rosrun atlascar2_calibration configure
Loading config file /home/miguel/catkin_ws/src/calibration/atlascar2/atlascar2_calibration/calibration/config.yml
Loading bagfile /home/miguel/bagfiles/atlascar/calibration_5_dez_2019_03m-07m.bag
Parsing description file /home/miguel/catkin_ws/src/calibration/atlascar2/atlascar2_description/urdf/atlascar2.urdf.xacro
Executing command: xacro /home/miguel/catkin_ws/src/calibration/atlascar2/atlascar2_description/urdf/atlascar2.urdf.xacro -o /tmp/description.urdf
Executing command: xacro /home/miguel/catkin_ws/src/calibration/atlascar2/atlascar2_description/urdf/atlascar2.urdf.xacro -o /tmp/description.urdf
Traceback (most recent call last):
  File "/home/miguel/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 115, in <module>
    description = URDF.from_xml_file(urdf_file)  # read teh urdf file
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 620, in from_xml_file
    xml_string = open(file_path, 'r').read()
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/description.urdf'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/miguel/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/miguel/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 118, in <module>
    raise ValueError('Could not parse description file ' + description_file)
ValueError: Could not parse description file /home/miguel/catkin_ws/src/calibration/atlascar2/atlascar2_description/urdf/atlascar2.urdf.xacro
```

Trying to launch the set_initial_estimate, got this error:
```
miguel@G50-80:~$ roslaunch atlascar2_calibration set_initial_estimate.launch 
... logging to /home/miguel/.ros/log/7f3d4930-d07a-11eb-ae0f-e9698ee7e75e/roslaunch-G50-80-23779.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

resource not found: pointgrey_camera_description
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/miguel/ws_moveit/src
ROS path [2]=/home/miguel/catkin_ws/src
ROS path [3]=/opt/ros/noetic/share
when processing file: /home/miguel/catkin_ws/src/calibration/atlascar2/atlascar2_calibration/urdf/initial_estimate.urdf.xacro
RLException: while processing /home/miguel/catkin_ws/src/calibration/atlascar2/atlascar2_calibration/launch/playbag.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '/home/miguel/catkin_ws/src/calibration/atlascar2/atlascar2_calibration/urdf/initial_estimate.urdf.xacro']] returned with code [2]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro $(arg description_file)"/>
The traceback for the exception was written to the log file
```

Any solution?

miguelriemoliveira, 2021-06-18:
Hi @MiguelDRPina ,

this is the problem:

resource not found: pointgrey_camera_description

You must install http://wiki.ros.org/pointgrey_camera_driver

using the command:

sudo apt-get install ros-noetic-pointgrey-camera-driver

MiguelDRPina, 2021-06-19:
Hello @miguelriemoliveira ,

I have tried to install it, but I get this error:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 ros-noetic-moveit-ros-planning-dbgsym : Depends: ros-noetic-moveit-ros-planning (= 1.1.2-1focal.20210424.003930) but 1.1.0-1focal.20210302.142257 is to be installed
 ros-noetic-pointgrey-camera-driver : Depends: ros-noetic-image-exposure-msgs but it is not going to be installed
                                      Depends: ros-noetic-wfov-camera-msgs but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

Trying to find the solution online, I then tried running sudo apt install --fix-broken and got this:
```
miguel@G50-80:~$ sudo apt install --fix-broken
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages were automatically installed and are no longer required:
  fltk1.3-doc fluid gazebo11 gazebo11-common gazebo11-plugin-base gir1.2-gtk-2.0
  hddtemp ignition-tools libavdevice-dev libavfilter-dev libdart-collision-bullet-dev
  libdart-collision-ode-dev libdart-dev libdart-external-ikfast-dev
  libdart-external-odelcpsolver-dev libdart-utils-dev libdart-utils-urdf-dev libdart6
  libdart6-collision-bullet libdart6-collision-ode libdart6-external-odelcpsolver
  libdart6-utils libdart6-utils-urdf libfcl-dev libfcl0.5 libfltk-cairo1.3
  libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3 libfltk1.3-dev
  libfreeimage-dev libgazebo11 libgazebo11-dev libgtk2.0-dev libgts-dev
  libignition-cmake2-dev libignition-common3-av libignition-common3-av-dev
  libignition-common3-core-dev libignition-common3-dev libignition-common3-events
  libignition-common3-events-dev libignition-common3-graphics-dev
  libignition-common3-profiler libignition-common3-profiler-dev
  libignition-fuel-tools4 libignition-fuel-tools4-dev libignition-math6-dev
  libignition-msgs5 libignition-msgs5-dev libignition-tools-dev
  libignition-transport8 libignition-transport8-core-dev libignition-transport8-dev
  libignition-transport8-log libignition-transport8-log-dev libnorm-dev
  liboctomap-dev liboctomap1.9 libpgm-dev libpostproc-dev libprotobuf-dev
  libprotoc-dev libprotoc17 libqwt-qt5-6 libqwt-qt5-dev libsdformat9 libsdformat9-dev
  libsimbody-dev libsimbody3.6 libsodium-dev libtar-dev libtar0 libusb-1.0-0-dev
  libusb-1.0-doc libxmu-dev libxmu-headers libyaml-dev libzip-dev libzip5 libzmq3-dev
  linux-headers-5.4.0-71 linux-headers-5.4.0-71-generic linux-image-5.4.0-71-generic
  linux-modules-5.4.0-71-generic linux-modules-extra-5.4.0-71-generic
  protobuf-compiler python-backports.functools-lru-cache python-bs4 python-html5lib
  python-lxml python-six python-soupsieve python-webencodings python3-opengl
  python3-pyqt5.qtopengl python3-pyqt5.qtwebkit ros-noetic-actionlib-tutorials
  ros-noetic-bond-core ros-noetic-bondpy ros-noetic-camera-calibration
  ros-noetic-common-msgs ros-noetic-common-tutorials
  ros-noetic-compressed-depth-image-transport ros-noetic-compressed-image-transport
  ros-noetic-diagnostic-aggregator ros-noetic-diagnostic-analysis
  ros-noetic-diagnostic-common-diagnostics ros-noetic-diagnostics
  ros-noetic-diff-drive-controller ros-noetic-executive-smach
  ros-noetic-forward-command-controller ros-noetic-gazebo-dev ros-noetic-gazebo-msgs
  ros-noetic-gazebo-plugins ros-noetic-gazebo-ros ros-noetic-gazebo-ros-control
  ros-noetic-gazebo-ros-pkgs ros-noetic-geometry ros-noetic-geometry-tutorials
  ros-noetic-gl-dependency ros-noetic-image-common ros-noetic-image-pipeline
  ros-noetic-image-rotate ros-noetic-image-transport-plugins
  ros-noetic-interactive-marker-tutorials ros-noetic-joint-state-controller
  ros-noetic-laser-assembler ros-noetic-laser-filters ros-noetic-laser-pipeline
  ros-noetic-librviz-tutorial ros-noetic-nodelet-core
  ros-noetic-nodelet-tutorial-math ros-noetic-perception-pcl
  ros-noetic-pluginlib-tutorials ros-noetic-polled-camera
  ros-noetic-position-controllers ros-noetic-qt-dotgraph ros-noetic-ros
  ros-noetic-ros-comm ros-noetic-ros-tutorials ros-noetic-rosbash
  ros-noetic-rosboost-cfg ros-noetic-roscpp-core ros-noetic-roscpp-tutorials
  ros-noetic-roscreate ros-noetic-roslang ros-noetic-roslisp ros-noetic-rosmake
  ros-noetic-rospy-tutorials ros-noetic-rqt-action ros-noetic-rqt-bag-plugins
  ros-noetic-rqt-common-plugins ros-noetic-rqt-console ros-noetic-rqt-dep
[... 33 lines cut ...]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Running sudo apt-get update, got this:
```
miguel@G50-80:~$ sudo apt-get update
Hit:1 http://ppa.launchpad.net/danielrichter2007/grub-customizer/ubuntu bionic InRelease
Hit:2 http://dl.google.com/linux/chrome/deb stable InRelease                          
Hit:3 http://ppa.launchpad.net/maarten-baert/simplescreenrecorder/ubuntu bionic InRelease
Hit:4 http://archive.canonical.com/ubuntu bionic InRelease                            
Hit:5 http://ppa.launchpad.net/peek-developers/stable/ubuntu bionic InRelease         
Get:6 http://packages.ros.org/ros/ubuntu focal InRelease [4676 B]                    
Hit:7 https://packages.microsoft.com/repos/ms-teams stable InRelease     
Err:6 http://packages.ros.org/ros/ubuntu focal InRelease
[... cut ...]
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros/ubuntu focal InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
W: Failed to fetch http://packages.ros.org/ros/ubuntu/dists/focal/InRelease  The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

MiguelDRPina, 2021-06-19:
Nevermind, I solved it.

MiguelDRPina, 2021-06-19:
> Hi @MiguelDRPina ,
> 
> this is the problem:
> 
> resource not found: pointgrey_camera_description
> 
> You must install http://wiki.ros.org/pointgrey_camera_driver
> 
> using the command:
> 
> sudo apt-get install ros-noetic-pointgrey-camera-driver

Hi @miguelriemoliveira ,
I installed that and I still get the same error.

MiguelDRPina, 2021-06-19:
Can we meet sometime today? In the midst of trying to make this work, I must have deleted something, now I can't even run the catkin_make command...

miguelriemoliveira, 2021-06-19:
Hi Miguel

Today i have family for lunch. We can talk at night or the end of the
afternoon.  Send me an email around 18h ok?

MiguelDRPina, 2021-06-19:
Ok, I will, thank you.

closed 2021-09-18 by miguelriemoliveira