lardemua/atom #303: Error in batch execution

danifpdra, 2021-05-07:
Hi @miguelriemoliveira ,

I'm getting this error when trying to execute the batch execution. Can you help me out here?

`Traceback (most recent call last):
  File "./batch_execution", line 95, in <module>
    main()
  File "./batch_execution", line 83, in main
    if not os.path.exists(file):
  File "/usr/lib/python3.8/genericpath.py", line 19, in exists
    os.stat(path)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType`

This is what I'm trying to run:

``
    - name: "partial_0"
      cmd: "{{ cmd_prefix }} -csf 'lambda name: int(name) in [0,3,5,7,9,10,11,12,15,16]' -nig 0.05 0.02 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro'
        -
    - name: "partial_1"
      cmd: "{{ cmd_prefix }} -csf 'lambda name: int(name) in [0,3,5,7,9,10,11,12,15,16,1]' -nig 0.05 0.02 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro'

    - name: "partial_3"
      cmd: "{{ cmd_prefix }} -csf 'lambda name: int(name) in [0,3,5,7,9,10,11,12,15,16,1,2,4]' -nig 0.05 0.02 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro'

    - name: "partial_6"
      cmd: "{{ cmd_prefix }} -csf 'lambda name: int(name) in [0,3,5,7,9,10,11,12,15,16,1,2,4,6,8,13]' -nig 0.05 0.02 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro'

    - name: "partial_9"
      cmd: "{{ cmd_prefix }} -csf 'lambda name: int(name) in [0,3,5,7,9,10,11,12,15,16,1,2,4,6,8,13,14,19,24]' -nig 0.05 0.02 -ss 1"
      files_to_collect:
        - '{{ dataset_path }}/atom_calibration.json'
        -  '/tmp/optimized.urdf.xacro' ``

danifpdra, 2021-05-10:
here https://github.com/miguelriemoliveira/mmtbot/blob/main/mmtbot_experiments/01_initial_estimate_vs_error/batch_mmtbot_partial.yml

miguelriemoliveira, 2021-05-10:
@danifpdra if you run with -v (verbose ) what do you see? (please git pull)

miguelriemoliveira, 2021-05-10:
how are you running the script ( BTW this should always be in the initial message of the issue)? 

I run with

`/home/mike/catkin_ws/src/calibration/mmtbot/mmtbot_experiments/01_initial_estimate_vs_error && rosrun atom_evaluation batch_execution -f batch_mmtbot.yml -v`

miguelriemoliveira, 2021-05-10:
Wait, you give two different outputs ? Which one is it?

miguelriemoliveira, 2021-05-10:
Please close if solved.

danifpdra, 2021-05-10:
I'm still getting the same problem after deleting the dash.

closed 2021-05-10 by danifpdra

miguelriemoliveira, 2021-05-10:
Great!