lardemua/atom #305: Unable to commit

MiguelDRPina, 2021-06-11:
Hello @miguelriemoliveira ,

I decided to come work just for a little bit on some pending details about the interface and I have tried to commit, without success. And it turns out, the last time I came here (a few days ago), it apparently did not commit as well, but I hadn't noticed.

[image]

I don't know if the permissions were changed or anything. Could you please check that?

miguelriemoliveira, 2021-06-11:
> Assigned #305 <https://github.com/lardemua/atom/issues/305> to

Hi Miguel

Did you pull? Do you have conflicts?

I can check it tomorrow.

Miguel

MiguelDRPina, 2021-06-11:
Yes, I did pull. But I keep getting that error saying I do not have permission.

MiguelDRPina, 2021-06-11:
Don't know if you changed anything or not, but somehow I was able to push now. Thank you.

miguelriemoliveira, 2021-06-11:
Changed nothing. Great!

closed 2021-06-12 by MiguelDRPina