lardemua/atom #306: Integration with RViz Panel - Data Collect - Type of sensor and marker representing it

MiguelDRPina, 2021-06-12:
Hi @miguelriemoliveira ,

For the data collection tab, I have a combobox to select which lidar marker I will be moving, like this:

[image]

The problem right now is that it has all the sensors in that combobox. My question is, how do I know which of my sensors is a lidar and which one is a camera? Obviously I know for the mmtbot, but I mean in a general sense, when I go through the list of all sensors of the robotic system, what should I look for to determine if that sensor is a lidar in order for me to put it on the combobox?

Also, the same way it was done in the Initial Estimate (transform between parent_link and child_link), I would like to get the current position of the marker(s) in this tab as I alternate through each sensor in the combobox. What are the links in this case?

miguelriemoliveira, 2021-06-12:
Hi @MiguelDRPina ,

**First question:**

> The problem right now is that it has all the sensors in that combobox. My question is, how do I know which of my sensors is a lidar and which one is a camera? Obviously I know for the mmtbot, but I [...]

You have to look and the calibration.yaml, and for each sensor there is a topic_name field for each sensor, e.g.:

```yaml
sensors:
    #       topic_name:
    #           Name of the ROS topic that contains the data produced by this sensor.
    #           If you are calibrating an camera, you should use the raw image produced by the
    #           sensors. Aditionally, it the topic is an image it will automatically use the
    #           respective `camera_info` topic.
    # Example:
    hand_camera:
        link: "hand_camera_rgb_optical_frame"
        parent_link: "ee_link"
        child_link: "hand_camera_link"
        topic_name: "/hand_camera/rgb/image_raw"
```

Now given a topic you can extract the type of message that is being published on that topic. In python this is quite easy to do since python has introspection features, e.g.:

https://github.com/lardemua/atom/blob/bef38d1a5fd868b63d2385a15aabbe1bfcc981e9/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py#L85-L93

In cpp this is much harder, but after searching around I found this

https://gist.github.com/bechu/6222399

So a code like:

```cpp
for (ros::master::V_TopicInfo::iterator it = master_topics.begin() ; it != master_topics.end(); it++) {
    const ros::master::TopicInfo& info = *it;
    std::cout << "Topic : " << it - master_topics.begin() << ": " << info.name << " -> " << info.datatype <<       std::endl;
}
```

Should give you all types for all topics. Now the ones that should have a marker are those which are of message type LaserScan of PointCloud2.

This is how the data_collector python node is doing it:

https://github.com/lardemua/atom/blob/bef38d1a5fd868b63d2385a15aabbe1bfcc981e9/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py#L145-L201

miguelriemoliveira, 2021-06-12:
Second question:

> Also, the same way it was done in the Initial Estimate (transform between parent_link and child_link), I would like to get the current position of the marker(s) in this tab as I alternate through each sensor in the combobox. What are the links in this case?

It should be the topic of the sensor data message, which you find in the calibration.yaml, concatenated with a "/labeled".
Check this marker creation for PointCloud2 message types:

https://github.com/lardemua/atom/blob/bef38d1a5fd868b63d2385a15aabbe1bfcc981e9/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py#L182-L186

MiguelDRPina, 2021-06-13:
Hello @miguelriemoliveira ,

Thank you. I really did not have time today, but tomorrow I will try it and then say something.

MiguelDRPina, 2021-06-14:
Hello @miguelriemoliveira ,

As for the first question, thanks, it worked well. The second one I am not so sure I understood. I launched the collect_data launchfile and listened to the /3dlidar/points/labeled topic in the terminal, and it was constantly receiving the following message:
```
header: 
  seq: 163
  stamp: 
    secs: 806
    nsecs: 290000000
  frame_id: "3dlidar"
height: 1
width: 333
fields: 
  - 
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  - 
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  - 
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  - 
    name: "rgba"
    offset: 12
    datatype: 6
    count: 1
is_bigendian: False
point_step: 16
row_step: 5328
data: [235, 123, 158, 63, 17, 230, 104, 63, 211, 48, 121, 190, 255, 255, 255, 150, 248, 139, 153, 63, 242, 147, 115, 63, 6, 83, 120, 190, 255, 255, 255, 150, 241, 155, 153, 63, 4, 92,................................]
```

Is this what you were referring to? Because I have changed the markers' pose in rviz and none of the information really changed.

miguelriemoliveira, 2021-06-14:
Hi @MiguelDRPina ,

sorry, I think I forgot about your question in the middle of the answer : ) .

Just listem to the topic rostopic /data_labeler/update, e.g.,

`echo /data_labeler/update`

will give you a lot of markers, you search for the one with the sensor name. Then you can get the pose of the marker w.r.t the frame_id (which is typically the sensor's data frame)

```yaml
- 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "3dlidar"
    pose: 
      position: 
        x: 1.0309501595303492
        y: 1.3522961691551403
        z: -0.16976502583489805
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    name: "3dlidar"
    description: "3dlidar_labeler"
    scale: 0.5
    menu_entries: 
      - 
        id: 1
        parent_id: 0
        title: "Save collection"
        command: ''
        command_type: 0
    controls: 
      - 
        name: ''
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 0.0
        orientation_mode: 0
        interaction_mode: 7
        always_visible: True
        markers: 
```

MiguelDRPina, 2021-06-14:
Hello @miguelriemoliveira , 

I have been trying to do this one, but I am having trouble creating the subscriber and cannot figure out why. I have tried several different ways, but always get some kind of error. Could you take a quick look into the code I pushed? The subscriber is being created in the header file (line 28) and the callback function in the data_collect.cpp (line 23), but it doesn't do anything so far.

I only want to make the subscriber work, I think (hope) I can manage after that.

miguelriemoliveira, 2021-06-14:
> I have been trying to do this one, but I am having trouble creating the subscriber and cannot figure out why. I have tried several different ways, but always get some kind of error. Could you take a [...]

So you say that the on message received callback **CalibrationPanel::positionCallback()** is not being called?

Did you put a PFLN there to make sure its not going inside the function?

Also, are you sure that messages are being published on that topic? Did you 

`rostopic echo /data_labeler/update`

Do you get any messages?

MiguelDRPina, 2021-06-14:
> So you say that the on message received callback **CalibrationPanel::positionCallback()** is not being called?
>
> Did you put a PFLN there to make sure its not going inside the function?
>
> Also, are [...]

I did not put a PFLN because the code does not compile, always get an error. I have checked for other examples and tried to create the subscriber other ways but I get different errors everytime.

That topic does receive messages, I checked first.

miguelriemoliveira, 2021-06-14:
Another suggestion. When using ros subscribers we should declare the class properties  as pointer to the variables we want, instead of the variables. We do this to make sure the subscribers are only created when the class constructor is called.

So instead of

```cpp
class CalibrationPanel: public rviz::Panel
{
  Q_OBJECT
 public:
  // ...
  ros::Subscriber data_collect_sub = nh.subscribe("/data_labeler/update", 1000, CalibrationPanel::positionCallback);

   CalibrationPanel(QWidget* parent = nullptr);
  

  void onInitialize()
  {
      // do some stuff

   }
```

we should:

```cpp
class CalibrationPanel: public rviz::Panel
{
  Q_OBJECT
 public:
  // ...
  boost::shared_ptr<ros::Subscriber> data_collect_sub;

  CalibrationPanel(QWidget* parent = nullptr);

  void onInitialize()
  {
      // do some stuff
     // Configure the subscriber     

     data_collect_sub  = boost::shared_ptr<ros::Subscriber> new(nh.subscribe("/data_labeler/update", 1000, CalibrationPanel::positionCallback));
   }
```

Not sure about the sintax, but I can try in the code if you want ...

> I did not put a PFLN because the code does not compile, always get an error.

What error?

MiguelDRPina, 2021-06-14:
The error was something along the lines of 'ros error: invalid use of non-static member function atom_rviz_plugin:CalibrationPanel:positionCallback'

MiguelDRPina, 2021-06-15:
Hello @miguelriemoliveira ,
Tried it again this morning. Still can't create a subscriber.

miguelriemoliveira, 2021-06-15:
Hi Miguel,

we can work this afternoon ... 14h?
*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

MiguelDRPina, 2021-06-15:
Hold on, I am seeing some light. It compiled.

MiguelDRPina, 2021-06-15:
I have already subscribed to the topic. But now I am not able to retrieve the message I want. Is it ok if we still meet this afternoon at 14h @miguelriemoliveira ?

miguelriemoliveira, 2021-06-15:
Better 14h10 ok?

MiguelDRPina, 2021-06-15:
Of course. See you soon.

MiguelDRPina, 2021-06-15:
Hey @miguelriemoliveira , I am already in the usual meeting room.

MiguelDRPina, 2021-06-15:
Message of topic /data_labeler/update:

``` yaml
server_id: "/collect_data"
seq_num: 587
type: 1
markers: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "3dlidar"
    pose: 
      position: 
        x: 1.0259728505032295
        y: 1.3491220353278954
        z: -0.18161773481469926
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    name: "3dlidar"
    description: "3dlidar_labeler"
    scale: 0.5
    menu_entries: 
      - 
        id: 1
        parent_id: 0
        title: "Save collection"
        command: ''
        command_type: 0
    controls: 
      - 
        name: ''
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 0.0
        orientation_mode: 0
        interaction_mode: 7
        always_visible: True
        markers: 
          - 
            header: 
              seq: 0
              stamp: 
                secs: 0
                nsecs:         0
              frame_id: ''
            ns: ''
            id: 0
            type: 2
            action: 0
            pose: 
              position: 
                x: 0.0
                y: 0.0
                z: 0.0
              orientation: 
                x: 0.0
                y: 0.0
                z: 0.0
                w: 0.0
            scale: 
              x: 0.15
              y: 0.15
              z: 0.15
            color: 
              r: 1.0
              g: 0.0
              b: 0.0
              a: 0.20000000298023224
            lifetime: 
              secs: 0
              nsecs:         0
            frame_locked: False
            points: []
            colors: []
            text: ''
            mesh_resource: ''
            mesh_use_embedded_materials: False
        independent_marker_orientation: False
        description: ''
      - 
        name: "move_x"
        orientation: 
          x: 1.0
          y: 0.0
          z: 0.0
          w: 1.0
        orientation_mode: 1
        interaction_mode: 3
        always_visible: False
        markers: []
        independent_marker_orientation: False
        description: ''
      - 
        name: "move_y"
        orientation: 
          x: 0.0
          y: 1.0
          z: 0.0
          w: 1.0
        orientation_mode: 1
        interaction_mode: 3
        always_visible: False
        markers: []
        independent_marker_orientation: False
        description: ''
      - 
        name: "move_z"
        orientation: 
          x: 0.0
          y: 0.0
          z: 1.0
          w: 1.0
        orientation_mode: 1
        interaction_mode: 3
        always_visible: False
        markers: []
        independent_marker_orientation: False
        description: ''
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "world"
    pose: 
      position: 
        x: 1.0
        y: 0.0
        z: 1.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    name: "menu"
    description: "menu"
    scale: 0.20000000298023224
    menu_entries: 
      - 
        id: 1
        parent_id: 0
        title: "Save collection"
        command: ''
        command_type: 0
    controls: 
      - 
        name: ''
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 0.0
        orientation_mode: 0
        interaction_mode: 7
        always_visible: True
        markers: 
          - 
            header: 
              seq: 0
              stamp: 
                secs: 0
                nsecs:         0
              frame_id: ''
            ns: ''
            id: 0
            type: 2
            action: 0
            pose: 
              position: 
                x: 0.0
                y: 0.0
                z: 0.0
              orientation: 
                x: 0.0
                y: 0.0
                z: 0.0
                w: 0.0
            scale: 
              x: 0.13999999999999999
              y: 0.13999999999999999
              z: 0.13999999999999999
            color: 
              r: 0.0
              g: 1.0
              b: 0.0
              a: 0.20000000298023224
            lifetime: 
              secs: 0
              nsecs:         0
            frame_locked: False
            points: []
            colors: []
            text: ''
            mesh_resource: ''
            mesh_use_embedded_materials: False
        independent_marker_orientation: False
        description: ''
      - 
        name: "move_x"
        orientation: 
          x: 1.0
          y: 0.0
          z: 0.0
          w: 1.0
        orientation_mode: 1
        interaction_mode: 3
        always_visible: False
        markers: []
        independent_marker_orientation: False
        description: ''
      - 
        name: "move_z"
        orientation: 
          x: 0.0
          y: 1.0
          z: 0.0
          w: 1.0
        orientation_mode: 1
        interaction_mode: 3
        always_visible: False
        markers: []
        independent_marker_orientation: False
        description: ''
      - 
        name: "move_y"
        orientation: 
          x: 0.0
          y: 0.0
          z: 1.0
          w: 1.0
        orientation_mode: 1
        interaction_mode: 3
        always_visible: False
        markers: []
        independent_marker_orientation: False
        description: ''
poses: []
erases: []
```

closed 2021-06-15 by MiguelDRPina