lardemua/atom #308: Python node for tf trees

MiguelDRPina, 2021-06-18:
Hi @miguelriemoliveira ,

Could you please send me the python node you showed me for making tf trees?

miguelriemoliveira, 2021-06-18:
I sent it to the email. Did you receive?

*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

MiguelDRPina, 2021-06-18:
Was just about to answer that email. The file is there, but for some reason I can't download it or open it.

miguelriemoliveira, 2021-06-18:
Here it is:

```python
#!/usr/bin/env python

# stdlib
import matplotlib
from colorama import Style, Fore
from graphviz import Digraph
import dot2tex
from matplotlib import cm

if __name__ == "__main__":
    cm1 = '#7fc97f'
    cm2 = '#beaed4'
    cm3 = '#fdc086'
    cm4 = '#386cb0'
    cm5 = '#8B4513'
    black = matplotlib.colors.rgb2hex([0, 0, 0])  # black by default
    white = matplotlib.colors.rgb2hex([1, 1, 1])
    grey = matplotlib.colors.rgb2hex([0.85, 0.85, 0.85])
    red = matplotlib.colors.rgb2hex([1, 0, 0])
    blue = matplotlib.colors.rgb2hex([0, 0, 1])
    dark_green = matplotlib.colors.rgb2hex([0, 0.6, 0])

    def get_node_label(top, top_color, bottom, bottom_color, top_size=16, bottom_size=16):
        if not bottom is None:
            s = '<<table border="0" cellborder="0" cellpadding="3" bgcolor="invis">' + \
                '<tr><td align="center" colspan="2"><font point-size="' + str(top_size) + '" color="' + top_color + '">' + \
                top + '</font></td></tr>' + \
                '<tr><td align="center" colspan="2"><font point-size="' + str(bottom_size) + '" color="' + \
                bottom_color + '">' + bottom + \
                '</font></td></tr>' + \
                '</table>>'
        else:
            s = '<<table border="0" cellborder="0" cellpadding="3" bgcolor="invis">' + \
                '<tr><td align="center" colspan="2"><font point-size="' + str(
                top_size) + '" color="' + top_color + '">' + \
                top + '</font></td></tr>' + \
                '</table>>'
        return s

    def get_edge_label(top, top_color, bottom, bottom_color, font_size=12):
        if top is None:
            s = ''
        elif not bottom is None:
            s = '<<table border="0" cellborder="0" cellpadding="0" bgcolor="invis">' + \
                '<tr><td align="center" colspan="2">' + \
                '<font point-size="' + str(font_size) + '" color="' + top_color + '"> ' + top + '</font> ' + \
                '<font point-size="' + str(font_size) + '" color="black">to</font> ' + \
                '<font point-size="' + str(font_size) + '" color="' + bottom_color + '">' + bottom + '</font> ' + \
                '</td></tr>' + \
                '<tr><td align="center" colspan="2"><font point-size="' + str(
                font_size) + '" color="black">calibration</font></td></tr>' + \
                '</table>>'
        else:
            s = '<<table border="0" cellborder="0" cellpadding="0" bgcolor="invis">' + \
                '<tr><td align="center" colspan="2">' + \
                '<font point-size="' + str(font_size) + '" color="' + top_color + '"> ' + top + '</font> ' + \
                '</td></tr>' + \
                '</table>>'

        return s

    # Graph 1
    g = Digraph('G', filename='graph1', directory='/tmp/', graph_attr={'title': 'asdads'})  # create a graph
    g.attr('node', shape='ellipse', style='solid')

    # Nodes

[... 119 lines cut ...]
    #        arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    # g.edge('wrist 1', 'wrist 2', label=get_edge_label('Dynamic', 'blue', None, 'red'), color=blue, style='solid',
    #        arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    # g.edge('wrist 2', 'wrist 3', label=get_edge_label('Dynamic', 'blue', None, 'red'), color=blue, style='solid',
    #        arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    # g.edge('wrist 3', 'end effector', label=get_edge_label('Dynamic', 'blue', None, 'red'), color=blue, style='solid',
    #        arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    # g.edge('shoulder', '...', label=get_edge_label('Dynamic', 'blue', None, 'red'), color=blue, style='solid',
    #        arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    g.edge('shoulder', '...', label=get_edge_label('Dynamic', 'blue', None, 'red'), color=blue, style='solid',
           arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    g.edge('...', 'wrist', label=get_edge_label('Dynamic', 'blue', None, 'red'), color=blue, style='solid',
           arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    g.edge('wrist', 'end effector', label=get_edge_label(None, 'blue', None, 'red'), color=black, style='solid',
           arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    g.edge('end effector', 'hand camera', label=get_edge_label(None, 'black', None, 'red'), color=black, style='solid',
           arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})
    g.edge('hand camera', 'hand camera optical', label=get_edge_label(None, 'black', None, 'red'), color=black, style='solid',
           arrowhead='normal', _attributes={'penwidth': '1', 'fontcolor': black})

    g.view()
    g.render(filename='graph1', directory='./', cleanup=True)

    exit(0)
```

closed 2021-06-18 by MiguelDRPina