lardemua/atom #309: Collect results with the atlascar2 for IF paper.

miguelriemoliveira, 2021-07-12:
Hi @danifpdra ,

as discussed by phone, you should:

1. get the bag file from the atlascar2

miguelriemoliveira, 2021-07-12:
I think this will get you started. If you need my help let me know. 

An important tip: because the detection of chessboards is much slower when compared to the charuco, you should reduce the bage playback rate when creating a collection. Something like this:

roslaunch atlascar2_calibration collect_data.launch output_folder:=/home/mike/datasets/atlascar2/atlascar2-1 overwrite:=true bag_rate:=0.2

danifpdra, 2021-07-13:
Hi @miguelriemoliveira ,

I've just tried to collect data with this dataset and after saving a bunch of collections and going through the entire bag file the output folder is empty...

When I run 

`roslaunch atlascar2_calibration collect_data.launch output_folder:=$ATOM_DATASETS/atlascar2_train_1 bag_rate:=0.5 `

I get the error 

```
[ERROR] [1626184826.863904, 1575573753.854089]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7f992d51bd30>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 219, in sensorDataReceivedCallback
    self.labelData  # label the data
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 366, in labelData
    self.pattern.drawKeypoints(image, result)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/patterns.py", line 43, in drawKeypoints
    cv2.drawMarker(image, tuple(point[0]), (0, 0, 255), cv2.MARKER_CROSS, 14)
cv2.error: OpenCV(4.5.2) :-1: error: (-5:Bad argument) in function 'drawMarker'
> Overload resolution failed:
>  - Can't parse 'position'. Sequence item with index 0 has a wrong type
>  - Can't parse 'position'. Sequence item with index 0 has a wrong type
```

Is this why?

danifpdra, 2021-07-13:
Ok, time to walk my dog!

danifpdra, 2021-07-14:
Hi @miguelriemoliveira ,

I'm having huge error when running the camera_to_camera_evaluation.py script with the results I got from the sc_stereo_calib.py :-(

---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
--------------------------------------------------------------------------
 All  275.2349  148.0138  355.5390  53.0881   19.3423  9258790.9486  1.8592  
---------------------------------------------------------------------------

miguelriemoliveira, 2021-07-14:
I need a bit more context. Using ATOM do you get nice results? Which?
Because that validates both ATOM and the evaluation script.

Then you say the opencv stereo is giving bad results, is that it?
Does the opencv calibration give some feedback before running the evaluation? Which is the result?

@aaguiar96 should be in on this since he's the one who worked on the stereo script (I think Afonso created it but André revised it in September, right?)

danifpdra, 2021-07-14:
Hi @miguelriemoliveira,

ATOM's results are good:

`rosrun atom_evaluation camera_to_camera_evalutation.py -train_json $ATOM_DATASETS/atlascar2_train_1/atom_calibration.json -test_json $ATOM_DATASETS/atlascar2_test_1/data_collected.json -ts top_left_camera -ss top_center_rgbd_camera_rgb `

```---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
 All   3.2830    3.3266    2.3136    3.8323    1.2133    0.0448    0.0899  
--------------------------------------------------------------------------- ```

`rosrun atom_evaluation camera_to_camera_evalutation.py -train_json $ATOM_DATASETS/atlascar2_train_1/atom_calibration.json -test_json $ATOM_DATASETS/atlascar2_test_1/data_collected.json -ss top_right_camera -ts top_center_rgbd_camera_rgb`

```---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
 All   2.3751    1.6692    2.5241    1.7741    0.7334    0.0737    0.0959  
--------------------------------------------------------------------------- ```

`rosrun atom_evaluation camera_to_camera_evalutation.py -train_json $ATOM_DATASETS/atlascar2_train_1/atom_calibration.json -test_json $ATOM_DATASETS/atlascar2_test_1/data_collected.json -ss top_left_camera -ts top_right_camera `

```---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
 All   2.9280    1.5601    1.7455    2.5387    3.1684    0.0443    0.0372  
--------------------------------------------------------------------------- ```

To evaluate the opencv I did this (example for one of the camera pairs)

`rosrun atom_evaluation cv_stereo_calib.py -json /home/daniela/datasets/atlascar2_train_1/data_collected.json -rc top_left_camera -lc top_center_rgbd_camera_rgb`

And then this

`rosrun atom_evaluation camera_to_camera_evalutation.py -train_json $ATOM_DATASETS/atlascar2_train_1/cv_calibration_left_to_center.json -test_json $ATOM_DATASETS/atlascar2_test_1/data_collected.json -ss top_left_camera -ts top_center_rgbd_camera_rgb`

(I tried to switch the ss with the ts because I didn't know which was the corrected order but both have bad results)

When I run the cv_stereo_calib.py I don't get any errors...When I run the camera_to_camera_evaluation.py I just get some outputs sayings its removing some collections because the pattern was not found in all sensors.

miguelriemoliveira, 2021-07-14:
OK, can you send me the datasets so I put them online? Then we can zoom ...

danifpdra, 2021-07-14:
here: [atlascar2_datasets.zip](https://github.com/lardemua/atom/files/6815272/atlascar2_datasets.zip)

danifpdra, 2021-07-14:
Hi @miguelriemoliveira,

I have already added the results to the camera to camera table in the paper.

miguelriemoliveira, 2021-07-14:
Thanks. That's it.

@eupedrosa , you can proceed with the analysis of the results.

closed 2021-07-14 by miguelriemoliveira