lardemua/atom #311: Create an atom dataset inspection tool

miguelriemoliveira, 2021-07-15:
Hi @danifpdra ,

I am doing this because I need to fill in the paper's table 2. 

Anything you can remember that would be interesting to see in a dataset let me know ...

danifpdra, 2021-07-15:
Hi @miguelriemoliveira,

train-1 : 39 collections, 32 incomplete 

test-1 : 21 collections, 18 incomplete

I'm not sure if the -rpd flag is not working or if there are no partial detections. When I calibrate with or without -rpd it uses the same number of collections.

miguelriemoliveira, 2021-07-15:
Thanks, can you check the style of table 2 of the paper., in particular the last collumn.

Some datasets say the partial collections, others dont.

Some have X collections, Y incimplete, Z partial
some others have X collections, Y incimplete Z of which are partial

Can you homogeneize please?

danifpdra, 2021-07-15:
done

miguelriemoliveira, 2021-07-15:
Thanks. The paper is getting better and better. Its great that we were able to get these new results for the atlascar2.

danifpdra, 2021-07-15:
I agree. Fingers crossed that it's accepted.

miguelriemoliveira, 2021-11-09:
Implemented a script that looks into the dataset and prints out some useful information.

[image]

@danifpdra can you test it? Something like this:

    rosrun atom_calibration inspect_dataset -json data_collected.json

eupedrosa, 2021-11-09:
Now, this is [rich](https://rich.readthedocs.io/en/latest/tables.html):exclamation:

miguelriemoliveira, 2021-11-09:
Thanks. Next time I will use that one.

danifpdra, 2021-11-10:
This works. Closing now.

closed 2021-11-10 by danifpdra