lardemua/atom #312: Move optimization utils inside ATOM

miguelriemoliveira, 2021-08-05:
Not sure whether to do this or not, but it would make things easier for people to use ATOM.

tiagomfmadeira, 2021-08-05:
In my opinion, we should add OptimizationUtils as a git submodule in the ATOM repo.

miguelriemoliveira, 2021-08-05:
We've tried this before and it did not work well. Moreover, in your Phd you probably are going to adjust the optimization utils code and it would be better if the calibration is not depending on these experiments ...

tiagomfmadeira, 2021-08-06:
How so? Did `git pull --recurse-submodules` work unexpectedly?
Well, we could have a branch in OptimizationUtils for experiments within my PhD... and some tests to make sure the integration didn't break A̶T̶O̶M̶ the main functionalities, when we merged to main.
Still, I'm not against the idea of integrating the needed functionalities into ATOM instead. If it becomes too cumbersome to manage OptimizationUtils in a way it can be reused by multiple projects, maybe we should not go down that road.

eupedrosa, 2021-08-06:
In the future, ATOM should be published as a ROS package. Reducing the dependencies is a good ideia. :+1:

miguelriemoliveira, 2022-05-04:
Hi @danifpdra and @eupedrosa ,

I have a working solution for #312. Can you test it before we consider the integration into ATOM? A lot of stuff changes...

miguelriemoliveira, 2022-05-14:
Implemented. Could have some bugs but lets see what happens.

closed 2022-05-14 by miguelriemoliveira