lardemua/atom #316: Verify the parent and child links for all sensors in the calibration.yaml

miguelriemoliveira, 2021-09-16:
This is important to verify the validity of the configuration file. 

The parent - child pair must be an existing atomic transformation, otherwise the configuration is invalid.

This should be checked in the configure_calibration_pkg.

@danifpdra can you do this? You know how to?

danifpdra, 2021-09-16:
I can do it but I don't know how.

I already solved the previous problem tho. We are calibrating successfully!

miguelriemoliveira, 2021-09-16:
> I can do it but I don't know how.

let me try to help you start. Here's where we are already doing some verifications:

https://github.com/lardemua/atom/blob/06de6901b5c7f93d72411c079d9764488928a3e4/atom_calibration/scripts/configure_calibration_pkg#L160-L243

So we should add one more that checks, for each sensor in the calibration file, if the parent to child transformation exists in the xacro tf treem and if it does if they are directly connected (to ensure they are atomic).

Here we are getting the path from A to B, which returns a list. If this list has lenght larger than 2, it means A and B are not directly connected.

What do you think?

danifpdra, 2021-09-16:
OK, I will do that. No problem!

commit 51d5ae7e (2022-04-29): Verify the parent and child links for all sensors in the calibration.yaml Fix #316
commit 68c299bb (2022-04-29): Merge pull request #435 from lardemua:miguelriemoliveira/issue316

Verify the parent and child links for all sensors in the calibration.yaml Fix #316
closed 2022-04-29 by miguelriemoliveira