lardemua/atom #317: Initial Guess noise is different from collection to collection

miguelriemoliveira, 2021-09-19:
@hi @eupedrosa ,

I just found out that the -nig mechanism randomizes noise for sensors, independently for each collection.

I think that should not be the case, we should randomize one sensor pose and then apply it to all collections, right?

FYI @danifpdra 

My experiment, if I print the transforms in a dataset inside the calibrate script (after the nig mechanism), with **-nig 0.0 0.0** I get

```
world to lidar_1:
Collection 0: 
[[-7.07106781e-01  8.65956056e-17 -7.07106781e-01 -9.81657926e-01]
 [ 8.65956056e-17  1.00000000e+00  3.58690743e-17 -9.00000000e-02]
 [ 7.07106781e-01 -3.58690743e-17 -7.07106781e-01  2.30834207e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
Collection 1: 
[[-7.07106781e-01  8.65956056e-17 -7.07106781e-01 -9.81657926e-01]
 [ 8.65956056e-17  1.00000000e+00  3.58690743e-17 -9.00000000e-02]
 [ 7.07106781e-01 -3.58690743e-17 -7.07106781e-01  2.30834207e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
Collection 2: 
[[-7.07106781e-01  8.65956056e-17 -7.07106781e-01 -9.81657926e-01]
 [ 8.65956056e-17  1.00000000e+00  3.58690743e-17 -9.00000000e-02]
 [ 7.07106781e-01 -3.58690743e-17 -7.07106781e-01  2.30834207e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
Collection 3: 
[[-7.07106781e-01  8.65956056e-17 -7.07106781e-01 -9.81657926e-01]
 [ 8.65956056e-17  1.00000000e+00  3.58690743e-17 -9.00000000e-02]
 [ 7.07106781e-01 -3.58690743e-17 -7.07106781e-01  2.30834207e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
Collection 6: 
[[-7.07106781e-01  8.65956056e-17 -7.07106781e-01 -9.81657926e-01]
 [ 8.65956056e-17  1.00000000e+00  3.58690743e-17 -9.00000000e-02]
 [ 7.07106781e-01 -3.58690743e-17 -7.07106781e-01  2.30834207e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
```

and using **-nig 0.05 0.02**

```
world to lidar_1:
Collection 0: 
[[-0.72110656 -0.01385556 -0.69268561 -0.97319017]
 [ 0.01385556  0.99931165 -0.03441295 -0.04551653]
 [ 0.69268561 -0.03441295 -0.72041821  2.28936527]
 [ 0.          0.          0.          1.        ]]
Collection 1: 
[[-0.72110656 -0.01385556 -0.69268561 -1.01643956]
 [ 0.01385556  0.99931165 -0.03441295 -0.0563579 ]
 [ 0.69268561 -0.03441295 -0.72041821  2.30224063]
 [ 0.          0.          0.          1.        ]]
Collection 2: 
[[-0.69282417 -0.01442117 -0.72096234 -1.01252303]
 [ 0.01442117  0.99932296 -0.03384746 -0.06581622]
 [ 0.72096234 -0.03384746 -0.69214713  2.27839189]
 [ 0.          0.          0.          1.        ]]
Collection 3: 
[[-0.72110656  0.01385556 -0.69268561 -0.97426852]
 [-0.01385556  0.99931165  0.03441295 -0.0930034 ]
 [ 0.69268561  0.03441295 -0.72041821  2.25849817]
 [ 0.          0.          0.          1.        ]]
Collection 6: 
[[-0.69282417  0.01442117 -0.72096234 -1.02300233]
 [ 0.01442117  0.99987715  0.00614188 -0.06180466]
 [ 0.72096234 -0.00614188 -0.69294703  2.30169352]
 [ 0.          0.          0.          1.        ]]
```

miguelriemoliveira, 2021-09-19:
I think I solved it (it was easier than anticipated).

Here's the new addNoise function:

```python
def addNoiseToInitialGuess(dataset, args, selected_collection_key):
    """
    Adds noise
    :param dataset:
    :param args: Makes use of nig, i.e., the amount of noise to add to the initial guess atomic transformations to be
                 calibrated
    """
    if args['sample_seed'] is not None:
        np.random.seed(args['sample_seed'])

    nig_trans = args['noisy_initial_guess'][0]
    nig_rot = args['noisy_initial_guess'][1]

    for sensor_key, sensor in dataset['sensors'].items():
        if sensor_key == dataset['calibration_config']['anchored_sensor']:
            continue

        calibration_child = sensor['calibration_child']
        calibration_parent = sensor['calibration_parent']
        tf_link = generateKey(calibration_parent, calibration_child, suffix='')

        # Get original transformation
        quat = dataset['collections'][selected_collection_key]['transforms'][tf_link]['quat']
        translation = dataset['collections'][selected_collection_key]['transforms'][tf_link]['trans']
        euler_angles = tf.transformations.euler_from_quaternion(quat)

        # Add noise to the 6 pose parameters
        v = np.random.uniform(-1.0, 1.0, 3)
        v = v / np.linalg.norm(v)
        new_translation = translation + v * nig_trans

        v = np.random.choice([-1.0, 1.0], 3) * nig_rot
        new_angles = euler_angles + v

        # Replace the original atomic transformations by the new noisy ones
        new_quat = tf.transformations.quaternion_from_euler(new_angles[0], new_angles[1], new_angles[2])
        dataset['collections'][selected_collection_key]['transforms'][tf_link]['quat'] = new_quat
        dataset['collections'][selected_collection_key]['transforms'][tf_link]['trans'] = list(new_translation)

        # Copy randomized transform to all collections
        for collection_key, collection in dataset['collections'].items():
            dataset['collections'][collection_key]['transforms'][tf_link]['quat'] = dataset['collections'][selected_collection_key]['transforms'][tf_link]['quat']
            dataset['collections'][collection_key]['transforms'][tf_link]['trans'] = dataset['collections'][selected_collection_key]['transforms'][tf_link]['trans']
```

eupedrosa, 2021-09-20:
Should we consider this a bug? We used this for your article. Redo the experiments??? :wink:

miguelriemoliveira, 2021-09-20:
I think not because the system will use one collection, the selected collection key and then copy the values to the others.
So my guess is that we were lucky that it worked, but were randomizing different initial poses for all collections when in fact only the randomized initial pose of the selected collection key was being used.

miguelriemoliveira, 2021-09-20:
> Redo the experiments???

NEVER : - )

closed 2021-09-20 by miguelriemoliveira