lardemua/atom #318: Initial estimate disappears when the computer is rebooted?

danifpdra, 2021-09-30:
HI @miguelriemoliveira,

Is the initial estimate that was set supposed to disappear when the computer is rebooted?  Shouldn't there be an option to save it?

[image]

miguelriemoliveira, 2021-09-30:
It should be stored in the file https://github.com/lardemua/larcc/blob/main/larcc_calibration/urdf/initial_estimate.urdf.xacro

Did you reconfigure the package? If you did it deletes the initial estimate ...

danifpdra, 2021-09-30:
No, I didn't reconfigure

miguelriemoliveira, 2021-09-30:
Strange ... you'll have to set the initial estimate again, and this time make a local backup of the file above. If it happens again we try to reuse the file.

danifpdra, 2021-09-30:
I already did that. Thanks

closed 2021-11-04 by danifpdra