lardemua/atom #325: Disparities in calibration results

danifpdra, 2021-11-09:
Hi @miguelriemoliveira ,

I just tested with mmtbot... you can still see the ghost but not like larcc

[image]

miguelriemoliveira, 2021-11-09:
Hi @danifpdra ,

I cannot run the mmtbot test. I think our bags and datasets are deprecated because of the changes in the description and gazebo packages. Which are you using? Perhaps you can add them to the readme.

Also, let me know the command.

I think that we should clarify why this is also ocurring in mmtbot (even if to a lesser extent)

danifpdra, 2021-11-10:
The bag is here: https://uapt33090-my.sharepoint.com/:u:/g/personal/danielarato_ua_pt/EQyry8hklGpPn_JPyUTYOy4BMAIDgD43Tosi6w7TafTiJQ?e=8FopnX

The dataset is here: https://uapt33090-my.sharepoint.com/:u:/g/personal/danielarato_ua_pt/EX8xycbjBS9PjEMsbr13bJoBZJcNCRVttXd7xijCj30RHg?e=Jw6VMz

I will add this to the readme also.

I used 

`rosrun atom_calibration calibrate -uic -json $ATOM_DATASETS/mmtbot/test_mmtbot/data_collected.json -nig 0.05 0.05 -rv -ipg -v -phased -si`

miguelriemoliveira, 2021-11-10:
Migrating to #327. This seems to be the problem...

miguelriemoliveira, 2022-01-09:
This was solved, I think. Closing.

closed 2022-01-09 by miguelriemoliveira